espressif/vscode-esp-idf-extension #421: [Bug Report]: Can't start OpenOCD

gfmoore, 2021-05-28:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [ x] Make sure you have searched for existing bugs and features request before you post an issue.
- [x] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [ x] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
I'm trying to learn how to debug the basic blinky. I can build and flash and get an led flashing on the breadboard (after a days work :( )

I'm following the instructions here: https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/debugging.md

I cannot get past the first step of opening the OpenOCD server.

When I open the blink.c and press F5 I get:
Open On-Chip Debugger  v0.10.0-esp32-20200709 (2020-07-09-08:54)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 20000 kHz

Info : Configured 2 cores
Info : Lis❌ tening on port 6666 for tcl connections
Error: couldn't bind telnet to socket on port 4444: No error
[Stopped] : OpenOCD Server

I can see the OpenOCD server (Stopped) in the status bar. I click on it, accept the Start OCD and same error.

A popup with the message: OpenOCD Exit with non-zero error code 1  
Not helpful
I click on report - what should I see?

**To Reproduce**
Steps to reproduce the behavior (including the full log).

As above

**Expected behavior**
A clear and concise description of what you expected to happen.

I expect to be able to set a breakpoint and then single step through application

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**

- OS Name & Version: [e.g. MacOS X 10.14.5]
- VSCode Version: [e.g. 1.35.1]
- ESP-IDF Version: [e.g. v3.2.2]
- Python Version: [e.g. 2.7.10]

> **NOTE**: You can use the `ESP-IDF: Doctor command` to generate a report of your configuration.

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19043 
System environment variable PATH 
 C:\Program Files (x86)\STMicroelectronics\st_toolset\asm;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\PuTTY\;C:\xampp\mysql\bin;C:\Program Files\cmder\vendor\git-for-windows\mingw64\bin;C:\Program Files\xpdf\bin64;C:\Program Files\Java\jdk-11.0.1\bin;C:\Program Files\Java\jdk-11.0.1;C:\Program Files\javafx-sdk-11.0.1;C:\Program Files\javafx-sdk-11.0.1\lib;C:\Program Files (x86)\AOMEI Backupper;C:\Program Files\R\R-3.5.1\bin;C:\Users\Gordon\AppData\Local\Programs\Python\Python37-32;C:\Users\Gordon\AppData\Local\Microsoft\WindowsApps;C:\Users\Gordon\AppData\Local\Programs\Microsoft VS Code\bin;C:\Program Files (x86)\sendmidi-windows-1.0.14;C:\Program Files (x86)\receivemidi-windows-1.1.0-beta1;C:\MinGW\bin;C:\MinGW\include;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files (x86)\Plantronics\Spokes3G\;C:\Program Files\dotnet\;C:\ProgramData\chocolatey\bin;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Git\cmd;C:\Program Files (x86)\dotnet\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin;C:\Program Files\jdk-15.0.2\bin;C:\Program Files\AdoptOpenJDK\jdk8u192-b12\bin;C:\Program Files\Java\jdk1.8.0_211\bin;C:\Android\android-sdk\tools;C:\Android\android-sdk\platform-tools;C:\Android\android-sdk\tools\bin;C:\Program Files\IVI Foundation\VISA\Win64\Bin\;C:\Program Files (x86)\Common Files\Sage SData\;C:\Program Files\PowerShell\7\;C:\Program Files (x86)\nodejs\;C:\Users\Gordon\AppData\Local\Programs\Python\Python39;C:\Users\Gordon\AppData\Local\Programs\Python\Python39\Scripts;C:\Program Files\Nordic Semiconductor\nrf-command-line-tools\bin;C:\Program Files\Git\bin;C:\Program Files\Git;C:\Users\Gordon\AppData\Local\Programs\Python\Python39\Scripts\;C:\Users\Gordon\AppData\Local\Programs\Python\Python39\;C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2020-q4-major\bin;C:\Users\Gordon\.cargo\bin;C:\Users\Gordon\AppData\Local\Microsoft\WindowsApps;C:\Program Files (x86)\sendmidi-windows-1.0.14;C:\Program Files (x86)\receivemidi-windows-1.1.0-beta1;C:\MinGW\bin;C:\MinGW\include;C:\xampp\php;C:\Users\Gordon\AppData\Local\Programs\Microsoft VS Code Insiders\bin;C:\Program Files\R\R-3.6.3\bin;C:\Program Files\R\R-3.6.3;C:\Program Files\Java\jdk-13.0.1\bin;C:\Users\Gordon\Documents\PowerShell\Scripts;C:\Users\Gordon\.dotnet\tools;C:\Android\android-sdk\emulator;C:\Android\android-sdk\emulator\qemu\windows-x86_64;C:\Program Files\SDCC\bin;C:\Users\Gordon\AppData\Roaming\Python\Python39\Scripts;C:\Users\Gordon\AppData\Local\Android\Sdk\platform-tools;C:\Users\Gordon\AppData\Roaming\npm;C:\Users\Gordon\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\Scripts;C:\Users\Gordon\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\Scripts;C:\Program Files\Nordic Semiconductor\nrf-command-line-tools\bin; 
Visual Studio Code version 1.57.0-insider 
Visual Studio Code language en 
Visual Studio Code shell C:\Program Files\PowerShell\7\pwsh.exe 
ESP-IDF Extension version 1.0.3 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-IDF Path (idf.espIdfPath) c:\Users\Gordon\esp-idf\esp-idf
Custom extra paths (idf.customExtraPaths) C:\Users\Gordon\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin;C:\Users\Gordon\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin;C:\Users\Gordon\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Users\Gordon\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Users\Gordon\.espressif\tools\cmake\3.16.4\bin;C:\Users\Gordon\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin;C:\Users\Gordon\.espressif\tools\ninja\1.10.0;C:\Users\Gordon\.espressif\tools\idf-exe\1.0.1;C:\Users\Gordon\.espressif\tools\ccache\3.7;C:\Users\Gordon\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"C:\\Users\\Gordon\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20200709/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) C:\Users\Gordon\.espressif\python_env\idf4.2_py3.9_env\Scripts\python.exe
Serial port (idf.port) /dev/ttyUSB1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,target/esp32.cfg
ESP-IDF Tools Path (idf.toolsPath) ${env:IDF_TOOLS_PATH}
[... 122 lines cut ...]
        }
    ],
    "version": 4
} 
--------------------------------------------------------------------------------------------------------------------------------------------

**Logs**
If applicable, please share the log file which can be obtained from

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

**Extension Configuration settings**
Open Command Palette (F1) and type `Preferences: Open Settings (JSON)` or use `{CURRENT_PROJECT}/.vscode/settings.json` to get the following extension configuration settings value:

- `idf.espIdfPath` (MacOS or Linux) or `idf.espIdfPathWin` (Windows) also called `Path to locate ESP-IDF framework (IDF_PATH)` in `Preferences: Open Settings (UI)`.
- `idf.pythonBinPath` (MacOS or Linux) or `idf.pythonBinPathWin` (Windows) also called `Python absolute binary path used to execute ESP-IDF Python Scripts` in `Preferences: Open Settings (UI)`.
- `idf.openOcdConfigs` also called `List of configuration files inside OpenOCD Scripts directory` in `Preferences: Open Settings (UI)`.
- `idf.customExtraPaths` also called `Paths to be appended to PATH` in `Preferences: Open Settings (UI)`.
- `idf.customExtraVars` also called `Variables to be added to system environment variables` in `Preferences: Open Settings (UI)`.
- `idf.adapterTargetName` also called `Target name for ESP-IDF Debug Adapter` in `Preferences: Open Settings (UI)`.
- `idf.customAdapterTargetName` is used when `idf.adapterTargetName` is set to `custom`.

**Additional context**
Add any other context about the problem here.

gfmoore, 2021-05-28:
Looked under Debug Console:

Debug adapter -> Extension: DEBUG_ADAPTER_STARTED
Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT
Failed to connect to "remote 127.0.0.1:3333"!

Tried turning firewall off, no difference.

gfmoore, 2021-05-28:
Uhmmm some progress.
At the Terminal tab I tried openocd --help and discovered that it wasn't in the path!!!! 
I checked my system path and discovered that there was so much junk there that it had broken a character limit 2000 chars or so. So cleaned all that out and added the path to openocd.exe (Had to reboot which is odd? for win10)

So openocd --help now works.

Tried opening the OpenOCD server: More errors
```
Open On-Chip Debugger  v0.10.0-esp32-20200709 (2020-07-09-08:54)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 20000 kHz

Info : Configured 2 cores
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
❌ Error: no d❌ evice found
Error: unable to open ftdi device with vid 0403, pid 6010, description '*', serial '*' at bus location '*'
❌ Error: no device found
❌ Error: unable to open ftdi device with vid 0403, pid 6014, description '*', serial '*' at bus location '*'

Assertion failed!

Program: C:\Users\Gordon\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin\openocd.exe
File: src/jtag/core.c, Line 343

Expression: jtag_trst == 0
[Stopped] : OpenOCD Server
```

I can understand that the install routine might not have recognised that the path wasn't added and had failed - but perhaps some attention can be given to that issue?

Gordon

gfmoore, 2021-05-28:
Looking at the error. I says something about a device not being found? Well the device is sitting here blinking. So, and I'm not quite clued up on this, is it looking for some other device like a "jtag" segger like debugger? I don't really understand these things (yet), but I was under the impression that the esp32 board (mine is from China and has 38 pins and looks like this:
[image]

Does the esp-idf debugger work with this type  of board?

Gordon

gfmoore, 2021-05-28:
Got  closer I think.

I went to device manager, found the Silicon Labs CP210x USB to UART Bridge (COM3)
And for the device ids found 
USB\VID_10C4&PID_EA60&REV_0100
USB\VID_10C4&PID_EA60

These do not match what is shown in the error message.

My new question is how do I change the OpenOCD config?

Gordon

(the adventure continues...)

gfmoore, 2021-05-28:
Uhmm, the only place I can find that mentions vid 0403 pid 6010 is in ...esp-idf\docs\en\api-guides\jtag-debugging\configure-ft2232h-jtag.rst, but it isn't hard-coded. I am getting more and more to the conclusion that this is some segger device and my board doesn't support it?

I've had enough now, I'm going to watch television  :(

Gordon

gfmoore, 2021-05-29:
So I've now come to the conclusion that my board has no ftdi? JTAG chip (WROVER?) so it cannot be debugged. 

I basically bought a device that isn't really suitable for development. 

To maintainers. Could you please add a section to the Debug section explaining for noobs like me that you need a proper setup for the debugging to work. Anyone who knows what they are doing doesn't need to be told these things do they!!!

Also, please let noobs know for the  that they need a breadboard and LED to actually see anything flash. After using a nrf52840 I was expecting to see some kind of onboard led flashing. I couldn't work out what GPIO pin to use to get a LED to flash!!! 
 Expectations eh.

Waste of a day.

Gordon

I'll let YOU close so I know you've read it.

brianignacio5, 2021-05-31:
Have you take a look to configure FTDI drivers as shown here https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/configure-ft2232h-jtag.html

gfmoore, 2021-05-31:
Yes,  but how can I check my board has this functionality? I don't think it does.

brianignacio5, 2021-06-01:
For openOCD compatible boards check [Debug Adapter Hardware](http://openocd.org/doc/html/Debug-Adapter-Hardware.html) and [ESP-IDF Selecting JTAG adapter](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/index.html#jtag-debugging-selecting-jtag-adapter).

gfmoore, 2021-06-01:
Thanks for your help, it's appreciated :)

closed 2021-06-01 by gfmoore