espressif/vscode-esp-idf-extension #424: Can't get debugging running within the extension

mickpf, 2021-06-02:
Hello,

Since several days I'm trying to get either the Eclipse plugin or the VS Code Extension running so that I'll be able to debug the code generated with ESP-IDF.
I gave up trying to get the Eclipse plug-in running properly because too much things do not work as they should do it, problems with the toolbar "Launch Target" and much more. I don't want to waste more time anymore!

Now I'm trying to set up the VS Code Extension.
I've installed the extension from "scratch", followed your instructions so far as possible, but I can't get the debugging runnung!

Could you help me, please?

In the following I have put the problems together.
Please let me know if you need any further data.

1. When I start VS Code with a fresh created workspace, a new entry appears in the log file "${HOME}/.vscode/extensions/espressif.esp-idf-extension-1.0.3/esp_idf_vsc_ext.log":

    {
        "message": "Open On-Chip Debugger  v0.10.0-esp32-20210401 (2021-04-01-15:45)
                    Licensed under GNU GPL v2\nFor bug reports, read
                        http://openocd.org/doc/doxygen/bugs.html",
        "stack": "Error: Open On-Chip Debugger  v0.10.0-esp32-20210401 (2021-04-01-15:45)
                    Licensed under GNU GPL v2
                    For bug reports, read
                        http://openocd.org/doc/doxygen/bugs.html
                        
                        at /home/mick/.vscode/extensions/espressif.esp-idf-extension-1.0.3/dist/extension.js:1:8461
                        at ChildProcess.exithandler (child_process.js:308:7)
                        at ChildProcess.emit (events.js:315:20)
                        at maybeClose (internal/child_process.js:1048:16)
                        at Socket.<anonymous> (internal/child_process.js:439:11)
                        at Socket.emit (events.js:315:20)\n\tat Pipe.<anonymous> (net.js:673:12)",
        "level": "error",
        "timestamp": "2021-06-02T06:41:08.752Z"
    }

2. Pressing the button "Debug" does not lead to any reaction or action (no log entries, no output, nothing!).

    ==> I changed the file "${workspaceFolder}/.vscode/launch.json" :

        {
            "version": "0.2.0",
            "configurations": [
                {
                    "type": "espidf",
                    "name": "Launch",
                    "request": "launch",
                    "debugPort": 9998,
                    "logLevel": 2,
                    "mode": "manual",
                    "initGdbCommands": [
                        "target remote :3333",
                        "symbol-file /home/mick/Projects/ESP32/EVE/build/EVE.elf",
                        "mon reset halt",
                        "flushregs",
                        "thb app-main"
                    ]
                }
            ]
        }    

    This change leads to the output as can be seen in the attachment "2. Output Button Debug 1.png".
    
    ==> Execution of "ESP-IDF: Select OpenOCD Board Configuration", selection "ESP32 chip (via ESP-PROG)"

        The attempt to start "OpenOCD Server" leads to the output as can be seen in the attachment "2. Output Start OpenOCD Server 1.png".

        Error detected: The file name "esp32_devkitjv1.cfg" is wrong! The correct file name is "esp32_devkitj_v1.cfg".

    ==> The execution of "ESP-IDF: Device configuration", selection "OpenOCD config files" displays and saves the correct file name.
    
    Starting "OpenOCD Server" again leads to the output as can be seen in the attachment "2. Output Start OpenOCD Server 2.png". It works now!
    I miss the info line(s) for esp32.cpu0 in this output! Sometimes this window displays esp32.cpu0 without the cpu1! Why it does display only
    one cpu?
    
    Pressing the "Debug" button does not lead to any reaction or action (no log entries, no output, nothing!).
    
At this point I can't get any further!

I've tried to attach the named screenshots (format: png, size < 120 kB per file), but it does not work!
How can I send them to you?

Kind Regards,
Michael

mickpf, 2021-06-02:
2nd try to attach some screenshots...
[image]
[image]
[image]

It workes now...

brianignacio5, 2021-06-03:
Please don't duplicate your issue.

closed 2021-06-03 by brianignacio5