espressif/vscode-esp-idf-extension #425: A couple of suggestions for improvement and questions

mickpf, 2021-06-02:
Hello,

1. After setting up a new workspace on the first attempt to flash, you will be asked for the type: UART or JTAG. There is no way to change this type later (except directly with a text editor in the file "${PROJECT_NAME}.code-workspace: <" idf.flashType ":" UART ">). Changing the file can possibly trigger side effects (affect other projects)!

2. Project-related settings for ESP-IDF are stored global under "Preferences: Open Settings (JSON)":
         "idf.customExtraPaths", "idf.customExtraVars", "idf.port" etc.
    That has already led to problems!

3. Is it possible to use variables like "${workspaceFolder}" and "${command: espIdf.getProjectName}" in the file "launch.json", section "initGdbCommands"?

Kind Regards,
Michael

brianignacio5, 2021-06-03:
With regards to 1 and 2:

Visual Studio has a priority layer of settings as shown in https://code.visualstudio.com/docs/editor/multi-root-workspaces#_settings where a `<project_dir>/.vscode/settings.json` would be chosen over any `workspace.code-workspace` and that over (global) user settings.

You can use the `idf.saveScope` configuration setting to determine where to save extension settings when using the extension commands as shown in [settings documentation](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/SETTINGS.md#extension-behaviour-settings)

About 3) we don't support it yet but we could do so. Thanks for the feature request.

brianignacio5, 2021-06-11:
Closing this issue for lack of response.

Please comment here to reopen.

closed 2021-06-11 by brianignacio5