espressif/vscode-esp-idf-extension #426: [Bug Report]: can not build on windows

chain710, 2021-06-03:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [x] Make sure you have searched for existing bugs and features request before you post an issue.
- [x] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [x] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
can not build on windows 10

**To Reproduce**
- press F1 
- New Project
- Create project using template hello-world
- click Build
- get error message like ```CMake Error at CMakeLists.txt:5 (include): include could not find requested file: /tools/cmake/project.cmake```

**Expected behavior**
a good example should build without error

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Environment (please complete the following information):**

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19042 
System environment variable PATH 
 C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\ProgramData\chocolatey\bin;C:\Program Files\Git\cmd;E:\Go\bin;C:\Program Files\Docker\Docker\resources\bin;C:\ProgramData\DockerDesktop\version-bin;C:\Users\chain710\AppData\Local\Programs\Python\Python37\Scripts\;C:\Users\chain710\AppData\Local\Programs\Python\Python37\;C:\Users\chain710\AppData\Local\Microsoft\WindowsApps;;C:\Users\chain710\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\chain710\go\bin 
Visual Studio Code version 1.56.2 
Visual Studio Code language zh-cn 
Visual Studio Code shell C:\WINDOWS\System32\wsl.exe 
ESP-IDF Extension version 1.0.3 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-IDF Path (idf.espIdfPath) E:\workspace\esp-idf
Custom extra paths (idf.customExtraPaths) C:\Users\chain710\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin;C:\Users\chain710\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin;C:\Users\chain710\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Users\chain710\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Users\chain710\.espressif\tools\cmake\3.16.4\bin;C:\Users\chain710\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin;C:\Users\chain710\.espressif\tools\ninja\1.10.0;C:\Users\chain710\.espressif\tools\idf-exe\1.0.1;C:\Users\chain710\.espressif\tools\ccache\3.7;C:\Users\chain710\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"C:\\Users\\chain710\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20200709/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) C:\Users\chain710\.espressif\python_env\idf4.2_py3.7_env\Scripts\python.exe
Serial port (idf.port) /dev/ttyUSB1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) ${env:IDF_TOOLS_PATH}
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-IDF Custom extra paths
Access to C:\Users\chain710\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin: true
Access to C:\Users\chain710\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin: true
Access to C:\Users\chain710\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin: true
Access to C:\Users\chain710\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin: true
Access to C:\Users\chain710\.espressif\tools\cmake\3.16.4\bin: true
Access to C:\Users\chain710\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin: true
Access to C:\Users\chain710\.espressif\tools\ninja\1.10.0: true
Access to C:\Users\chain710\.espressif\tools\idf-exe\1.0.1: true
Access to C:\Users\chain710\.espressif\tools\ccache\3.7: true
Access to C:\Users\chain710\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) false
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version 2.24.1.windows.2
ESP-IDF version 4.2.1
Python version 3.7.3
Python's pip version 21.1.2
-------------------------------------------------- Python packages in idf.pythonBinPath ----------------------------------------------------
certifi version: 2021.5.30
chardet version: 4.0.0
click version: 8.0.1
colorama version: 0.4.4
contextlib2 version: 0.6.0.post1
future version: 0.18.2
gcovr version: 4.2
idf-component-manager version: 0.2.1a1
idna version: 2.10
importlib-metadata version: 4.4.0
Jinja2 version: 3.0.1
lxml version: 4.6.3
MarkupSafe version: 2.0.1
pip version: 21.1.2
psutil version: 5.8.0
pygdbmi version: 0.9.0.2
pywin32 version: 301
PyYAML version: 5.4.1
requests version: 2.25.1
requests-toolbelt version: 0.9.1
schema version: 0.7.4
semantic-version version: 2.8.5
setuptools version: 40.8.0
six version: 1.16.0
tqdm version: 4.61.0
typing-extensions version: 3.10.0.0
urllib3 version: 1.26.5
websocket-client version: 1.0.1
wheel version: 0.36.2
xmlrunner version: 1.7.7
zipp version: 3.4.1
---------------------------------------------------- Check ESP-IDF python requirements.txt -------------------------------------------------
Check ESP-IDF Python packages undefined
---------------------------------------------------- Check extension requirements.txt ------------------------------------------------------
Check Extension Python packages undefined
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages undefined
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at Command failed: C:\Users\chain710\.espressif\python_env\idf4.2_py3.7_env\Scripts\python.exe E:\workspace\esp-idf\tools\check_python_dependencies.py -r "E:\workspace\esp-idf\requirements.txt"

--------------------------------------------------------------------------------------------------------------------------------------------

**Logs**
If applicable, please share the log file which can be obtained from

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

**Extension Configuration settings**
```
"idf.espIdfPathWin": "E:\\workspace\\esp-idf",
    "idf.pythonBinPathWin": "C:\\Users\\chain710\\.espressif\\python_env\\idf4.2_py3.7_env\\Scripts\\python.exe",
    "idf.customExtraPaths": "C:\\Users\\chain710\\.espressif\\tools\\xtensa-esp32-elf\\esp-2020r3-8.4.0\\xtensa-esp32-elf\\bin;C:\\Users\\chain710\\.espressif\\tools\\xtensa-esp32s2-elf\\esp-2020r3-8.4.0\\xtensa-esp32s2-elf\\bin;C:\\Users\\chain710\\.espressif\\tools\\esp32ulp-elf\\2.28.51-esp-20191205\\esp32ulp-elf-binutils\\bin;C:\\Users\\chain710\\.espressif\\tools\\esp32s2ulp-elf\\2.28.51-esp-20191205\\esp32s2ulp-elf-binutils\\bin;C:\\Users\\chain710\\.espressif\\tools\\cmake\\3.16.4\\bin;C:\\Users\\chain710\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20200709\\openocd-esp32\\bin;C:\\Users\\chain710\\.espressif\\tools\\ninja\\1.10.0;C:\\Users\\chain710\\.espressif\\tools\\idf-exe\\1.0.1;C:\\Users\\chain710\\.espressif\\tools\\ccache\\3.7;C:\\Users\\chain710\\.espressif\\tools\\dfu-util\\0.9\\dfu-util-0.9-win64",
    "idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"C:\\\\Users\\\\chain710\\\\.espressif\\\\tools\\\\openocd-esp32\\\\v0.10.0-esp32-20200709/openocd-esp32/share/openocd/scripts\",\"IDF_CCACHE_ENABLE\":\"1\"}"
```

**Additional context**
Add any other context about the problem here.

brianignacio5, 2021-06-04:
Are you using the extension command `ESP-IDF: Build your project` to build your project ? From the output seems you are trying to use the CMake extension instead.

chain710, 2021-06-04:
> Are you using the extension command `ESP-IDF: Build your project` to build your project ? From the output seems you are trying to use the CMake extension instead.

Built with `esp-idf: build your project` again with error:

```
CMake Error: The current CMakeCache.txt directory /mnt/e/workspace/esp32/playground/helloworld/hello_world/build/CMakeCache.txt is different than the directory e:/workspace/esp32/playground/helloworld/hello_world/build where CMakeCache.txt was created. This may result in binaries being created in the wrong place. If you are not sure, reedit the CMakeCache.txt
CMake Error: The source "/mnt/e/workspace/esp32/playground/helloworld/hello_world/CMakeLists.txt" does not match the source "E:/workspace/esp32/playground/helloworld/hello_world/CMakeLists.txt" used to generate cache.  Re-run cmake with a different source directory.
```

It seems extension tried to build in wsl, and something went wrong

brianignacio5, 2021-06-07:
Can you try to delete the build folder in your project an try again to make a fresh build? The previous build attempt cache has conflicts.

chain710, 2021-06-08:
modified vscode settings.json , then finally build ok :D

```
"terminal.integrated.defaultProfile.windows": "Command Prompt"
```

closed 2021-06-08 by chain710