espressif/vscode-esp-idf-extension #429: [Bug Report]: Some weird ESP-IDF extension behavior when use ESP-IDF: Create project from extension template >template-app

samsam4, 2021-06-09:
**Describe the bug**

1.	Start VS Code  
2.	Open an empty folder VS_WSP1
3.	View>Command Palette..> ESP-IDF: Create project from extension template> Select VS_WSP1  folder
4.	VSC opens additional window where creates template app
5.	Click on the new window to see main.c code and I see it one problem on #include <stdio>: Include file not found… (pic.1)
6.	Click back on the original VSC window to open in it main.c
7.	Now I see main.c with 6 problems on all 6 #unclude directives (pic2)
8.	Next behavior depend which window will close first and last at this step:
If I first close the window that shows 1 error (the newest window) and then close the very first window (the one from when I start VSC, that currently shows 6 errors), then next time when I reopen VSC – it will show me main.c with the 6 errors (pic3), BUT If I instead last closed  VSC window was the one with the 1 error, next time when I reopen VSC will see main.c with 1 error.
(To see both behaviors will need to delete content of  VS_WSP1 folder when VSC is not opened and repeat steps 1-7 and in step 8 choose alternate sequence of closing the two VSC windows then was used in last test)

**Expected behavior**
Consistent output in both windows that open same file 

**Screenshots**
pic1:
[image]

pic2:
[image]

pic3:
[image]

**Environment (please complete the following information):**

- OS Name & Version: OS win32 x64 10.0.19041
- VSCode Version: 1.56.2
- ESP-IDF Version: 4.2.1
- Python Version: 3.8.7

**Logs**

[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/6624436/esp_idf_vsc_ext.log)

**Extension Configuration settings**
Open Command Palette (F1) and type `Preferences: Open Settings (JSON)` or use `{CURRENT_PROJECT}/.vscode/settings.json` to get the following extension configuration settings value:

VS_WSP1\template-app\.vscode\settings.json : 
{
  "C_Cpp.intelliSenseEngine": "Tag Parser"
}

Preferences: Open Settings (JSON):
{
    "idf.espIdfPathWin": "d:\\ESP4_2_1\\ESP-IDF\\esp-idf",
    "idf.pythonBinPathWin": "d:\\ESP2_9b_tools\\.espressif\\python_env\\idf4.2_py3.8_env\\Scripts\\python.exe",
    "idf.customExtraPaths": "",
    "idf.customExtraVars": "{}"
}

brianignacio5, 2021-06-10:
Have you looked into [C/C++ Configuration](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/C_CPP_CONFIGURATION.md) ?

samsam4, 2021-06-10:
Here is my c_cpp_properties.json (ziiped) as per your request.
[c_cpp_properties.zip](https://github.com/espressif/vscode-esp-idf-extension/files/6630893/c_cpp_properties.zip)

Thanks

brianignacio5, 2021-06-10:
This is probably an issue for the Microsoft C/C++ extension itself. From the original picture, it seems headers are resolved accordingly to c_cpp_properties.json definition.

samsam4, 2021-06-10:
Could you, please, post  some example (or link to existing) settings.json for windows environment as you can see from the copy of mine that I posted before : neither "idf.customExtraPaths" nor "idf.customExtraVars" are filled from the ESP-IDF: Configure ESP-IDF extension > Advanced,  and the documentation is a little messy so I cant figure what could manually fill there?

Also when run Doctor, I  see plenty of missing paths:
```
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-IDF Custom extra paths
Access to : false
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) false
```
[Doctor_10-6-21.log](https://github.com/espressif/vscode-esp-idf-extension/files/6632593/Doctor_10-6-21.log)

Also when I try to set target (ESP-IDF: Set Espressif device target)  I get:
`{ error: [Error: non zero exit code 2 ESP-IDF v4.2.1-dirty 'cmake' must be available on the PATH to use d:\ESP4_2_1\ESP-IDF\esp-idf\tools\idf.py at ChildProcess.<anonymous> (c:\Users\SS\.vscode\extensions\espressif.esp-idf-extension-1.0.3\dist\extension.js:1:5627) at ChildProcess.emit (events.js:315:20) at Process.ChildProcess._handle.onexit (internal/child_process.js:277:12)] }`

Thanks

brianignacio5, 2021-06-11:
As shown in  [Setup documentation](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/SETUP.md#example-configuration-setting-values)

For example if required ESP-IDF Tools are:

- OpenOCD executable path is `/home/myUser/.espressif/tools/openocd-esp32/version/openocd-esp32/bin/openocd` or `C:\Users\myUser\.espressif\tools\openocd-esp32\version\openocd-esp32\bin\openocd` (Windows)
- XtensaEsp32 executable path is `/home/myUser/.espressif/tools/xtensa-esp32/version/xtensa-esp32/bin/xtensa-esp32-gcc` or `C:\Users\myUser\.espressif\tools\xtensa\version\xtensa-esp32\bin\xtensa-esp32-gcc` (Windows)

you need to set in `idf.customExtraPaths`:

- Linux/MacOS

```
/home/myUser/.espressif/tools/openocd/version/openocd-esp32/bin:/home/myUser/.espressif/tools/xtensa-esp32/version/xtensa-esp32/bin
```

- Windows

```
C:\Users\myUser\.espressif\tools\openocd-esp32\version\openocd-esp32\bin;C:\Users\myUser\.espressif\tools\xtensa-esp32\version\xtensa-esp32\bin
```

`idf.customExtraVars` is an stringified JSON object saved in Visual Studio Code's settings.json (**Make sure to replace \${TOOL_PATH} with the existing tool directory path**):

```
"idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"/home/myUser/.espressif/tools/openocd-esp32/version/openocd-esp32/share/openocd/scripts\"}"

samsam4, 2021-06-14:
Will close this issue as the new version of the extensions overrided the one where this problem was exhibit, but with the new version of the extension I cant eve pass the 'Config extension' in order to get to the same point where this problem was observed   before :(

closed 2021-06-14 by samsam4