espressif/vscode-esp-idf-extension #434: [Bug Report]:

samsam4, 2021-06-14:
**Describe the bug**
Bugs, bugs, bugs :( 
The new version 1.1.0 of the extension makes a total mess when trying to use existing ESP-IDF and ESP-IDF Tools and every attempt to Configure ESP-IDF extension finishes with "Invalid ESP-IDF"
I've made and a completely new clean installation of ESP-IDF and ESP-IDF Tools, tried with Eclipse and everything works fine, but as soon as try to use the same installation with VSC and become a total mess :( 

**To Reproduce**

1. Uninstall/delete all existing ESP-IDF and ESP-IDF Tools.
2. Download 2.8 Offline Installer from here: https://dl.espressif.com/dl/esp-idf-tools-setup-offline-2.8.exe

3. Run Installer, Run just installed Eclipse, compile 'blink' example, run Help>Product Information - check if everything is Ok.

VSC:
4. Try to Config ESP-IDF extension (doesn't give me option to use 'Existing' - only Express and Advanced) 
Bug1: Although selected Advanced>Existing ESP-IDF, it starts downloading git and python
idf-git-2.30.1-win64.zip
idf-python-3.8.7-embed-win64.zip
to D:\ESP2_8\.espressif\dist\
but same versions git and python was already in D:\ESP2_8\.espressif\ 

Downloading D:\ESP2_8\.espressif\dist\idf-git-2.30.1-win64.zip...
idf-git-2.30.1-win64.zip has (27218) KB
idf-git-2.30.1-win64.zip progress: 0.06% (15.64 / 27217.34) KB
...
idf-git-2.30.1-win64.zip progress: 100.00% (27217.34 / 27217.34) KB
Using existing D:\ESP2_8\.espressif\tools\idf-git
idf-python-3.8.7-embed-win64.zip has (19616) KB
idf-python-3.8.7-embed-win64.zip progress: 0.08% (15.64 / 19615.46) KB
...
idf-python-3.8.7-embed-win64.zip progress: 100.00% (19615.46 / 19615.46) KB
Extracted D:\ESP2_8\.espressif\tools\idf-python ...
Invalid ESP-IDF

Bug2: At the end finish with Invalid ESP-IDF (see pic. 2 and 1)

Bug3: New Downloaded idf-git-2.30.1-win64.zip was not extracted at all
Bug4: idf.gitPath that was set from the Config does not exist:

config idf extension makes VS_WSP1\.vscode\settings.json:
{
    "idf.gitPath": "D:\\ESP2_8\\.espressif\\tools\\idf-git\\cmd\\git.exe"
}
but the actual git path is D:\ESP2_8\.espressif\tools\idf-git\2.30.1\cmd\git.exe  (see pic. 2 and 3)

Bug5: Downloaded idf-python-3.8.7-embed-win64.zip duplicated already existing same files that were already there in D:\ESP2_8\.espressif\tools\idf-python\3.8.7\...
but the Config extension put them one step up in D:\ESP2_8\.espressif\tools\idf-python\...
(see pic.3)

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**

[image]

[image]

[image]

**Environment (please complete the following information):**

- OS Name & Version: OS win32 x64 10.0.19041
- VSCode Version:1.57.0
- ESP-IDF Version: 4.2 (from v2.8 offline installer)
- Python Version: 3.8.7  (from v2.8 offline installer + duplicate downloaded from ESP-IDF extension)

[DoctorReport.txt](https://github.com/espressif/vscode-esp-idf-extension/files/6648544/DoctorReport.txt)

**Logs**

[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/6648543/esp_idf_vsc_ext.log)

**Extension Configuration settings**
`Preferences: Open Settings (JSON)`  : 
{
}

`{CURRENT_PROJECT}/.vscode/settings.json` :
{
    "idf.gitPath": "D:\\ESP2_8\\.espressif\\tools\\idf-git\\cmd\\git.exe"
}

samsam4, 2021-06-14:
**Bug6**: Config ESP-IDF extension: Doesn't show option to use 'Existing' - only Express and Advanced. Although already have installed and verified with Eclipse ESP-IDF and ESP-IDF Tools installations.
Also have Windows environment variable  `IDF_TOOLS_PATH=D:\ESP2_8\.espressif`
and in the same path have a file ` D:\ESP2_8\.espressif\esp_idf.json` : 

```
{
    "$schema": "http://json-schema.org/schema#",
    "$id": "http://dl.espressif.com/dl/schemas/esp_idf",
    "_comment": "Configuration file for ESP-IDF Eclipse plugin.",
    "_warning": "Use / or \\ when specifying path. Single backslash is not allowed by JSON format.",
    "gitPath": "D:/ESP2_8/.espressif/tools/idf-git/2.30.1/cmd/git.exe",
    "idfToolsPath": "D:/ESP2_8/.espressif",
    "idfSelectedId": "esp-idf-f4a3ead534e060e74bbaf7cf3e4cf80b",
    "idfInstalled": {
        "esp-idf-f4a3ead534e060e74bbaf7cf3e4cf80b": {
            "version": "4.2",
            "python": "D:/ESP2_8/.espressif/python_env/idf4.2_py3.8_env/Scripts/python.exe",
            "path": "D:/ESP2_8/esp-idf/"
        }
    }
}
```

which has correct data for most of the paths/variables required for settings.json

This is the screen shot while still was the old version of the extension (I think 1.0.3 was the version): 
[image]

Here is what I have with the new version 1.1.0:

[image]

andoriyu, 2021-06-16:
I don't know if you already listed it, but extension is all fails at "Install Python blahbla" step because file paths are all wrong. There is no way to resume installation. Pretty bad release :(

samsam4, 2021-06-16:
I didn't notice exact same message (unless is from the esp_idf_vsc_ext.log which I didn't explore in details), but my  situation might be a little different  than yours, because I already had ESP-IDF and tools installed and tested with Eclipse (incl. Python) and the VSC Extension actually installed a duplicate of the already existing Python, but shifted one step up on the directory tree (see pic.3 and Bug5 in my first post here), so I either way had at least one Python with correct path available and maybe that is why I didn't see same message. 
But if this is the only problem you have, you can always move the installed Python from your first try in the correct path (see my pic3 above) then run again    the extension Config and probably you will get rid of at least this obstacle :(

brianignacio5, 2021-06-22:
We're very sorry for these issues @andoriyu @samsam4 

I've made a PR to fix these issues. Can you try this [installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/69475619.zip)?

With regards to bug 6, if your extensions has a valid `idf.espIdfPath` it will not search in `esp_idf.json`. Maybe you can try to remove all `idf.*` settings from settings.json to test esp_idf.json values are being read.

samsam4, 2021-06-23:
@brianignacio5

Going even worse ☹

Tried 3 different combinations:
1.	Install vsix without uninstalling already there extension 1.1.0 (esp_idf_vsc_ext - Copy03.log)
[esp_idf_vsc_ext - Copy03.log](https://github.com/espressif/vscode-esp-idf-extension/files/6703248/esp_idf_vsc_ext.-.Copy03.log)

2.	Uninstall extension 1.1.0 then install vsix (esp_idf_vsc_ext - Copy04.log)
[esp_idf_vsc_ext - Copy04.log](https://github.com/espressif/vscode-esp-idf-extension/files/6703250/esp_idf_vsc_ext.-.Copy04.log)

3.	Delete current working copy of ESP-IDF tools and replaced with a virgin copy I made after successfully installed it before (esp_idf_vsc_ext - Copy05.log)
[esp_idf_vsc_ext - Copy05.log](https://github.com/espressif/vscode-esp-idf-extension/files/6703255/esp_idf_vsc_ext.-.Copy05.log)

All results are identical:
-	When run 'Configure ESP-IDF extension' it doesn’t show me ‘Use Existing’ option – only Express and Advanced,
-	when select Advanced > Find ESP-IDF in your system, point to folder with ESP-IDF, then press ‘Install’, the screen frizzes for couple minutes then release in same place. No any output on the OUTPUT pane.
-	From what I can see from error log: … `Command failed: git --version\n'git' is not recognized as an internal or external command` …

So maybe you should change a little logic workflow when ‘Configure ESP-IDF extension ’using existing ESP-IDF installation?

My suggestion:
As soon as user select > 'Find ESP-IDF in your system', don’t ask even for the path to ESP-IDF yet, just try to locate existing ESP-IDF tools, git, python and whatever is necessary. In my case (and everybody who used ESP-IDF installer and Windows OS) this would be quite easy as I have already from installer set a Windows env var: IDF_TOOLS_PATH=D:\ESP2_8\.espressif    and there I have a esp_idf.json file where correct file paths to ESP-IDF, git, python etc. are available;
 and if extension can’t find it – to ask the user to point to existing Tools folder, resp. git and python if cant find them in default subdirectory of the tools folder!
Same  procedure could be used  for the extension to decide if can find and shows "Use Existing Setup" at the first screen of Configure Extension.

Regards

[image]

samsam4, 2021-06-24:
After heavy fight got some workaround :) :
Adding to Windows 'Path' variable paths to existing git and python made "Use Existing Setup" to reappear. So I used this option and setup finish without exceptions/errors. Didn't try yet "Advanced option".
So you need somehow to add the git and python paths from existing installation to the VSC environment as better not to pollute all the time Windows Path with those extra paths.

Create and build template-app didn't go so smooth though - same old problem that we discuss earlier [here](https://github.com/espressif/vscode-esp-idf-extension/issues/429), and some other problems that appears occasionally, but will need some more experimenting to find some pattern when which problem popups :(
Regards

brianignacio5, 2021-06-24:
Sorry I have updated the vsix a couple of times. Can you delete current extension folder and try downloading the last link again ?

For the `Use existing` part:
* IF the extension is already configured (meaning `idf.espIdfPathWin` and other settings are already set) then Use Existing will show this one to use.
* If the extension is not configured, it will try to find `esp_idf.json` in `idf.toolsPathWin` (which default value is the environment variable IDF_TOOLS_PATH) and get gitPath, espIdfPath and toolsPath from there.

I've tested on Windows 10 in a clean setup and it is working.

1) Make sure to delete all `idf.*` configuration settings from your user, workspace and workspace folder settings.json if you are trying to configure the extension with `Use existing` setup mode.
2) Git path and python paths are added to the environment variable Path from the setup download result. After the extension is configured the git and python path are added to the environment variable Path from `idf.gitPath` and `idf.pythonBinPathWin`

samsam4, 2021-06-25:
Still 'ESP-IDF: Configure ESP-IDF extension' from esp-idf-extension.vsix (creted 23-‎Jun-‎21, ‏‎12:54:50 PM) doesn't work as suppose to work without set some extra  path to git and python in the Windows system variable 'Path'

What I did:

Delete used, Copy virgin esp tools
Delelete `C:\Users\xx\.vscode\extensions\espressif.esp-idf-extension-1.1.0`
rename Windows paths that I put extra before to git and python to non existing paths
```
	D:\ESP2_8\.espressif\python_env\idf4.2_py3.8_env\Scripts\00
	D:\ESP2_8\.espressif\tools\idf-git\2.30.1\cmd\00
```
create new folder vs_wsp2
start VSC
open folder vs_wsp2
install from vsix >  esp-idf-extension.vsix (creted 23-‎Jun-‎21, ‏‎12:54:50 PM)
dont have any .log so far
ESP-IDF: Configure ESP-IDF extension

[esp_idf_vsc_ext - Copy01.log](https://github.com/espressif/vscode-esp-idf-extension/files/6716529/esp_idf_vsc_ext.-.Copy01.log)

Doesnt show USE EXISTING... - only Express and Advanced
Exit VSC
____
clear esp_idf_vsc_ext.log
start VSC
ESP-IDF: Configure ESP-IDF extension
same story as before :(

[esp_idf_vsc_ext - Copy02.log](https://github.com/espressif/vscode-esp-idf-extension/files/6716530/esp_idf_vsc_ext.-.Copy02.log)

Exit VSC
__________

clear esp_idf_vsc_ext.log
restore Windows paths that I put extra to git and python to existing paths
```
	D:\ESP2_8\.espressif\python_env\idf4.2_py3.8_env\Scripts\
	D:\ESP2_8\.espressif\tools\idf-git\2.30.1\cmd\
```
start VSC (folder vs_wsp2 is open/active)
	 (now take some more times to process, but at the end shows EXPRESS, ADVACED, **USE EXISTING SETUP**)

[esp_idf_vsc_ext - Copy03.log](https://github.com/espressif/vscode-esp-idf-extension/files/6716531/esp_idf_vsc_ext.-.Copy03.log)

and have a  C:\Users\..\AppData\Roaming\Code\User\settings.json , but this is maybe from before???
```
{
    "C_Cpp.updateChannel": "Insiders",
    "idf.espIdfPathWin": "D:/ESP2_8/esp-idf/",
    "idf.pythonBinPathWin": "D:/ESP2_8/.espressif/python_env/idf4.2_py3.8_env/Scripts/python.exe",
    "idf.toolsPathWin": "D:\\ESP2_8\\.espressif",
    "idf.customExtraPaths": "D:\\ESP2_8\\.espressif\\tools\\xtensa-esp32-elf\\esp-2020r3-8.4.0\\xtensa-esp32-elf\\bin;D:\\ESP2_8\\.espressif\\tools\\xtensa-esp32s2-elf\\esp-2020r3-8.4.0\\xtensa-esp32s2-elf\\bin;D:\\ESP2_8\\.espressif\\tools\\esp32ulp-elf\\2.28.51-esp-20191205\\esp32ulp-elf-binutils\\bin;D:\\ESP2_8\\.espressif\\tools\\esp32s2ulp-elf\\2.28.51-esp-20191205\\esp32s2ulp-elf-binutils\\bin;D:\\ESP2_8\\.espressif\\tools\\cmake\\3.16.4\\bin;D:\\ESP2_8\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20200709\\openocd-esp32\\bin;D:\\ESP2_8\\.espressif\\tools\\ninja\\1.10.0;D:\\ESP2_8\\.espressif\\tools\\idf-exe\\1.0.1;D:\\ESP2_8\\.espressif\\tools\\ccache\\3.7;D:\\ESP2_8\\.espressif\\tools\\dfu-util\\0.9\\dfu-util-0.9-win64",
    "idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"D:\\\\ESP2_8\\\\.espressif\\\\tools\\\\openocd-esp32\\\\v0.10.0-esp32-20200709/openocd-esp32/share/openocd/scripts\",\"IDF_CCACHE_ENABLE\":\"1\"}",
    "idf.gitPath": "D:/ESP2_8/.espressif/tools/idf-git/2.30.1/cmd/git.exe"

}
```

clear C:\Users\xx\AppData\Roaming\Code\User\settings.json
Exit VSC
_______________
clear esp_idf_vsc_ext.log
start VSC
open folder vs_wsp2
ESP-IDF: Configure ESP-IDF extension
fills  C:\Users\..\AppData\Roaming\Code\User\settings.json :
```
{
    "idf.espIdfPathWin": "D:/ESP2_8/esp-idf/",
    "idf.pythonBinPathWin": "D:/ESP2_8/.espressif/python_env/idf4.2_py3.8_env/Scripts/python.exe",
    "idf.toolsPathWin": "D:\\ESP2_8\\.espressif",
    "idf.customExtraPaths": "D:\\ESP2_8\\.espressif\\tools\\xtensa-esp32-elf\\esp-2020r3-8.4.0\\xtensa-esp32-elf\\bin;D:\\ESP2_8\\.espressif\\tools\\xtensa-esp32s2-elf\\esp-2020r3-8.4.0\\xtensa-esp32s2-elf\\bin;D:\\ESP2_8\\.espressif\\tools\\esp32ulp-elf\\2.28.51-esp-20191205\\esp32ulp-elf-binutils\\bin;D:\\ESP2_8\\.espressif\\tools\\esp32s2ulp-elf\\2.28.51-esp-20191205\\esp32s2ulp-elf-binutils\\bin;D:\\ESP2_8\\.espressif\\tools\\cmake\\3.16.4\\bin;D:\\ESP2_8\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20200709\\openocd-esp32\\bin;D:\\ESP2_8\\.espressif\\tools\\ninja\\1.10.0;D:\\ESP2_8\\.espressif\\tools\\idf-exe\\1.0.1;D:\\ESP2_8\\.espressif\\tools\\ccache\\3.7;D:\\ESP2_8\\.espressif\\tools\\dfu-util\\0.9\\dfu-util-0.9-win64",
    "idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"D:\\\\ESP2_8\\\\.espressif\\\\tools\\\\openocd-esp32\\\\v0.10.0-esp32-20200709/openocd-esp32/share/openocd/scripts\",\"IDF_CCACHE_ENABLE\":\"1\"}",
    "idf.gitPath": "D:/ESP2_8/.espressif/tools/idf-git/2.30.1/cmd/git.exe"
   

}
```

[esp_idf_vsc_ext - Copy04.log](https://github.com/espressif/vscode-esp-idf-extension/files/6716536/esp_idf_vsc_ext.-.Copy04.log)

shows EXPRESS, ADVACED, USE EXISTING SETUP

samsam4, 2021-06-27:
By the way: the problem with not finding Git and Python might be not from VSC extension, but from some of the ESP-IDF tools modules, as with Eclipse I also have similar issue, but there at least the first run goes Ok, then nexts fail unless add git and python in Windows path
https://github.com/espressif/idf-eclipse-plugin/issues/283

brianignacio5, 2021-06-28:
Needed to test in a clean Windows machine. Turns out git and python was not being added for processes like checking ESP-IDF version for given folders.

Please try this latest [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/70972541.zip)

samsam4, 2021-06-28:
@brianignacio5 
This one works (at least with USE EXISTING SETUP and tested only up to build stage so far) .
Not tested Advanced setup yet.
There are some errors in the .log, but on the VSC screens everything  looks Ok.  
Didn't  flash anything yet.

After .vsix (from 28-6-21) installed and Configure Extension: 
[esp_idf_vsc_ext11.log](https://github.com/espressif/vscode-esp-idf-extension/files/6729563/esp_idf_vsc_ext11.log)
[Doctor11.txt](https://github.com/espressif/vscode-esp-idf-extension/files/6729565/Doctor11.txt)

After ESP-IDF:Create project from etension template  ->TEMPLATE-APP
[esp_idf_vsc_ext12.log](https://github.com/espressif/vscode-esp-idf-extension/files/6729570/esp_idf_vsc_ext12.log)
[Doctor12.txt](https://github.com/espressif/vscode-esp-idf-extension/files/6729571/Doctor12.txt)

After build:
[esp_idf_vsc_ext13.log](https://github.com/espressif/vscode-esp-idf-extension/files/6729575/esp_idf_vsc_ext13.log)
[Doctor13.txt](https://github.com/espressif/vscode-esp-idf-extension/files/6729576/Doctor13.txt)

Thanks

PR #444 (2021-06-29): fix idf git python setup
Fix #434 
Fix #436
Fix #439 
Fix #447

Fixes for IDF Git and IDF Python setup.
closed 2021-06-29 by brianignacio5