espressif/vscode-esp-idf-extension #435: [Bug Report]: v1.1.0 broke Apple Silicon M1 chip Darwin-arm64 support

poky, 2021-06-15:
Newly updated to v1.1.0 on my Apple M1, whenever I'm trying to switch my versions between ESP-IDF Extension, this message pops up:

```
Command failed: /usr/local/bin/python /Users/lee/espidf/esp-idf_3.3/esp-idf/tools/idf_tools.py install-python-env
ERROR: Platform Darwin-arm64 appears to be unsupported
```

Have no problem if I rollback to previous working version which is v1.0.3

brianignacio5, 2021-06-21:
This is an issue within ESP-IDF itself, please take a look at [Darwin-arm64](https://github.com/espressif/esp-idf/issues/6113#issuecomment-756335935) which is fixed for ESP-IDF master branch but has yet to be back ported to previous releases.

In the first link there is an step by step guide to fix the issue, just manually add Darwin-arm64 as PLATFORM_MACOS as done in [here](https://github.com/espressif/esp-idf/commit/5739127412b6756801a38c700d068dd1d937ed41#diff-873391a19109de0fec9486d3ef26704a32fe81560a38f2e1f6c4f3db99d19350)

poky, 2021-06-23:
Hi, thank you for reply!
I have setup ESP-IDF env already, so I know it works fine on M1.
And have it worked on extension v1.0.3, but stop working after the latest v1.1.0

brianignacio5, 2021-06-24:
Can you try using `https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/69825436.zip` ?

poky, 2021-06-26:
> Can you try using this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/69825436.zip) ?
> 
> [Installing extension from vsix in Visual Studio Code](https://code.visualstudio.com/docs/editor/extension-marketplace#_install-from-a-vsix)

Hi, I've installed the vsix as provided, however, it still shows these message when installing the ESP-IDF tool

```
ERROR: Platform Darwin-arm64 appears to be unsupported
```

brianignacio5, 2021-06-29:
It seems you are trying to use an ESP-IDF version older than ESP-IDF master since this is a ESP-IDF error itself. Make you sure you are using the right path of ESP-IDF when configure the vscode extension.

KevinDaddy, 2021-07-01:
> It seems you are trying to use an ESP-IDF version older than ESP-IDF master since this is a ESP-IDF error itself. Make you sure you are using the right path of ESP-IDF when configure the vscode extension.

Same problem to me, on macbook m1 chip.  
Everything works fine if I downgrade the plugin to 1.0.3, with the same esp-idf 4.2 release installation.

brianignacio5, 2021-07-12:
Closing this issue since, as mentioned, it is an issue with ESP-IDF itself.

Please try the ESP-IDF master repository until the fix is available for older versions patches.

closed 2021-07-12 by brianignacio5