espressif/vscode-esp-idf-extension #436: [Bug Report]: First use of extension, configuration locked up during "Installing Python virtual environment" phase

jmporter34, 2021-06-15:
**Describe the bug**
A clear and concise description of what the bug is (current behavior).
During the first use of the extension, it locked up during the "Installing Python virtual environment" phase.  The circular progress bar kept spinning for at least an hour.

**To Reproduce**
I was installing software on a new system.  I first installed Git, Python 3.9, VS Code, ESP-IDF tools separately, then ESP-IDF extension for VS Code.  After I clicked on the extension icon, it presented a menu for configuring the extension: Simple, Advanced and Use Existing (it found both the ESP-IDF code library and tools that I had just installed).  I selected Use Existing and it began configuring things.  It reported successful completion of the "Installing IDF Prerequisites", "Installing ESP-IDF" and "Installing ESP-IDF Tools" phases (although it reported "Checksum: invalid" on each file, it still reported successful completion of the "Installing ESP-IDF Tools" phase).  Then, "Installing Python virtual environment for ESP-IDF" appeared.  It ran for a while, generating lines of output.  Eventually it stopped generating new lines of output, but the "Installing Python ..." message never disappeared and the circular progress bar never stopped spinning.  I let it run for at least an hour, and nothing ever changed.

**Expected behavior**
I expected the installation to report that it was finished, whether successful or not.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Environment (please complete the following information):**
- OS Name & Version: Windows 10 Enterprise 10.0.18363 Build 18363
- VSCode Version: 1.5.7
- ESP-IDF Version: 2.8 (offline installer), 1.1.0 (extension)
- Python Version: 3.9.5

> **NOTE**: You can use the `ESP-IDF: Doctor command` to generate a report of your configuration.
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.18363 
System environment variable PATH 
 C:\Program Files\Python39\Scripts\;C:\Program Files\Python39\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\Program Files\dotnet\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\170\Tools\Binn\;C:\Program Files\TortoiseGit\bin;C:\Program Files\WinMerge;C:\Program Files\Microsoft VS Code\bin;D:\Git\cmd;D:\Python\WPy64-3940\python-3.9.4.amd64;C:\Users\porterj02\AppData\Local\Microsoft\WindowsApps;C:\Users\porterj02\AppData\Local\Programs\Git\cmd 
Visual Studio Code version 1.57.0 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.1.0 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) ${env:IDF_PATH}
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
Custom extra paths (idf.customExtraPaths) 
Custom extra vars (idf.customExtraVars) 
Virtual env Python Path (idf.pythonBinPath) python
Serial port (idf.port) COM1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) ${env:IDF_TOOLS_PATH}
Git Path (idf.gitPath) git
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espIdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) false
Access to ESP-MDF Path (idf.espIdfPath) false
Access to ESP-IDF Custom extra paths
Access to Virtual env Python Path (idf.pythonBinPath) false
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) false
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version 2.32.0.windows.1
ESP-IDF version Not found
Python version 3.9.5
Python's pip version 21.1.1
-------------------------------------------------- Python packages in idf.pythonBinPath ----------------------------------------------------
pip version: 21.1.1
setuptools version: 56.0.0
---------------------------------------------------- Check ESP-IDF python requirements.txt -------------------------------------------------
Check ESP-IDF Python packages Error
---------------------------------------------------- Check extension requirements.txt ------------------------------------------------------
Check Extension Python packages Error
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages Error
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at Command failed: python ${env:IDF_PATH}\tools\check_python_dependencies.py -r "c:\Users\porterj02\.vscode\extensions\espressif.esp-idf-extension-1.1.0\esp_debug_adapter\requirements.txt"
python: can't open file 'c:\Users\porterj02\.vscode\extensions\espressif.esp-idf-extension-1.1.0\${env:IDF_PATH}\tools\check_python_dependencies.py': [Errno 22] Invalid argument

--------------------------------------------------------------------------------------------------------------------------------------------

**Logs**
[... 7 lines cut ...]
{"message":"spawn C:\\WINDOWS\\system32\\cmd.exe ENOENT","stack":"Error: spawn C:\\WINDOWS\\system32\\cmd.exe ENOENT\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:269:19)\n\tat onErrorNT (internal/child_process.js:465:16)\n\tat processTicksAndRejections (internal/process/task_queues.js:80:21)","level":"error","timestamp":"2021-06-15T10:26:01.260Z"}
{"message":"Open On-Chip Debugger  v0.10.0-esp32-20200709 (2020-07-09-08:54)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n","stack":"Error: Open On-Chip Debugger  v0.10.0-esp32-20200709 (2020-07-09-08:54)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n\n\tat c:\\Users\\porterj02\\.vscode\\extensions\\espressif.esp-idf-extension-1.1.0\\dist\\extension.js:1:6837\n\tat ChildProcess.exithandler (child_process.js:308:7)\n\tat ChildProcess.emit (events.js:315:20)\n\tat ChildProcess.EventEmitter.emit (domain.js:467:12)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-06-15T10:26:02.038Z"}
{"message":"WARNING: You are using pip version 20.3.3; however, version 21.1.2 is available.\r\nYou should consider upgrading via the 'C:\\Users\\porterj02\\.espressif\\python_env\\idf4.2_py3.8_env\\Scripts\\python.exe -m pip install --upgrade pip' command.\r\n","stack":"Error: WARNING: You are using pip version 20.3.3; however, version 21.1.2 is available.\r\nYou should consider upgrading via the 'C:\\Users\\porterj02\\.espressif\\python_env\\idf4.2_py3.8_env\\Scripts\\python.exe -m pip install --upgrade pip' command.\r\n\n\tat c:\\Users\\porterj02\\.vscode\\extensions\\espressif.esp-idf-extension-1.1.0\\dist\\extension.js:1:6837\n\tat ChildProcess.exithandler (child_process.js:308:7)\n\tat ChildProcess.emit (events.js:315:20)\n\tat ChildProcess.EventEmitter.emit (domain.js:467:12)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-06-15T10:26:05.262Z"}
{"message":"WARNING: You are using pip version 20.3.3; however, version 21.1.2 is available.\r\nYou should consider upgrading via the 'C:\\Users\\porterj02\\.espressif\\python_env\\idf4.2_py3.8_env\\Scripts\\python.exe -m pip install --upgrade pip' command.\r\n","stack":"Error: WARNING: You are using pip version 20.3.3; however, version 21.1.2 is available.\r\nYou should consider upgrading via the 'C:\\Users\\porterj02\\.espressif\\python_env\\idf4.2_py3.8_env\\Scripts\\python.exe -m pip install --upgrade pip' command.\r\n\n\tat c:\\Users\\porterj02\\.vscode\\extensio [... cut ...] nonymous>)\n\tat s (c:\\Users\\porterj02\\.vscode\\extensions\\espressif.esp-idf-extension-1.1.0\\dist\\extension.js:8:237105)\n\tat runMicrotasks (<anonymous>)\n\tat processTicksAndRejections (internal/process/task_queues.js:93:5)","level":"error","timestamp":"2021-06-15T11:37:07.408Z"}

**Extension Configuration settings**
There is no project open.  This is the entire contents of settings.JSON:
{
    "window.zoomLevel": -1
}

**Additional context**
Add any other context about the problem here.

brianignacio5, 2021-06-21:
Can you share the output of the Output `ESP-IDF` to see if there was any error while installing the python packages ?

brianignacio5, 2021-06-22:
Can you try this [installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/69475619.zip) ?

jmporter34, 2021-06-22:
Hi Brian,
I was installing the extension on my work computer with the help of an IT
person at the office, since my company does not allow users to install
software on their own.  I will be happy to try again with your new
installer the next time I am in the office.  But I'm not sure when that
will be, as I'm mostly still working from home for now.
I will update you when I have some more news to report.
Thanks
John Porter

closed 2021-06-29 by brianignacio5

KonssnoK, 2021-12-23:
having the same issue

brianignacio5, 2021-12-23:
Have you tried the linked vsix installer @KonssnoK  ?

KonssnoK, 2022-01-14:
sadly i solved but forgot how... probably i just went on, since the problem is just related to the tool not continuing, not to a real error ..

PR #444 (2021-06-29): fix idf git python setup
Fix #434 
Fix #436
Fix #439 
Fix #447

Fixes for IDF Git and IDF Python setup.