espressif/vscode-esp-idf-extension #439: [Bug Report]: ESP-IDF Setup stuck at EXPRESS Install

mrwappa, 2021-06-17:
OS: Windows 10 Pro

When going through the installer (ESP-IDF: Configure ESP-IDF extension), it gets stuck at this step:
[image]

Clicking the "Install" button does nothing.

I've followed the [instructions ](https://marketplace.visualstudio.com/items?itemName=espressif.esp-idf-extension#Prerequisites )and installed the Build Tools as stated in the Prerequisites:
[image]

What might be the problem?

brianignacio5, 2021-06-21:
Can you please provide more information about your environment to understand the issue better ?

**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [ ] Make sure you have searched for existing bugs and features request before you post an issue.
- [ ] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [ ] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
A clear and concise description of what the bug is (current behavior).

**To Reproduce**
Steps to reproduce the behavior (including the full log).

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**

- OS Name & Version: [e.g. MacOS X 10.14.5]
- VSCode Version: [e.g. 1.35.1]
- ESP-IDF Version: [e.g. v3.2.2]
- Python Version: [e.g. 2.7.10]

> **NOTE**: You can use the `ESP-IDF: Doctor command` to generate a report of your configuration.

**Logs**
If applicable, please share the log file which can be obtained from

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

mrwappa, 2021-06-21:
`Environment`
* Windows 10 Pro - Version 20H2
* Python version 3.9
* VSCode Version 1.57.1
* ESP-IDF Version 4.3

**Logs**
Here's the full `esp_idf_vsc_ext.log` log file on my system:
```
{"message":"non zero exit code 1\r\n\r\nINFO: Could not find files for the given pattern(s).\r\n","stack":"Error: non zero exit code 1\r\n\r\nINFO: Could not find files for the given pattern(s).\r\n\n\tat ChildProcess.<anonymous> (c:\\Users\\user\\.vscode\\extensions\\espressif.esp-idf-extension-1.1.0\\dist\\extension.js:1:4067)\n\tat ChildProcess.emit (events.js:315:20)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:277:12)","level":"error","timestamp":"2021-06-17T11:29:51.647Z"}
{"message":"Cannot access filePath: openocd","level":"error","timestamp":"2021-06-17T11:29:51.649Z"}
{"message":"Command failed: git describe --tags\nfatal: not a git repository (or any of the parent directories): .git\n","stack":"Error: Command failed: git describe --tags\nfatal: not a git repository (or any of the parent directories): .git\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-06-17T11:29:51.706Z"}
{"message":"Command failed: reg query HKEY_LOCAL_MACHINE\\SOFTWARE\\PYTHON\nERROR: The system was unable to find the specified registry key or value.\r\n","stack":"Error: Command failed: reg query HKEY_LOCAL_MACHINE\\SOFTWARE\\PYTHON\nERROR: The system was unable to find the specified registry key or value.\r\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-06-17T11:29:53.641Z"}
{"message":"Error looking for python in windows","stack":"Error: Command failed: reg query HKEY_LOCAL_MACHINE\\SOFTWARE\\PYTHON\nERROR: The system was unable to find the specified registry key or value.\r\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-06-17T11:29:53.642Z"}
{"message":"spawn C:\\WINDOWS\\system32\\cmd.exe ENOENT","stack":"Error: spawn C:\\WINDOWS\\system32\\cmd.exe ENOENT\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:269:19)\n\tat onErrorNT (internal/child_process.js:465:16)\n\tat processTicksAndRejections (internal/process/task_queues.js:80:21)","level":"error","timestamp":"2021-06-17T11:29:53.861Z"}
{"message":"Open On-Chip Debugger  v0.10.0-esp32-20210401 (2021-04-01-15:46)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n","stack":"Error: Open On-Chip Debugger  v0.10.0-esp32-20210401 (2021-04-01-15:46)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n\n\tat c:\\Users\\user\\.vscode\\extensions\\espressif.esp-idf-extension-1.1.0\\dist\\extension.js:1:6837\n\tat ChildProcess.exithandler (child_process.js:308:7)\n\tat ChildProcess.emit (events.js:315:20)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-06-17T11:29:54.691Z"}
{"message":"Command failed: \"Provide python executable path\" -c \"import sys; print('{}.{}'.format(sys.version_info.major, sys.version_info.minor))\"\n'\"Provide python executable path\"' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n","stack":"Error: Command failed: \"Provide python executable path\" -c \"import sys; print('{}.{}'.format(sys.version_info.major, sys.version_info.minor))\"\n'\"Provide python executable path\"' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-06-17T11:29:55.178Z"}
{"message":"Command failed: \"Provide python executable path\" -c \"import sys; print('{}.{}'.format(sys.version_info.major, sys.version_info.minor))\"\n'\"Provide python executable path\"' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n","stack":"Error: Command failed: \"Provide python executable path\" -c \"import sys; print('{}.{}'.format(sys.version_info.major, sys.version_info.minor))\"\n'\"Provide python executable path\"' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-06-17T11:29:55.179Z"}
{"message":"Command failed: reg query HKEY_LOCAL_MACHINE\\SOFTWARE\\PYTHON\nERROR: The system was unable to find the specified registry key or value.\r\n","stack":"Error: Command failed: reg query HKEY_LOCAL_MACHINE\\SOFTWARE\\PYTHON\n [... cut ...]
{"message":"Command failed: git describe --tags\nfatal: not a git repository (or any of the parent directories): .git\n","stack":"Error: Command failed: git describe --tags\nfatal: not a git repository (or any of the parent directories): .git\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-06-21T09:08:13.309Z"}

```

brianignacio5, 2021-06-22:
Can you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/69475619.zip)?

[Installing extension from vsix in Visual Studio Code](https://code.visualstudio.com/docs/editor/extension-marketplace#_install-from-a-vsix)

mrwappa, 2021-06-24:
Yes, I can install it with the vsix file, but I still get the same problem with the **ESP-IDF: Configure ESP-IDF extension**,

brianignacio5, 2021-06-24:
Did you delete the previous extension before installing this vsix ?

Which error are you getting ?

Sorry what do you mean with you can install it with the vsix file? Means `ESP-IDF: Configure ESP-IDF extension` works for this vsix file ?

oelison, 2021-06-26:
Had the same problem and the vsix installer helped here.

brianignacio5, 2021-06-29:
@mrwappa What about the latest [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/70982396.zip) ?

closed 2021-06-29 by brianignacio5

mrwappa, 2021-06-29:
The new version worked!
Thanks for the help :)

PR #444 (2021-06-29): fix idf git python setup
Fix #434 
Fix #436
Fix #439 
Fix #447

Fixes for IDF Git and IDF Python setup.