espressif/vscode-esp-idf-extension #441: [Bug Report]: Wont compile arduino as component

Olgidos, 2021-06-20:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [ x] Make sure you have searched for existing bugs and features request before you post an issue.
- [ x] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [ x] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
Wont compile arduino-as-component. 

**To Reproduce**
I am on version 1.0.3. pressed F1 "create project from extension template" -> arduino as component -> built

**Expected behavior**
A correct built

**Environment (please complete the following information):**

- OS Name & Version: [Win 10]
- VSCode Version: [1.57.1]
- ESP-IDF Version: [4.3]
- Python Version: [3.9]

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19042 
System environment variable PATH 
 C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v11.0\bin;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v11.0\libnvvp;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\Git\cmd;C:\Program Files\TortoiseGit\bin;C:\Program Files\TortoiseSVN\bin;C:\Program Files\PuTTY\;C:\Program Files\NVIDIA Corporation\Nsight Compute 2020.1.2\;C:\Program Files\MATLAB\R2020b\bin;C:\Users\TheOlg\Desktop\svn\somp\shaders\glslViewer-1.6.0-win-64bit;;C:\Program Files\Git LFS;C:\Users\TheOlg\AppData\Local\Microsoft\WindowsApps;C:\Users\TheOlg\AppData\Local\Programs\Microsoft VS Code\bin 
Visual Studio Code version 1.57.1 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.0.3 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-IDF Path (idf.espIdfPath) C:\Users\TheOlg\esp\esp-idf
Custom extra paths (idf.customExtraPaths) C:\Users\TheOlg\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin;C:\Users\TheOlg\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin;C:\Users\TheOlg\.espressif\tools\xtensa-esp32s3-elf\esp-2020r3-8.4.0\xtensa-esp32s3-elf\bin;C:\Users\TheOlg\.espressif\tools\riscv32-esp-elf\1.24.0.123_64eb9ff-8.4.0\riscv32-esp-elf\bin;C:\Users\TheOlg\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Users\TheOlg\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Users\TheOlg\.espressif\tools\cmake\3.16.4\bin;C:\Users\TheOlg\.espressif\tools\openocd-esp32\v0.10.0-esp32-20210401\openocd-esp32\bin;C:\Users\TheOlg\.espressif\tools\ninja\1.10.2;C:\Users\TheOlg\.espressif\tools\idf-exe\1.0.1;C:\Users\TheOlg\.espressif\tools\ccache\3.7;C:\Users\TheOlg\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64;C:\Users\TheOlg\.espressif\tools\idf-python-wheels\idf4.3_py3.9_2021-01-07
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"C:\\Users\\TheOlg\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20210401/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) C:\Users\TheOlg\.espressif\python_env\idf4.3_py3.9_env\Scripts\python.exe
Serial port (idf.port) /dev/ttyUSB1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) ${env:IDF_TOOLS_PATH}
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-IDF Custom extra paths
Access to C:\Users\TheOlg\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin: true
Access to C:\Users\TheOlg\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin: true
Access to C:\Users\TheOlg\.espressif\tools\xtensa-esp32s3-elf\esp-2020r3-8.4.0\xtensa-esp32s3-elf\bin: true
Access to C:\Users\TheOlg\.espressif\tools\riscv32-esp-elf\1.24.0.123_64eb9ff-8.4.0\riscv32-esp-elf\bin: true
Access to C:\Users\TheOlg\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin: true
Access to C:\Users\TheOlg\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin: true
Access to C:\Users\TheOlg\.espressif\tools\cmake\3.16.4\bin: true
Access to C:\Users\TheOlg\.espressif\tools\openocd-esp32\v0.10.0-esp32-20210401\openocd-esp32\bin: true
Access to C:\Users\TheOlg\.espressif\tools\ninja\1.10.2: true
Access to C:\Users\TheOlg\.espressif\tools\idf-exe\1.0.1: true
Access to C:\Users\TheOlg\.espressif\tools\ccache\3.7: true
Access to C:\Users\TheOlg\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to C:\Users\TheOlg\.espressif\tools\idf-python-wheels\idf4.3_py3.9_2021-01-07: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) false
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version 2.27.0.windows.1
ESP-IDF version 4.3
Python version 3.9.5
Python's pip version 21.1.1
-------------------------------------------------- Python packages in idf.pythonBinPath ----------------------------------------------------
bitstring version: 3.1.7
Brotli version: 1.0.9
certifi version: 2021.5.30
cffi version: 1.14.5
chardet version: 4.0.0
[... 388 lines cut ...]
../components/arduino/libraries/WiFi/src/WiFiGeneric.cpp: At global scope:
../components/arduino/libraries/WiFi/src/WiFiGeneric.cpp:188:21: warning: 'const char* auth_mode_str(int)' defined but not used [-Wunused-function]
 static const char * auth_mode_str(int authmode)
                     ^~~~~~~~~~~~~
[1069/1108] Building CXX object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/libraries/WiFi/src/WiFiScan.cpp.obj
../components/arduino/libraries/WiFi/src/WiFiScan.cpp:45:21: warning: 'const char* cipher_str(int)' defined but not used [-Wunused-function]
 static const char * cipher_str(int cipher)
                     ^~~~~~~~~~
[1070/1108] Building CXX object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/libraries/WiFi/src/WiFiSTA.cpp.obj
ninja: build stopped: subcommand failed.
The terminal process "C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command cmake --build ." terminated with exit code: 1.

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

**Extension Configuration settings**
Open Command Palette (F1) and type `Preferences: Open Settings (JSON)` or use `{CURRENT_PROJECT}/.vscode/settings.json` to get the following extension configuration settings value:

- `idf.espIdfPath` (MacOS or Linux) or `idf.espIdfPathWin` (Windows) also called `Path to locate ESP-IDF framework (IDF_PATH)` in `Preferences: Open Settings (UI)`.
- `idf.pythonBinPath` (MacOS or Linux) or `idf.pythonBinPathWin` (Windows) also called `Python absolute binary path used to execute ESP-IDF Python Scripts` in `Preferences: Open Settings (UI)`.
- `idf.openOcdConfigs` also called `List of configuration files inside OpenOCD Scripts directory` in `Preferences: Open Settings (UI)`.
- `idf.customExtraPaths` also called `Paths to be appended to PATH` in `Preferences: Open Settings (UI)`.
- `idf.customExtraVars` also called `Variables to be added to system environment variables` in `Preferences: Open Settings (UI)`.
- `idf.adapterTargetName` also called `Target name for ESP-IDF Debug Adapter` in `Preferences: Open Settings (UI)`.
- `idf.customAdapterTargetName` is used when `idf.adapterTargetName` is set to `custom`.

brianignacio5, 2021-06-21:
Hi @Olgidos I think you need to post this issue in the [Arduino-ESP32 repository](https://github.com/espressif/arduino-esp32) since this is outside the scope of this extension.

Olgidos, 2021-06-21:
Thanks, @brianignacio5, for your response! I will add an issue over there. Do you think a downgrade of the IDF could solve this issue? Did you test it with a specific version?
Sorry for close and reopen. My keyboard bugged, and I was clicking around out of frustration

brianignacio5, 2021-06-22:
I think you can use steps in [here](https://github.com/espressif/arduino-esp32/issues/1142#issuecomment-367867856) to check which ESP-IDF commit you can use without conflicts. I would recommend to use 4.0 or 4.2 though.

closed 2021-06-22 by brianignacio5