espressif/vscode-esp-idf-extension #447: [Bug Report]:

Tanishq-Banyal, 2021-06-26:
**Describe the bug :-**
I installed esp-idf using Windows installer from espressif website. It installed successfully and I am able to use esp-idf in commandline.

The tools exist in proper directories and I can manually run them but this ***extension*** isn't detecting them.
During setup of esp-idf extension, its saying `INVALID ESP-IDF` an throws out error saying `git not found in current environment`.

Then I also tried uninstalling esp-idf and letting the extension download esp-idf itself, still same error. And I even compared the checksum of the `esp-idf-v4.3.zip` downloaded by extension as well as by windows installer, its same, so my download is not currupt.

Please fix this extension. I have a limited internet and due to this extension and its countless failed downloads i already used up a lot of my data plan and my time.

**To Reproduce :-**
1. Install VSCode
2. Install C/C++ & ESP-IDF Extension
3. Install esp-idf at custom loctaion

**Expected behavior :-**
I expect it to detect the tools properly.

**Screenshots :-**
[image]

**Environment :-**

- OS Name & Version: Windows 10 Pro (version 20H2)
- VSCode Version: 1.57.1
- ESP-IDF Version: v4.3
- Python Version: 3.8.7 64bit (embedded version downloaded by esp-idf installer)(I also have python 3.9.4 32bit in path)
**esp_idf_vsc_ext.log :-**
```
{"message":"Command failed: G:\\.espressif\\tools\\idf-git\\cmd\\git.exe --version\nThe system cannot find the path specified.\r\n","stack":"Error: Command failed: G:\\.espressif\\tools\\idf-git\\cmd\\git.exe --version\nThe system cannot find the path specified.\r\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-06-26T04:17:38.843Z"}
{"user":true,"message":"Git is not found in current environment","stack":"Error: Command failed: G:\\.espressif\\tools\\idf-git\\cmd\\git.exe --version\nThe system cannot find the path specified.\r\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-06-26T04:17:38.844Z"}
{"user":true,"message":"Git is not found in current environment","stack":"Error: git is not found\n\tat Object.<anonymous> (c:\\Users\\Tanishq Banyal\\.vscode\\extensions\\espressif.esp-idf-extension-1.1.0\\dist\\extension.js:1:8860)\n\tat Generator.next (<anonymous>)\n\tat s (c:\\Users\\Tanishq Banyal\\.vscode\\extensions\\espressif.esp-idf-extension-1.1.0\\dist\\extension.js:1:2617)\n\tat processTicksAndRejections (internal/process/task_queues.js:93:5)","level":"error","timestamp":"2021-06-26T04:17:38.845Z"}
{"message":"Command failed: G:\\.espressif\\tools\\idf-git\\cmd\\git.exe --version\nThe system cannot find the path specified.\r\n","stack":"Error: Command failed: G:\\.espressif\\tools\\idf-git\\cmd\\git.exe --version\nThe system cannot find the path specified.\r\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-06-26T04:17:38.918Z"}
{"user":true,"message":"Git is not found in current environment","stack":"Error: Command failed: G:\\.espressif\\tools\\idf-git\\cmd\\git.exe --version\nThe system cannot find the path specified.\r\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-06-26T04:17:38.919Z"}
{"user":true,"message":"Git is not found in current environment","stack":"Error: git is not found\n\tat Object.<anonymous> (c:\\Users\\Tanishq Banyal\\.vscode\\extensions\\espressif.esp-idf-extension-1.1.0\\dist\\extension.js:1:8860)\n\tat Generator.next (<anonymous>)\n\tat s (c:\\Users\\Tanishq Banyal\\.vscode\\extensions\\espressif.esp-idf-extension-1.1.0\\dist\\extension.js:1:2617)\n\tat processTicksAndRejections (internal/process/task_queues.js:93:5)","level":"error","timestamp":"2021-06-26T04:17:38.919Z"}
{"message":"Command failed: reg query HKEY_CURRENT_USER\\SOFTWARE\\PYTHON\nERROR: The system was unable to find the specified registry key or value.\r\n","stack":"Error: Command failed: reg query HKEY_CURRENT_USER\\SOFTWARE\\PYTHON\nERROR: The system was unable to find the specified registry key or value.\r\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-06-26T04:17:40.101Z"}
{"message":"Error looking for python in windows","stack":"Error: Command failed: reg query HKEY_CURRENT_USER\\SOFTWARE\\PYTHON\nERROR: The system was unable to find the specified registry key or value.\r\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-06-26T04:17:40.102Z"}
{"message":"Command failed: reg query HKEY_LOCAL_MACHINE\\SOFTWARE\\PYTHON\nERROR: The system was unable to find the specified registry key or value.\r\n","stack":"Error: Command failed: reg query HKEY_LOCAL_MACHINE\\SOFTWARE\\PYTHON\nERROR: The system was unable to find the specified registry key or value.\r\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-06-26T04:17:40.139Z"}
{"message":"Error looking for python in windows","stack":"Error: Command failed: reg query HKEY_LOCAL_MACHINE\\SOFTWARE\\PYTHON\nERROR: The system was unable to find the specified registry key or value.\r\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-06-26T04:17:40.140Z"}
{"message":"Command failed: G:\\.espressif\\tools\\idf-git\\cmd\\git.exe --version\nThe system cannot find the path specified.\r\n","stack":"Error: Command failed: G:\\.espressif\\tools\\idf-git\\cmd\\git.exe --version\nThe system cannot find the path specified.\r\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-06-26T04:17:40.340Z"}
{"user":true,"message":"Git is not found in current environment","stack":"Error: Command failed: G:\\.espressif\\tools\\idf-git\\cmd\\git.exe --version\nThe system cannot find the path specified.\r\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-06-26T04:17:40.340Z"}
{"message":"Command failed: G:\\.espressif\\tools\\idf-git\\cmd\\git.exe --version\nThe system cannot find the path specified.\r\n","stack":"Error: Command failed: G:\\.espressif\\tools\\idf-git\\cmd\\git.exe --version\nThe system cannot find the path specified.\r\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-06-26T04:17:40.386Z"}
{"user":true,"message":"Git is not found in current environment","stack":"Error: Command failed: G:\\.espressif\\tools\\idf-git\\cmd\\git.exe --version\nThe system cannot find the path specified.\r\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js: [... cut ...] \\Users\\Tanishq Banyal\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20200709\\openocd-esp32\\bin;C:\\Users\\Tanishq Banyal\\.espressif\\tools\\ninja\\1.10.0;C:\\Users\\Tanishq Banyal\\.espressif\\tools\\idf-exe\\1.0.1;C:\\Users\\Tanishq Banyal\\.espressif\\tools\\ccache\\3.7;C:\\Users\\Tanishq Banyal\\.espressif\\tools\\dfu-util\\0.9\\dfu-util-0.9-win64",
    "idf.customExtraVars": "{}",
    "workbench.startupEditor": "newUntitledFile"
```

**Additional context :-**
#444 may fix this issue.

samsam4, 2021-06-28:
I have similar issue and the workarounds are either to add path to git and python in the Windows environment Path,
or install VSC extension from this .vsix file from here:
https://github.com/espressif/vscode-esp-idf-extension/issues/434#issuecomment-869739163

(Just make sure not to try some of the previous versions posted in the same thread as they don't work) :(

Tanishq-Banyal, 2021-06-29:
> I have similar issue and the workarounds are either to add path to git and python in the Windows environment Path

Latest Git and python are already in path, but in off logs it says git is incompatible so it's going to download again.

Btw merging #444 and building a vsix works too.

Except now my problem is that I am unable to change port, it's defaulting to COM1, and COM1 is not physically accessible on my PC :(

samsam4, 2021-06-29:
@Tanishq-Banyal 
This is the same .vsix I gave you the link before - just be sure you get the last one (from 28-6-21) as there were couple tries before that didn't fix the issue :(.
I used offline installer v2.8 with IDF 4.2.1 selected and with the latest .vsix now working and I dont need to pollute Windows Path with the git and python :)  Still didn't make thorough test, but at least this obstacle I think is fixed  :)
Regarding duplicate downloading git and python - you can refer to the same link I gave you before, but my [very first posting](https://github.com/espressif/vscode-esp-idf-extension/issues/434#issue-920420077): bugs1 to 4, but I think this issue was fixed with some of the first .vsix brianignacio5  posted  in the same thread (which are actually from fix idf git python setup #444)

brianignacio5, 2021-06-29:
To change ports you can use the `ESP-IDF: Select port to use` command or change `idf.portWin` in the Visual Studio Code User/Workspace/Workspace Folder configuration settings

Tanishq-Banyal, 2021-06-29:
> To change ports you can use the `ESP-IDF: Select port to use` command or change `idf.portWin` in the Visual Studio Code User/Workspace/Workspace Folder configuration settings

I have already tried both ways. Even though I can see port is COM3 in settings file, but it still uses COM1 (which is not physically accessible on my PC)

I suspect this happened due to some bug in #444 (I am using master branch of extension merged with #444 in vscode)

brianignacio5, 2021-06-29:
There is a configuration setting called `idf.saveScope` or `Idf: Save Scope` to choose where to save configuration settings and its default is User or Global settings. This is used by the extension to know where to save settings.

Maybe there is a setting in your `<project directory>/.vscode/settings.json` with `idf.portWin` that is overriding your User settings

Tanishq-Banyal, 2021-06-29:
> Maybe there is a setting in your `<project directory>/.vscode/settings.json` with `idf.portWin` that is overriding your User settings. Please take a look at [vscode settings precedence](https://code.visualstudio.com/docs/getstarted/settings#_settings-precedence)

Thanks, that worked, didnt knew that there were 2 settings.json (one global and one per project)

closed 2021-06-29 by brianignacio5

ideapad2009, 2021-07-29:
I have the same problem but I have solved. My finding is that esp-idf-tool can not be installed in folder whose name has space such as Program Files folder.

PR #444 (2021-06-29): fix idf git python setup
Fix #434 
Fix #436
Fix #439 
Fix #447

Fixes for IDF Git and IDF Python setup.