espressif/vscode-esp-idf-extension #449: "path" argument must be of type string. Received undefined.

igrr, 2021-06-29:
Okay, so I installed the esp-idf extension for vs-code and tried out their blinky program. I am now facing extension errors when I tried to build the application : 
[image]

I get this one whenever I try to configure the sdk options:
[image]

I'm using the latest espressif idf version V1.1.0

_Originally posted by @nalostta in https://github.com/espressif/esp32-camera/issues/289#issuecomment-870411571_

igrr, 2021-06-29:
@nalostta I've moved your issue into https://github.com/espressif/vscode-esp-idf-extension repository so that we can track it.

brianignacio5, 2021-06-29:
Did you configure the extension ?

Please fill this issue template:

**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [ ] Make sure you have searched for existing bugs and features request before you post an issue.
- [ ] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [ ] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
A clear and concise description of what the bug is (current behavior).

**To Reproduce**
Steps to reproduce the behavior (including the full log).

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**

- OS Name & Version: [e.g. MacOS X 10.14.5]
- VSCode Version: [e.g. 1.35.1]
- ESP-IDF Version: [e.g. v3.2.2]
- Python Version: [e.g. 2.7.10]

Use the `ESP-IDF: Doctor command` to generate a report of your configuration.

**Logs**
If applicable, please share the log file which can be obtained from

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

**Additional context**
Add any other context about the problem here.

nalostta, 2021-06-29:
Hello, 
PFA: config report.
[esp_idf_vscode_config_report.txt](https://github.com/espressif/vscode-esp-idf-extension/files/6732171/esp_idf_vscode_config_report.txt)

bug:
" The Path argument must be of type string. Received undefined. " when running the command 'SDK configuration editor'.

Steps to reproduce bug: 
> installed esp-idf vscode extension.
> select express installation, (latest esp-idf version v4.3, vscode extension version v1.1.0)
> after download and idf setup finishes, close vscode. Reopen it and start a new esp-idf project.
> go to command view and select esp sdk configuration command.

    I checked the esp installation folder and it is not empty. Upon reading the config report, i realize that the extension is not able to find the idf path. I tried to re-install it by selecting : "Find existing esp-idf on your system",  and then select install. There is no response from the extension for this. 

OS: Windows 10 pro 64-bit  (build: 19042.1052)
Python : python39 (3.9.1) => checked on powershell
ESP-IDF Version: [v4.3]
VSCode version : v1.57.1 (user setup)

[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/6732290/esp_idf_vsc_ext.log)

brianignacio5, 2021-06-30:
After your extension is configured (as shown in the latest config report), there is an Output channel (menu View -> Output) called `ESP-IDF: Menuconfig` where the sdkconfig conserver output is printed. Do you see any error there ?

You can try to delete the build folder before trying to run the ESP-IDF Menuconfig again to make sure no bad cache files are being used.

nalostta, 2021-07-01:
It is stuck at this :
[image]

brianignacio5, 2021-07-01:
Can you show the `SDK Configuration Editor` output ? In your screenshot, use the dropdown that says `Tasks` so we can the error you are getting.

The sdkconfig file is generated if not available when you build the project and default values are given by ESP-IDF itself. When the file is created though it can contain any project configuration you choose.

IF you want to start a project from start you might want to read [Example project structure](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#example-project) in ESP-IDF documentation. 

Basically any project has a that structure with a root CMakeLists.txt with

```
cmake_minimum_required(VERSION 3.5)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(myProject)
```
and a `main` directory (also known as main component) which is is the entry point of your application. All components have a CMakeLists.txt with CMake configuration for each. The documentation should give you more idea about what is possible to do.

nalostta, 2021-07-02:
So the issue is coming only when I try the command with the esp32-cam repository. I tested out the command with blinky from examples and it worked properly. 

The output ran commands and then quickly cleared the window so I was unable to get the out from there. However, I opened the esp-idf terminal and ran the command 'idf.py menuconfig', I got the following output:
[image]

brianignacio5, 2021-07-05:
So I'm closing this issue since it is not related to the extension. 

You might need to consider ESP32-cam works with specific version of ESP-IDF and might not work with 4.3 (your current version)

closed 2021-07-05 by brianignacio5

nalostta, 2021-07-05:
Hello, I found out the cause for the issue, the esp32 cam was supposed to be used as a component. I was trying to build it as the main application since it had an example code for the same. 

Because of the above, the CMakeLists.txt was in the correct format causing the extension to pause whenever I tried to run the SDK config. While my usage was improper, I fell it would've be nice if the extension could've sent an error instead being stuck at loading.