espressif/vscode-esp-idf-extension #450: esp-idf-extension  failed to check command version from Version 0.6

JerryZheng89, 2021-07-01:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [ ] Make sure you have searched for existing bugs and features request before you post an issue.
- [ ] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [ ] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
A clear and concise description of what the bug is (current behavior).

**To Reproduce**
Steps to reproduce the behavior (including the full log).

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**

- OS Name & Version: [e.g. MacOS X 10.14.5]
- VSCode Version: [e.g. 1.35.1]
- ESP-IDF Version: [e.g. v3.2.2]
- Python Version: [e.g. 2.7.10]

> **NOTE**: You can use the `ESP-IDF: Doctor command` to generate a report of your configuration.

**Logs**
If applicable, please share the log file which can be obtained from

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

**Extension Configuration settings**
Open Command Palette (F1) and type `Preferences: Open Settings (JSON)` or use `{CURRENT_PROJECT}/.vscode/settings.json` to get the following extension configuration settings value:

- `idf.espIdfPath` (MacOS or Linux) or `idf.espIdfPathWin` (Windows) also called `Path to locate ESP-IDF framework (IDF_PATH)` in `Preferences: Open Settings (UI)`.
- `idf.pythonBinPath` (MacOS or Linux) or `idf.pythonBinPathWin` (Windows) also called `Python absolute binary path used to execute ESP-IDF Python Scripts` in `Preferences: Open Settings (UI)`.
- `idf.openOcdConfigs` also called `List of configuration files inside OpenOCD Scripts directory` in `Preferences: Open Settings (UI)`.
- `idf.customExtraPaths` also called `Paths to be appended to PATH` in `Preferences: Open Settings (UI)`.
- `idf.customExtraVars` also called `Variables to be added to system environment variables` in `Preferences: Open Settings (UI)`.
- `idf.adapterTargetName` also called `Target name for ESP-IDF Debug Adapter` in `Preferences: Open Settings (UI)`.
- `idf.customAdapterTargetName` is used when `idf.adapterTargetName` is set to `custom`.

**Additional context**
Add any other context about the problem here.

brianignacio5, 2021-07-01:
Can you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/71213648.zip) ?

JerryZheng89, 2021-07-02:
[](url)

> Can you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/71213648.zip) ?
> 
> This happens when the extension is not configured properly. The `ESP-IDF: Doctor command` will show that configurations settings values and if they are accessible.

I try the vsix installer, but problem still there.  here is my log file, which is doctorCommand output!  [doctorCommand.log](https://ynote-resource-dl.lx.netease.com/ydl/resource?key=woshizhengjie89%40163.com%2F455F91EBC42A430D941AA81859283904&mime=application%2Foctet-stream&name=doctorCommand.log&reqId=919a457a-ae6d-44e2-8ab4-b70b0423176e&ts=1625204737815&type=NOS&sign=8dbf235b23a273bd673f0bcb5aa93416)

brianignacio5, 2021-07-06:
IS it possible you have `.vscode/settings.json` in your currently opened directory overriding global ones ?

brianignacio5, 2021-07-12:
Closing this issue due to lack of response.

closed 2021-07-12 by brianignacio5

arfaali-naikar, 2025-02-16:
Hi everyone,

I am trying to configure ESP-IDF using the Express Install method in Visual Studio Code with the ESP-IDF extension. However, I am facing the following issues during the setup:

    I get the error:

        The path for ESP-IDF is not valid: /tools/idf.py not found.

    Additionally, I receive the warning:

        Unable to write into user settings. Please open the user settings to correct errors/warnings in it and try again.

My current configuration:

    VSCode
    ESP-IDF Extension 
    ESP-IDF Version: v5.4 (release version)
    OS:  Windows 11

Questions:

    How can I resolve the "/tools/idf.py not found" error? Do I need to manually set a different path or reinstall the ESP-IDF?
    How do I fix the “Unable to write into user settings” error? Is there a specific permission or setting I need to check?

Any guidance or suggestions would be greatly appreciated!

Thank you!

[image]

brianignacio5, 2025-02-17:
Hi @arfaali-naikar It seems that the extension is unable to write settings from the ESP-IDF setup you just installed in vscode user settings. Do you have user permission to modify this file `APPDATA%\Code\User\settings.json` ?

arfaali-naikar, 2025-02-17:
@brianignacio5 Thank you for your support! After reinstalling Visual Studio Code and ensuring the proper permissions for settings.json, the issue was resolved. The ESP-IDF extension is now working perfectly.