espressif/vscode-esp-idf-extension #453: [Bug Report]: "ESP-IDF: New Project" wizard, show blank template screen without having an open folder.

zfields, 2021-07-02:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [X] Make sure you have searched for existing bugs and features request before you post an issue.
- [X] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [X] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
Cannot use "ESP-IDF: New Project" wizard, without having a folder open.

**To Reproduce**
1. In VSCode, select "File > Close Folder [Ctrl+K F]" from the menu
2. Press F1
3. Select "ESP-IDF: New Project"
4. Fill in the template information.
5. Select "Choose Template".

**Expected behavior**
The list of templates would appear as intended.

**Screenshots**
[image]

**Environment (please complete the following information):**

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS linux x64 5.11.0-7620-generic 
System environment variable PATH 
 /home/zak/.local/bin:/home/zak/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin 
Visual Studio Code version 1.57.1 
Visual Studio Code language en 
Visual Studio Code shell bash 
ESP-IDF Extension version 1.1.0 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) ${env:IDF_PATH}
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
Custom extra paths (idf.customExtraPaths) 
Custom extra vars (idf.customExtraVars) 
Virtual env Python Path (idf.pythonBinPath) python
Serial port (idf.port) /dev/ttyUSB1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) ${env:IDF_TOOLS_PATH}
Git Path (idf.gitPath) git
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espIdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) false
Access to ESP-MDF Path (idf.espIdfPath) false
Access to ESP-IDF Custom extra paths
Access to Virtual env Python Path (idf.pythonBinPath) false
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) false
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version 2.27.0
ESP-IDF version Not found
Python version 3.8.10
Python's pip version 20.1.1
-------------------------------------------------- Python packages in idf.pythonBinPath ----------------------------------------------------
adafruit-nrfutil version: 0.5.3.post13
appdirs version: 1.4.4
blinker version: 1.4
Brlapi version: 0.7.0
certifi version: 2020.4.5.1
chardet version: 3.0.4
chrome-gnome-shell version: 0.0.0
click version: 7.1.2
colorama version: 0.4.3
command-not-found version: 0.3
cryptography version: 3.0
cupshelpers version: 1.0
dbus-python version: 1.2.16
defer version: 1.0.6
distlib version: 0.3.2
distro version: 1.5.0
distro-info version: 0.23ubuntu1
ecdsa version: 0.16.1
evdev version: 1.3.0
filelock version: 3.0.12
gpg version: 1.14.0-unknown
gyp version: 0.1
hidpidaemon version: 18.4.6
httplib2 version: 0.18.1
idna version: 2.10
jeepney version: 0.4.3
kernelstub version: 3.1.4
keyring version: 21.3.0
language-selector version: 0.1
launchpadlib version: 1.10.13
lazr.restfulclient version: 0.14.2
lazr.uri version: 1.0.5
louis version: 3.14.0
macaroonbakery version: 1.3.1
netifaces version: 0.10.4
oauthlib version: 3.1.0
pip version: 20.1.1
pop-transition version: 1.1.2
protobuf version: 3.12.3
pycairo version: 1.16.2
pycups version: 2.0.1
pydbus version: 0.6.0
Pygments version: 2.3.1
PyGObject version: 3.38.0
PyJWT version: 1.7.1
pymacaroons version: 0.13.0
PyNaCl version: 1.4.0
pyRFC3339 version: 1.1
pyserial version: 3.4
python-apt version: 2.1.2pop0-1587756471-20.04-cd2988e
python-debian version: 0.1.37
python-xlib version: 0.27
pytz version: 2020.1
pyxdg version: 0.26
PyYAML version: 5.3.1
repolib version: 1.5.2
repoman version: 1.4.0
requests version: 2.23.0
requests-unixsocket version: 0.2.0
screen-resolution-extra version: 0.0.0
SecretStorage version: 3.1.2
sessioninstaller version: 0.0.0
setuptools version: 49.3.1
simplejson version: 3.17.0
six version: 1.15.0
system76driver version: 20.4.36
systemd-python version: 234
ubuntu-advantage-tools version: 27.0
ubuntu-drivers-common version: 0.0.0
ufw version: 0.36
urllib3 version: 1.25.9
virtualenv version: 20.4.7
wadllib version: 1.3.4
wheel version: 0.34.2
xkit version: 0.0.0
---------------------------------------------------- Check ESP-IDF python requirements.txt -------------------------------------------------
Check ESP-IDF Python packages Error
---------------------------------------------------- Check extension requirements.txt ------------------------------------------------------
Check Extension Python packages Error
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages Error
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at Command failed: python ${env:IDF_PATH}/tools/check_python_dependencies.py -r "/home/zak/.vscode/extensions/espressif.esp-idf-extension-1.1.0/esp_debug_adapter/requirements.txt"
/bin/sh: 1: Bad substitution

--------------------------------------------------------------------------------------------------------------------------------------------

> **NOTE**: You can use the `ESP-IDF: Doctor command` to generate a report of your configuration.

edouardreg, 2021-07-02:
Hello,

SAme problem for me.

zfields, 2021-07-02:
The workaround, is to have another (any) folder open when starting a new project, then the templates appear.

edouardreg, 2021-07-02:
Thanks !

Will try soon !

edouardreg, 2021-07-02:
So : not working for me. And then try to make a clean install of all : no more possible after several tries...

Command failed: C:\Users\edoua\AppData\Local\Programs\Python\Python39\python.exe d:\ESP-IDF\esp-idf\tools\idf_tools.py install-python-env
  DEPRECATION: A future pip version will change local packages to be built in-place without first copying to a temporary directory. We recommend you use --use-feature=in-tree-build to test your packages with this new behavior before it becomes the default.
   pip 21.3 will remove support for this functionality. You can find discussion regarding this at https://github.com/pypa/pip/issues/7555.
ERROR: Exception:
Traceback (most recent call last):
  File "C:\Users\edoua\.espressif\python_env\idf4.4_py3.9_env\lib\site-packages\pip\_internal\cli\base_command.py", line 180, in _main
    status = self.run(options, args)
  File "C:\Users\edoua\.espressif\python_env\idf4.4_py3.9_env\lib\site-packages\pip\_internal\cli\req_command.py", line 204, in wrapper
    return func(self, options, args)
  File "C:\Users\edoua\.espressif\python_env\idf4.4_py3.9_env\lib\site-packages\pip\_internal\commands\install.py", line 318, in run
    requirement_set = resolver.resolve(
  File "C:\Users\edoua\.espressif\python_env\idf4.4_py3.9_env\lib\site-packages\pip\_internal\resolution\resolvelib\resolver.py", line 127, in resolve
    result = self._result = resolver.resolve(
  File "C:\Users\edoua\.espressif\python_env\idf4.4_py3.9_env\lib\site-packages\pip\_vendor\resolvelib\resolvers.py", line 473, in resolve
    state = resolution.resolve(requirements, max_rounds=max_rounds)
  File "C:\Users\edoua\.espressif\python_env\idf4.4_py3.9_env\lib\site-packages\pip\_vendor\resolvelib\resolvers.py", line 341, in resolve
    name, crit = self._merge_into_criterion(r, parent=None)
  File "C:\Users\edoua\.espressif\python_env\idf4.4_py3.9_env\lib\site-packages\pip\_vendor\resolvelib\resolvers.py", line 172, in _merge_into_criterion
    if not criterion.candidates:
  File "C:\Users\edoua\.espressif\python_env\idf4.4_py3.9_env\lib\site-packages\pip\_vendor\resolvelib\structs.py", line 139, in __bool__
    return bool(self._sequence)
  File "C:\Users\edoua\.espressif\python_env\idf4.4_py3.9_env\lib\site-packages\pip\_internal\resolution\resolvelib\found_candidates.py", line 143, in __bool__
    return any(self)
  File "C:\Users\edoua\.espressif\python_env\idf4.4_py3.9_env\lib\site-packages\pip\_internal\resolution\resolvelib\found_candidates.py", line 129, in <genexpr>
    return (c for c in iterator if id(c) not in self._incompatible_ids)
  File "C:\Users\edoua\.espressif\python_env\idf4.4_py3.9_env\lib\site-packages\pip\_internal\resolution\resolvelib\found_candidates.py", line 30, in _iter_built
    for version, func in infos:
  File "C:\Users\edoua\.espressif\python_env\idf4.4_py3.9_env\lib\site-packages\pip\_internal\resolution\resolvelib\factory.py", line 269, in iter_index_candidate_infos
    result = self._finder.find_best_candidate(
  File "C:\Users\edoua\.espressif\python_env\idf4.4_py3.9_env\lib\site-packages\pip\_internal\index\package_finder.py", line 879, in find_best_candidate
    candidates = self.find_all_candidates(project_name)
  File "C:\Users\edoua\.espressif\python_env\idf4.4_py3.9_env\lib\site-packages\pip\_internal\index\package_finder.py", line 824, in find_all_candidates
    page_candidates = list(page_candidates_it)
  File "C:\Users\edoua\.espressif\python_env\idf4.4_py3.9_env\lib\site-packages\pip\_internal\index\sources.py", line 134, in page_candidates
    yield from self._candidates_from_page(self._link)
  File "C:\Users\edoua\.espressif\python_env\idf4.4_py3.9_env\lib\site-packages\pip\_internal\index\package_finder.py", line 783, in process_project_url
    html_page = self._link_collector.fetch_page(project_url)
  File "C:\Users\edoua\.espressif\python_env\idf4.4_py3.9_env\lib\site-packages\pip\_internal\index\collector.py", line 512, in fetch_page
    return _get_html_page(location, session=self.session)
  File "C:\Users\edoua\.espressif\python_env\idf4.4_py3.9_env\lib\site-packages\pip\_internal\index\collector.py", line 422, in _get_html_page
    resp = _get_html_response(url, session=session)
  File "C:\Users\edoua\.espressif\python_env\idf4.4_py3.9_env\lib\site-packages\pip\_internal\index\collector.py", line 120, in _get_html_response
    resp = session.get(
  File "C:\Users\edoua\.espressif\python_env\idf4.4_py3.9_env\lib\site-packages\pip\_vendor\requests\sessions.py", line 555, in get
    return self.request('GET', url, **kwargs)
  File "C:\Users\edoua\.espressif\python_env\idf4.4_py3.9_env\lib\site-packages\pip\_internal\network\session.py", line 449, in request
    return super().request(method, url, *args, **kwargs)
  File "C:\Users\edoua\.espressif\python_env\idf4.4_py3.9_env\lib\site-packages\pip\_vendor\requests\sessions.py", line 542, in request
    resp = self.send(prep, **send_kwargs)
  File "C:\Users\edoua\.espressif\python_env\idf4.4_py3.9_env\lib\site-packages\pip\_vendor\requests\sessions.py", line 655, in send
    r = adapter.send(request, **kwargs)
  File "C:\Users\edoua\.espressif\python_env\idf4.4_py3.9_env\lib\site-packages\pip\_vendor\cachecontrol\adapter.py", line 53, in send
    resp = super(CacheControlAdapter, self).send(request, **kw)
  File "C:\Users\edoua\.espressif\python_env\idf4.4_py3.9_env\lib\site-packages\pip\_vendor\requests\adapters.py", line 439, in send
    resp = conn.urlopen(
[... cut ...]
    subprocess.check_call(run_args, stdout=sys.stdout, stderr=sys.stderr)
  File "C:\Users\edoua\AppData\Local\Programs\Python\Python39\lib\subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['C:\\Users\\edoua\\.espressif\\python_env\\idf4.4_py3.9_env\\Scripts\\python.exe', '-m', 'pip', 'install', '--no-warn-script-location', '-r', 'D:\\ESP-IDF\\esp-idf\\requirements.txt', '--extra-index-url', 'https://dl.espressif.com/pypi']' returned non-zero exit status 2.

No more possible to do anything...

brianignacio5, 2021-07-05:
I don't think the issue is related to having a folder open but your extension not being configured properly.

Have you tested the current master? We have fixed many setup issues in v1.1.0 [vsix installer link](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/71213648.zip)

zfields, 2021-07-06:
> `ESP-IDF Extension version 1.1.0.`

I have a fresh install, and I've only used plug-ins distributed by VSCode. I went through the wizard and let the ESP-IDF extension download dependencies and configure itself. Nothing strange.

The doctor output (above) mentions it doesn't know how to find my ESP-IDF version, which is strange since it installed it.

```
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version 2.27.0
ESP-IDF version Not found
Python version 3.8.10
Python's pip version 20.1.1
```

edouardreg, 2021-07-07:
Thanks Brianignacio : fresh install and ok now.

brianignacio5, 2021-07-08:
@zfields based on your doctor command output it seems that the extension settings are not configured properly.

We have fixed many issues in extension setup but not available in the vscode marketplace yet. Can you uninstall your current extension and try installing from this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/71213648.zip) ?

PR #461 (2021-07-13): add extension templates in new project
Fix #453 

Add extension templates in new project wizard.
closed 2021-07-13 by brianignacio5