espressif/vscode-esp-idf-extension #454: [Bug Report]: Launching "ESP-IDF: Configure project sdkconfig for coverage" breaks build

zfields, 2021-07-02:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [X] Make sure you have searched for existing bugs and features request before you post an issue.
- [X] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [X] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
Launching "ESP-IDF: Configure project sdkconfig for coverage", then immediately exiting the new tab breaks the build. A new `sdkconfig` file is generated, the previous `sdkconfig` is saved as `sdkconfig.old`, and the build will error.

The build can be fix by deleting `sdkconfig`, and renaming `sdkconfig.old` to `sdkconfig`.

**To Reproduce**
1. Create a new project using the `blink` template.
2. Press F1
3. Select, "ESP-IDF: Build project"
4. Confirm successful build
5. Press F1
6. Select, "ESP-IDF: Configure project sdkconfig for coverage"
7. Immediately close the newly opened tab.
8. Press F1
9. Select, "ESP-IDF: Build project"
10. Build fails

**Expected behavior**
The project to build successfully.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS linux x64 5.11.0-7620-generic 
System environment variable PATH 
 /home/zak/.local/bin:/home/zak/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin 
Visual Studio Code version 1.57.1 
Visual Studio Code language en 
Visual Studio Code shell bash 
ESP-IDF Extension version 1.1.0 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) /home/zak/Development/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
Custom extra paths (idf.customExtraPaths) /home/zak/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin:/home/zak/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin:/home/zak/.espressif/tools/xtensa-esp32s3-elf/esp-2020r3-8.4.0/xtensa-esp32s3-elf/bin:/home/zak/.espressif/tools/riscv32-esp-elf/1.24.0.123_64eb9ff-8.4.0/riscv32-esp-elf/bin:/home/zak/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/home/zak/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/home/zak/.espressif/tools/openocd-esp32/v0.10.0-esp32-20210401/openocd-esp32/bin
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"/home/zak/.espressif/tools/openocd-esp32/v0.10.0-esp32-20210401/openocd-esp32/share/openocd/scripts"}
Virtual env Python Path (idf.pythonBinPath) /home/zak/.espressif/python_env/idf4.3_py3.8_env/bin/python
Serial port (idf.port) /dev/ttyUSB0
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) ${env:IDF_TOOLS_PATH}
Git Path (idf.gitPath) git
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espIdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espIdfPath) false
Access to ESP-IDF Custom extra paths
Access to /home/zak/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin: true
Access to /home/zak/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin: true
Access to /home/zak/.espressif/tools/xtensa-esp32s3-elf/esp-2020r3-8.4.0/xtensa-esp32s3-elf/bin: true
Access to /home/zak/.espressif/tools/riscv32-esp-elf/1.24.0.123_64eb9ff-8.4.0/riscv32-esp-elf/bin: true
Access to /home/zak/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin: true
Access to /home/zak/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin: true
Access to /home/zak/.espressif/tools/openocd-esp32/v0.10.0-esp32-20210401/openocd-esp32/bin: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) false
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version 2.27.0
ESP-IDF version 4.3
[... 207 lines cut ...]
{"user":true,"message":"Something went wrong while trying to build the project","stack":"Error: Task ESP-IDF Build exited with code 1\n\tat Function.<anonymous> (/home/zak/.vscode/extensions/espressif.esp-idf-extension-1.1.0/dist/extension.js:1:66256)\n\tat Generator.next (<anonymous>)\n\tat /home/zak/.vscode/extensions/espressif.esp-idf-extension-1.1.0/dist/extension.js:1:65362\n\tat new Promise (<anonymous>)\n\tat o (/home/zak/.vscode/extensions/espressif.esp-idf-extension-1.1.0/dist/extension.js:1:65107)\n\tat /home/zak/.vscode/extensions/espressif.esp-idf-extension-1.1.0/dist/extension.js:1:66099\n\tat l.fire (/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:57:1835)\n\tat w.$onDidEndTaskProcess (/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:88:50925)\n\tat runMicrotasks (<anonymous>)\n\tat processTicksAndRejections (internal/process/task_queues.js:93:5)","level":"error","timestamp":"2021-07-02T12:39:08.999Z"}
{"message":"Open On-Chip Debugger  v0.10.0-esp32-20210401 (2021-04-01-15:45)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n","stack":"Error: Open On-Chip Debugger  v0.10.0-esp32-20210401 (2021-04-01-15:45)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n\n\tat /home/zak/.vscode/extensions/espressif.esp-idf-extension-1.1.0/dist/extension.js:1:6837\n\tat ChildProcess.exithandler (child_process.js:308:7)\n\tat ChildProcess.emit (events.js:315:20)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Socket.<anonymous> (internal/child_process.js:439:11)\n\tat Socket.emit (events.js:315:20)\n\tat Pipe.<anonymous> (net.js:673:12)","level":"error","timestamp":"2021-07-02T12:40:54.540Z"}
{"user":true,"message":"Something went wrong while trying to build the project","stack":"Error: Task ESP-IDF Build exited with code 1\n\tat Function.<anonymous> (/home/zak/.vscode/extensions/espressif.esp-idf-extension-1.1.0/dist/extension.js:1:66256)\n\tat Generator.next (<anonymous>)\n\tat /home/zak/.vscode/extensions/espressif.esp-idf-extension-1.1.0/dist/extension.js:1:65362\n\tat new Promise (<anonymous>)\n\tat o (/home/zak/.vscode/extensions/espressif.esp-idf-extension-1.1.0/dist/extension.js:1:65107)\n\tat /home/zak/.vscode/extensions/espressif.esp-idf-extension-1.1.0/dist/extension.js:1:66099\n\tat l.fire (/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:57:1835)\n\tat w.$onDidEndTaskProcess (/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:88:50925)\n\tat runMicrotasks (<anonymous>)\n\tat processTicksAndRejections (internal/process/task_queues.js:93:5)","level":"error","timestamp":"2021-07-02T12:42:38.846Z"}
{"message":"Cannot access filePath: ${env:IDF_TOOLS_PATH}","stack":"Error: ENOENT: no such file or directory, access '${env:IDF_TOOLS_PATH}'\n\tat Object.accessSync (fs.js:220:3)\n\tat Object.e.accessSync (electron/js2c/asar_bundle.js:5:7099)\n\tat I (/home/zak/.vscode/extensions/espressif.esp-idf-extension-1.1.0/dist/extension.js:1:4226)\n\tat /home/zak/.vscode/extensions/espressif.esp-idf-extension-1.1.0/dist/extension.js:8:238580\n\tat Generator.next (<anonymous>)\n\tat /home/zak/.vscode/extensions/espressif.esp-idf-extension-1.1.0/dist/extension.js:8:233432\n\tat new Promise (<anonymous>)\n\tat l (/home/zak/.vscode/extensions/espressif.esp-idf-extension-1.1.0/dist/extension.js:8:233177)\n\tat /home/zak/.vscode/extensions/espressif.esp-idf-extension-1.1.0/dist/extension.js:8:238500\n\tat /home/zak/.vscode/extensions/espressif.esp-idf-extension-1.1.0/dist/extension.js:8:239580\n\tat Generator.next (<anonymous>)\n\tat s (/home/zak/.vscode/extensions/espressif.esp-idf-extension-1.1.0/dist/extension.js:8:237105)\n\tat runMicrotasks (<anonymous>)\n\tat pro [... cut ...] inutils/bin:/home/zak/.espressif/tools/openocd-esp32/v0.10.0-esp32-20210401/openocd-esp32/bin",
    "idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"/home/zak/.espressif/tools/openocd-esp32/v0.10.0-esp32-20210401/openocd-esp32/share/openocd/scripts\"}",
    "idf.espIdfPath": "/home/zak/Development/esp-idf",
    "idf.port": "/dev/ttyUSB0",
    "idf.pythonBinPath": "/home/zak/.espressif/python_env/idf4.3_py3.8_env/bin/python"
}
```
**Additional context**
Add any other context about the problem here.

brianignacio5, 2021-07-05:
Please consider what the `ESP-IDF: Configure project sdkconfig for coverage` command is doing to your project configuration. The command is updating this in your sdkconfig:

```json
{"version": 2, "set": { "APPTRACE_DEST_TRAX": true, "APPTRACE_GCOV_ENABLE": true }}
```

This might change other settings in your project due to dependencies of settings defined in ESP-IDF components Kconfig.

What is the build output, maybe enabling Coverage breaks something you are doing in your project.

Documentation about [Coverage](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/COVERAGE.md) and [ESP-IDF Coverage](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/app_trace.html#gcov-source-code-coverage)

zfields, 2021-07-06:
Please read the report carefully. It contains the logs you requested.

Again, I have a fresh install and the project is the unmodified "Blink" template project. It compiles successfully, before `sdkconfig` is modified by clicking on,"ESP-IDF: Configure project sdkconfig for coverage", and cancelling out.

zfields, 2021-07-09:
For the record, this is a new ESP-IDF install, I'm compiling and unmodified Blink template.
When I accidently selected that option and then closed it without saving, it modified `sdkconfig` in such a way that it would no longer compile.

I'm trying to report that it's breaking something in YOUR project. I expect that it is very easy to reproduce, and I am hopeful you will investigate.

brianignacio5, 2021-07-12:
I've reproduce your steps and it is compiling fine after running `ESP-IDF: Configure project sdkconfig for coverage` for ESP-IDF v4.2, 4.3 and master branch.

**What is your build output?**

brianignacio5, 2021-07-19:
Closing this for lack of response.

Please comment here if you still face this issue.

closed 2021-07-19 by brianignacio5