espressif/vscode-esp-idf-extension #456: [Bug Report]: ESP-IDF: Menuconfig: Loading default values windows hung

xiaolongba, 2021-07-04:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [x] Make sure you have searched for existing bugs and features request before you post an issue.
- [x] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [x] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
the Menuconfig information pop up windows hung when I tried to reset the menuconfig parameters, the details as below:
[image]

**To Reproduce**
1. click **ESP IDF Launch GUI Configuration tool** button

  [image]

2.  click **Reset** button

  [image]

**Expected behavior**
 the pop up windows can disappear and goes back to menuconfig setting windows

**Screenshots**

I have mentioned above

**Environment (please complete the following information):**

- OS Name & Version: win10
- VSCode Version: 1.57.1
- ESP-IDF Version: v1.1.0
- Python Version: 3.9.0

> **NOTE**: You can use the `ESP-IDF: Doctor command` to generate a report of your configuration.

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19042 
System environment variable PATH 
 D:\Python39\Scripts\;D:\Python39\;%ALLEGRO_BRD2ODB%\nv\bin;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\windows\system32;C:\windows;C:\windows\System32\Wbem;C:\windows\System32\WindowsPowerShell\v1.0\;C:\windows\System32\OpenSSH\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;c:\Program Files (x86)\HP\HP Performance Advisor;D:\PuTTY\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin\;C:\Program Files\IVI Foundation\VISA\Win64\Bin\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin;D:\Git\cmd;C:\Program Files (x86)\STMicroelectronics\STM32 ST-LINK Utility\ST-LINK Utility;C:\Program Files (x86)\IVI Foundation\IVI\bin;C:\Program Files\IVI Foundation\IVI\bin;D:\CMake\bin;C:\Program Files\Nordic Semiconductor\nrf-command-line-tools\bin\;D:\doxygen\bin;D:\Mscgen;D:\stlink-1.3.0-win64\bin;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Users\wireless_tech\AppData\Local\Microsoft\WindowsApps;D:\Microsoft VS Code\bin 
Visual Studio Code version 1.57.1 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.1.0 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) F:\BLE_WIFI\Espressif\SDK\esp-idf-c3
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
Custom extra paths (idf.customExtraPaths) F:\Tools\Espressif\IDF_Tools\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin;F:\Tools\Espressif\IDF_Tools\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin;F:\Tools\Espressif\IDF_Tools\.espressif\tools\xtensa-esp32s3-elf\esp-2020r3-8.4.0\xtensa-esp32s3-elf\bin;F:\Tools\Espressif\IDF_Tools\.espressif\tools\riscv32-esp-elf\1.24.0.123_64eb9ff-8.4.0\riscv32-esp-elf\bin;F:\Tools\Espressif\IDF_Tools\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;F:\Tools\Espressif\IDF_Tools\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;F:\Tools\Espressif\IDF_Tools\.espressif\tools\cmake\3.16.4\bin;F:\Tools\Espressif\IDF_Tools\.espressif\tools\openocd-esp32\v0.10.0-esp32-20210401\openocd-esp32\bin;F:\Tools\Espressif\IDF_Tools\.espressif\tools\ninja\1.10.2;F:\Tools\Espressif\IDF_Tools\.espressif\tools\idf-exe\1.0.1;F:\Tools\Espressif\IDF_Tools\.espressif\tools\ccache\3.7;F:\Tools\Espressif\IDF_Tools\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"F:/Tools/Espressif/IDF_Tools/.espressif/tools/openocd-esp32/v0.10.0-esp32-20210401/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) F:\Tools\Espressif\IDF_Tools\.espressif\python_env\idf4.4_py3.9_env\Scripts\python.exe
Serial port (idf.port) COM35
OpenOCD Configs (idf.openOcdConfigs) board/esp32c3-builtin.cfg
ESP-IDF Tools Path (idf.toolsPath) ${env:IDF_TOOLS_PATH}
Git Path (idf.gitPath) git
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espIdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espIdfPath) false
Access to ESP-IDF Custom extra paths
Access to F:\Tools\Espressif\IDF_Tools\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin: true
Access to F:\Tools\Espressif\IDF_Tools\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin: true
Access to F:\Tools\Espressif\IDF_Tools\.espressif\tools\xtensa-esp32s3-elf\esp-2020r3-8.4.0\xtensa-esp32s3-elf\bin: true
Access to F:\Tools\Espressif\IDF_Tools\.espressif\tools\riscv32-esp-elf\1.24.0.123_64eb9ff-8.4.0\riscv32-esp-elf\bin: true
Access to F:\Tools\Espressif\IDF_Tools\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin: true
Access to F:\Tools\Espressif\IDF_Tools\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin: true
Access to F:\Tools\Espressif\IDF_Tools\.espressif\tools\cmake\3.16.4\bin: true
Access to F:\Tools\Espressif\IDF_Tools\.espressif\tools\openocd-esp32\v0.10.0-esp32-20210401\openocd-esp32\bin: true
Access to F:\Tools\Espressif\IDF_Tools\.espressif\tools\ninja\1.10.2: true
Access to F:\Tools\Espressif\IDF_Tools\.espressif\tools\idf-exe\1.0.1: true
[... 160 lines cut ...]
requests version: 2.25.1
requests-toolbelt version: 0.9.1
schema version: 0.7.4
semantic-version version: 2.8.5
setuptools version: 49.2.1
six version: 1.16.0
tqdm version: 4.61.0
urllib3 version: 1.26.5
websocket-client version: 1.0.1
Werkzeug version: 0.16.1
wheel version: 0.36.2
windows-curses version: 2.2.0
xmlrunner version: 1.7.7
---------------------------------------------------- Check ESP-IDF python requirements.txt -------------------------------------------------
Check ESP-IDF Python packages Python requirements from F:\BLE_WIFI\Espressif\SDK\esp-idf-c3\requirements.txt are satisfied.
---------------------------------------------------- Check extension requirements.txt ------------------------------------------------------
Check Extension Python packages Python requirements from c:\Users\wireless_tech\.vscode\extensions\espressif.esp-idf-extension-1.1.0\requirements.txt are satisfied.
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages Python requirements from c:\Users\wireless_tech\.vscode\extensions\espressif.esp-idf-extension-1.1.0\esp_debug_adapter\requirements.txt are satisfied.
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at ENOENT: no such file or directory, open 'f:\BLE_WIFI\Espressif\HX_DK_FOR_ESP32C3\Software\Example_Project\1_Advertisement\.vscode\launch.json'
--------------------------------------------------------------------------------------------------------------------------------------------

**Additional context**
Add any other context about the problem here.

brianignacio5, 2021-07-05:
Do you see any error in the `SDK Configuration Editor` Output ?

xiaolongba, 2021-07-05:
@brianignacio5 

Hi Bro.

Nothing can be found, the details as below:
[image]

brianignacio5, 2021-07-12:
@xiaolongba Can you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/73592906.zip) ?

xiaolongba, 2021-07-12:
Hi @brianignacio5 

I have tried with this new extension. it works but it took me for 5~6 seconds and then back to the SDK configuration editor. is it the same as you?

brianignacio5, 2021-07-12:
It needs some time because the reset action needs:

1) Delete the `sdkconfig` file
2) Run `idf.py menuconfig` to create default `sdkconfig` and reload `conserver.py` which needs to run CMake configure build step

Best we can do is to make sure the progress bar is showing during this process. The windows will disappear because those values are going to be replaced with default ones.

xiaolongba, 2021-07-12:
@brianignacio5 

got it. it will be better if the time can be decreased.

PR #459 (2021-07-12): Fix load initial values menuconfig
Fix #456 

Fix openOCD version validation not picking the current openOCD value in settings.json
closed 2021-07-12 by brianignacio5