espressif/vscode-esp-idf-extension #457: [Feature Request]:

samsam4, 2021-07-04:
**Is your feature request related to a problem? Please describe.**
Missing way to import existing IDF project (not from ESP-IDF examples)

**Describe the solution you'd like**
Same or similar to how is made in Eclipse

**Describe alternatives you've considered**
Copy the existing IDF project in the ESP-IDF examples tree

**Additional context**
[image]

[image]
.

brianignacio5, 2021-07-05:
I'm not sure what is the feature request here. If a directory is an ESP-IDF project (contains root CMakeLists.txt) you can just open the directory in Visual Studio Code and the extension is activated for this directory.

Same case for an existing workspace, just add a folder to current workspace.

Am I missing something else that the Eclipse plugin is doing when importing an existing project ?

samsam4, 2021-07-05:
I just tried this way, but dont see the .vscode folder  with *.json  files that are normally created when I use ESP-IDF: Create project from extension template or ESP-IDF: Show Examples Projects ?

brianignacio5, 2021-07-05:
You can add the `.vscode folder` with the `ESP-IDF: Add vscode configuration folder` command

samsam4, 2021-07-06:
Ok, it works, but there are couple problems:
1. Is not intuitive, so at least should be some instructions in the readme
2. Is not foolproof and there is a very high chance to become a mess if "ESP-IDF: Add vscode configuration folder" is not applied immediately after the folder with the project is open or at least before VSC start popping  notification: "Would you like to use a compile_commands.json file to autoconfigure IntelliSensse for this folder?" as else in case accepting this proposal, VSC will create a .vscode folder with only c_cpp_properties.json in it which looks like:

```
{
    "configurations": [
        {
            "name": "Win32",
            "includePath": [
                "${workspaceFolder}/**"
            ],
            "defines": [
                "_DEBUG",
                "UNICODE",
                "_UNICODE"
            ],
            "cStandard": "c17",
            "cppStandard": "c++17",
            "intelliSenseMode": "windows-msvc-x64",
            "compileCommands": "${workspaceFolder}/build/compile_commands.json"
        }
    ],
    "version": 4
}
```

and even subsequent call to "ESP-IDF: Add vscode configuration folder" will add only a single line to this file:
```
{
    "configurations": [
        {
            "name": "Win32",
            "includePath": [
                "${workspaceFolder}/**"
            ],
            "defines": [
                "_DEBUG",
                "UNICODE",
                "_UNICODE"
            ],
            "cStandard": "c17",
            "cppStandard": "c++17",
            "intelliSenseMode": "windows-msvc-x64",
            "compileCommands": "${workspaceFolder}/build/compile_commands.json",
            "compilerPath": "D:\\ESP2_8\\.espressif\\tools\\xtensa-esp32-elf\\esp-2020r3-8.4.0\\xtensa-esp32-elf\\bin\\xtensa-esp32-elf-gcc.exe"
        }
    ],
    "version": 4
}
```
______________
while normal c_cpp_properties.json file for ESP-IDF extension looks like:
```
{
    "configurations": [
        {
            "name": "ESP-IDF",
            "compilerPath": "D:\\ESP2_8\\.espressif\\tools\\xtensa-esp32-elf\\esp-2020r3-8.4.0\\xtensa-esp32-elf\\bin\\xtensa-esp32-elf-gcc.exe",
            "cStandard": "c11",
            "cppStandard": "c++17",
            "includePath": [
                "${config:idf.espIdfPath}/components/**",
                "${config:idf.espIdfPathWin}/components/**",
                "${config:idf.espAdfPath}/components/**",
                "${config:idf.espAdfPathWin}/components/**",
                "${workspaceFolder}/**"
            ],
            "browse": {
                "path": [
                    "${config:idf.espIdfPath}/components",
                    "${config:idf.espIdfPathWin}/components",
                    "${config:idf.espAdfPath}/components/**",
                    "${config:idf.espAdfPathWin}/components/**",
                    "${workspaceFolder}"
                ],
                "limitSymbolsToIncludedHeaders": false
            }
        }
    ],
    "version": 4
}

```

closed 2021-07-13 by brianignacio5