espressif/vscode-esp-idf-extension #458: [Bug Report]: Initial Config forgets where Python is on Linux

GeorgeFlorian, 2021-07-08:
**Pre Bug Report Checklist**

**Describe the bug**
Each time I open VSCode and click on the ESP-IDF Explorer the ESP-IDF extension install window pops-up and I get the following errors in the terminal:
```
-- Checking Python dependencies...
The following Python requirements are not satisfied:
python-socketio<5
kconfiglib==13.7.1
construct==2.10.54
To install the missing packages, please run "/home/georgeflorian/esp/esp-idf/install.sh"
Diagnostic information:
    IDF_PYTHON_ENV_PATH: /home/georgeflorian/.espressif/python_env/idf4.2_py3.8_env
    Python interpreter used: /home/georgeflorian/.espressif/python_env/idf4.2_py3.8_env/bin/python
CMake Error at /home/georgeflorian/esp/esp-idf/tools/cmake/build.cmake:274 (message):
  Some Python dependencies must be installed.  Check above message for
  details.
Call Stack (most recent call first):
  /home/georgeflorian/esp/esp-idf/tools/cmake/build.cmake:400 (__build_check_python)
  /home/georgeflorian/esp/esp-idf/tools/cmake/project.cmake:396 (idf_build_process)
  CMakeLists.txt:6 (project)
```
To fix it I have to delete ESP-IDF extension and reinstall it. I have to do this each time I open VSCode.

**To Reproduce**
Open Folder with project inside in VSCode.
Try to build the project for the first time.

**Expected behavior**
I would like for the ESP-IDF to find python and work properly.

**Screenshots**
I have python here:
```
georgeflorian@georgeflorian-desktop:~$ ls /usr/bin/python*

/usr/bin/python   /usr/bin/python2.7  /usr/bin/python3.8         /usr/bin/python3-config
/usr/bin/python2  /usr/bin/python3    /usr/bin/python3.8-config
```
[image]
[image]

As you can see I've installed:
 - ESP-IDF in /home/georgeflorian/esp/esp-idf
 - ESP-IDF Tools in /home/georgeflorian/.espressif
 - python is in /usr/bin/python3.8
 - and the projects folder is in /home/georgeflorian/NewVolume/ESP-IDF

**Environment (please complete the following information):**
- OS Name & Version: Linux Mint 20.1

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS linux x64 5.4.0-77-generic 
System environment variable PATH 
 /home/georgeflorian/.local/bin:/home/georgeflorian/.nvm/versions/node/v16.1.0/bin:/home/georgeflorian/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/georgeflorian/esp/esp-idf/tools:/home/georgeflorian/esp/esp-idf/tools 
Visual Studio Code version 1.57.1 
Visual Studio Code language en 
Visual Studio Code shell bash 
ESP-IDF Extension version 1.1.0 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) /home/georgeflorian/esp/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
Custom extra paths (idf.customExtraPaths) /home/georgeflorian/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin:/home/georgeflorian/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin:/home/georgeflorian/.espressif/tools/xtensa-esp32s3-elf/esp-2020r3-8.4.0/xtensa-esp32s3-elf/bin:/home/georgeflorian/.espressif/tools/riscv32-esp-elf/1.24.0.123_64eb9ff-8.4.0/riscv32-esp-elf/bin:/home/georgeflorian/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/home/georgeflorian/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/home/georgeflorian/.espressif/tools/openocd-esp32/v0.10.0-esp32-20210401/openocd-esp32/bin
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"/home/georgeflorian/.espressif/tools/openocd-esp32/v0.10.0-esp32-20210401/openocd-esp32/share/openocd/scripts"}
Virtual env Python Path (idf.pythonBinPath) /home/georgeflorian/.espressif/python_env/idf4.3_py3.8_env/bin/python
Serial port (idf.port) /dev/ttyUSB1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) ${env:IDF_TOOLS_PATH}
Git Path (idf.gitPath) git
[... 68 lines cut ...]
Werkzeug version: 0.16.1
wheel version: 0.36.2
xmlrunner version: 1.7.7
---------------------------------------------------- Check ESP-IDF python requirements.txt -------------------------------------------------
Check ESP-IDF Python packages Python requirements from /home/georgeflorian/esp/esp-idf/requirements.txt are satisfied.
---------------------------------------------------- Check extension requirements.txt ------------------------------------------------------
Check Extension Python packages Python requirements from /home/georgeflorian/.vscode/extensions/espressif.esp-idf-extension-1.1.0/requirements.txt are satisfied.
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages Python requirements from /home/georgeflorian/.vscode/extensions/espressif.esp-idf-extension-1.1.0/esp_debug_adapter/requirements.txt are satisfied.
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at ENOENT: no such file or directory, open '/home/georgeflorian/NewVolume/ESP-IDF/.vscode/launch.json'
--------------------------------------------------------------------------------------------------------------------------------------------

**Logs**
[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/6784194/esp_idf_vsc_ext.log)
[CMakeOutput.log](https://github.com/espressif/vscode-esp-idf-extension/files/6784197/CMakeOutput.log)

**Extension Configuration settings**
```
 "idf.espIdfPath": "/home/georgeflorian/esp/esp-idf",
    "idf.pythonBinPath": "/home/georgeflorian/.espressif/python_env/idf4.2_py3.8_env/bin/python",
    "idf.customExtraPaths": "/home/georgeflorian/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin:/home/georgeflorian/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin:/home/georgeflorian/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/home/georgeflorian/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/home/georgeflorian/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/bin",
    "idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"/home/georgeflorian/.espressif/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/share/openocd/scripts\"}",
 
```

brianignacio5, 2021-07-08:
Have you try the extension current master [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/71213648.zip) ?

GeorgeFlorian, 2021-07-08:
> Have you try the extension current master [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/71213648.zip) ?

No. I have only installed the extension from the extension manager in VSCode.
How do I use vsix installer on Linux ?

GeorgeFlorian, 2021-07-08:
> Have you try the extension current master [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/71213648.zip) ?

Using this prompts a ton of `not included` errors.
Almost all the header files are underlined with the squiggly red lines.

CMake says:
```
Some Python dependencies must be installed. Check above message for details.
```

brianignacio5, 2021-07-12:
The `not included` errors are because of the definition of `c_cpp_properties.json` file which is used by the Microsoft C/C++ extension.

Did you look into [C/C++ Configuration](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/C_CPP_CONFIGURATION.md) as specified in this project readme ?

GeorgeFlorian, 2021-07-12:
> Is the initial issue fixed with the new vsix ?

Nope. It is not.
I've build and flashed the project with the esp-idf python tool and it worked way better. Even the project works better now.

brianignacio5, 2021-07-12:
This seems to be an issue with your configuration.

In the doctor command you have python virtual env path `/home/georgeflorian/.espressif/python_env/idf4.3_py3.8_env/bin/python`

While in the error output you have `/home/georgeflorian/.espressif/python_env/idf4.2_py3.8_env`

Visual Studio Code can save settings on User Settings, Workspace settings (.codeworkspace) and workspace folder (.vscode/settings.json) as shown in [vscode settings precedence](https://code.visualstudio.com/docs/getstarted/settings#_settings-precedence). 

Make sure your project's `.vscode/settings.json` doesn't override the global settings or has the proper setting.

You can use the `Install ESP-IDF Python Packages` to install missing python packages without running the whole setup.

The extension setup will use the `idf.saveScope` (Idf Save Scope) setting to choose where to save the settings. This setting too can be override locally.

closed 2021-07-12 by brianignacio5