espressif/vscode-esp-idf-extension #460: [Bug Report]: Unable to complete install after failure initial corrected. (Windows 10)

RudyFiero, 2021-07-11:
I had tried to install the ESP-IDF extension but there was an initial failure. It seemed that Git could not be found so the install aborted with errors listed in the console Output.  Prior to doing the ESP-IDF extension install, I had installed the latest Git on my computer. I was able to do a version check of the installed Git in a Windows console, and it was there.

I thought maybe the problem was a path issue. So I went to add Git manually in the Path. The path was set for Git/cmd but not for Git/bin. So I added Git/bin. I hoped this was the reason for the failure. 

I then went back to VS Code and tried to continue with the install of ESP-IDF. I have not found a way to continue with the install. I have not found a way to completely remove the partial install and then to redo the install again.

When I remove ESP-IDF as a component it is shown as have been removed. Then when I select install there is a half second delay, and then it is shown to have been installed. 

I have removed VS Code from my Windows 10, and then I re-installed it. After the install the previous information is still present. I was not able to clear out the previously installed extension.

I ran ESP-IDF Doctor and this is what I got.
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19041 
System environment variable PATH 
 C:\ProgramData\Oracle\Java\javapath;C:\Program Files\Common Files\Microsoft Shared\Windows Live;C:\Program Files (x86)\Common Files\Microsoft Shared\Windows Live;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Windows Live\Shared;C:\Program Files (x86)\QuickTime\QTSystem\;C:\Program Files (x86)\Skype\Phone\;C:\Program Files (x86)\GtkSharp\2.12\bin;C:\TDM-GCC-64\bin;C:\Program Files\Microsoft DNX\Dnvm\;C:\Program Files\Microsoft SQL Server\110\Tools\Binn\;C:\Program Files\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files (x86)\PuTTY\;C:\Program Files\dotnet\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\LLVM\bin;C:\Program Files (x86)\STMicroelectronics\STM32 ST-LINK Utility\ST-LINK Utility;C:\Program Files\Microchip\xc8\v2.32\bin;C:\Program Files\Git\cmd;C:\Program Files\bin;C:\Users\Rudy\.dnx\bin;C:\Python27;C:\Users\Rudy\AppData\Local\Microsoft\WindowsApps;C:\Users\Rudy\AppData\Local\atom\bin;C:\Users\Rudy\AppData\Local\Microsoft\WindowsApps;C:\Users\Rudy\Documents\stm32;C:\Users\Rudy\AppData\Local\Programs\Microsoft VS Code\bin 
Visual Studio Code version 1.58.0 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.1.0 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) ${env:IDF_PATH}
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
Custom extra paths (idf.customExtraPaths) 
Custom extra vars (idf.customExtraVars) 
Virtual env Python Path (idf.pythonBinPath) python
Serial port (idf.port) COM1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) ${env:IDF_TOOLS_PATH}
Git Path (idf.gitPath) C:\Users\Rudy\.espressif\tools\idf-git\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espIdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) false
Access to ESP-MDF Path (idf.espIdfPath) false
Access to ESP-IDF Custom extra paths
Access to Virtual env Python Path (idf.pythonBinPath) false
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) false
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version 2.30.1.windows.1
ESP-IDF version Not found
Python version 3.8.10
Python's pip version 20.1.1
-------------------------------------------------- Python packages in idf.pythonBinPath ----------------------------------------------------
altgraph version: 0.17
future version: 0.18.2
pefile version: 2019.4.18
pip version: 20.1.1
PyInstaller version: 3.6
pywin32-ctypes version: 0.2.0
---------------------------------------------------- Check ESP-IDF python requirements.txt -------------------------------------------------
Check ESP-IDF Python packages Error
---------------------------------------------------- Check extension requirements.txt ------------------------------------------------------
Check Extension Python packages Error
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages Error
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at ENOENT: no such file or directory, open 'c:\Users\Rudy\Documents\PythonProgramming\.vscode\launch.json'
--------------------------------------------------------------------------------------------------------------------------------------------

I have not used VS Code before. I am not a programmer. I am a hardware developer. I have only used the Arduino IDE, which is simple. My manager On Friday I convinced my manager that we should use the Espressif ESP32 for a series of IOT products, rather than his choice of Microchip, since we use mainly Microchip microcontrollers in our products. Because of my insistence of using the ESP32 series of part, I was asked to also do the programming along with the hardware design. So far I have wasted two days and I still do mot have a working IDE. Please tell me how I can resolve my issues.

brianignacio5, 2021-07-12:
We're very sorry @RudyFiero there are some bugs in the current marketplace version.

Can you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/74375661.zip) ?

RudyFiero, 2021-07-12:
Thank you for the quick reply. 

How do I try that installer? I am not very familiar with VS Code.

brianignacio5, 2021-07-12:
You can uninstall the extension, delete the folder `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-1.1.0` and follow the steps [here](https://code.visualstudio.com/docs/editor/extension-marketplace#_install-from-a-vsix)

RudyFiero, 2021-07-12:
I had tried to install ESP-IDF on my home machine, and that is where I had the problems I listed above in this issue. 

Now I am at work and trying to install ESP-IDF on my work machine. I have installed Visual Studio Code. I then did the  Install from VSIX command with the file you provided. VS Code shows that the ESP-IDF extension is installed. But it only took seconds for that to happen so I assumed that the toolchain was not installed. 

I then did, F1  ESP-IDF doctor  and the setup menu EXPRESS/ADVANCED came up. And I also got an error that said, Git is not found in the current environment.  I then opened up a Windows shell and tried Git --version, the result came back with, git version 2.21.0.windows.1. 

So now I don't know what to do. Will the install fail because of the _Git not found in the current environment_ problem? I assume it will. What should I do?

brianignacio5, 2021-07-12:
Well I think is because it is using the previous attempt settings.  try to run the express setup once again to override existing settings with working ones. In Windows, we use a portable embeddable  git executable that is part of the extension setup.

RudyFiero, 2021-07-12:
It looks like it all installed without errors. Next step is to try a project.

Thank you for your help.

closed 2021-07-13 by brianignacio5

RudyFiero, 2021-07-13:
Hi Brian. 
I tried the new install on my home machine, but I got an error. 

Check Debug AdapterPython packages Python requirements from c:\Users\Rudy\.vscode\extensions\espressif.esp-idf-extension-1.1.0\esp_debug_adapter\requirements.txt are satisfied.
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at ENOENT: no such file or directory, open 'c:\Users\Rudy\Documents\PythonProgramming\.vscode\launch.json'
--------------------------------------------------------------------------------------------------------------------------------------------

Did I do something wrong? Is this error significant?