espressif/vscode-esp-idf-extension #466: [Bug Report]: Flash does not work on windows when MSYS2 is installed.

cilliemalan, 2021-07-14:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [x] Make sure you have searched for existing bugs and features request before you post an issue.
- [x] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [x] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
The command "ESP-IDF: Flash (UART) your project" fails. It attempts to run a command with bash but with windows/cmd.exe-style paths.

**To Reproduce**
1. Install MSYS2. Make sure bash is in your path.
2. Install ESP-IDF.
3. Install the ESP IDF VSCode Extension.
4. Create a new project and build (build succeeds).
5. Run the command "ESP-IDF: Flash (UART) your project"
6. This is the resulting terminal output:
```
> Executing task: C:\Users\blah\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe c:\tools\esp-idf\components\esptool_py\esptool\esptool.py -p COM28 -b 460800 --before default_reset --after hard_reset --chip esp32 write_flash --flash_mode dio --flash_freq 40m --flash_size detect 0x10000 geyser.bin 0x1000 bootloader/bootloader.bin 0x8000 partition_table/partition-table.bin <

/usr/bin/bash: C:Usersblah.espressifpython_envidf4.3_py3.8_envScriptspython.exe: command not found
The terminal process "C:\tools\msys64\usr\bin\bash.exe '-c', 'C:\Users\blah\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe c:\tools\esp-idf\components\esptool_py\esptool\esptool.py -p COM28 -b 460800 --before default_reset --after hard_reset --chip esp32 write_flash --flash_mode dio --flash_freq 40m --flash_size detect 0x10000 geyser.bin 0x1000 bootloader/bootloader.bin 0x8000 partition_table/partition-table.bin'" terminated with exit code: 127.
```
7. As you can see it's running `bash` with a windows-style path with backslashes, which bash does not understand.

**Expected behavior**
It flashes the built executable.

**Screenshots**
[image]

**Environment (please complete the following information):**

- Windows
- VSCode Version: 1.58.0
- ESP-IDF Version: 1.1.1
- Python Version: 3.8.10

**Output**
```
Installing ESP-IDF Debug Adapter python packages in C:/Users/blah/.espressif/python_env/idf4.3_py3.8_env/Scripts/python.exe ...

Ignoring typing: markers 'python_version < "3.5"' don't match your environment
Requirement already satisfied: setuptools>=21 in c:\users\blah\.espressif\python_env\idf4.3_py3.8_env\lib\site-packages (from -r c:\Users\blah\.vscode\extensions\espressif.esp-idf-extension-1.1.1\esp_debug_adapter\requirements.txt (line 1)) (57.1.0)
Requirement already satisfied: click in c:\users\blah\.espressif\python_env\idf4.3_py3.8_env\lib\site-packages (from -r c:\Users\blah\.vscode\extensions\espressif.esp-idf-extension-1.1.1\esp_debug_adapter\requirements.txt (line 3)) (8.0.1)
Requirement already satisfied: psutil>=5.5.1 in c:\users\blah\.espressif\python_env\idf4.3_py3.8_env\lib\site-packages (from -r c:\Users\blah\.vscode\extensions\espressif.esp-idf-extension-1.1.1\esp_debug_adapter\requirements.txt (line 4)) (5.8.0)
Requirement already satisfied: pygdbmi<=0.9.0.2 in c:\users\blah\.espressif\python_env\idf4.3_py3.8_env\lib\site-packages (from -r c:\Users\blah\.vscode\extensions\espressif.esp-idf-extension-1.1.1\esp_debug_adapter\requirements.txt (line 5)) (0.9.0.2)
Requirement already satisfied: pywin32>=227 in c:\users\blah\.espressif\python_env\idf4.3_py3.8_env\lib\site-packages (from -r c:\Users\blah\.vscode\extensions\espressif.esp-idf-extension-1.1.1\esp_debug_adapter\requirements.txt (line 6)) (301)
Requirement already satisfied: requests>=2.21.0 in c:\users\blah\.espressif\python_env\idf4.3_py3.8_env\lib\site-packages (from -r c:\Users\blah\.vscode\extensions\espressif.esp-idf-extension-1.1.1\esp_debug_adapter\requirements.txt (line 7)) (2.26.0)
Requirement already satisfied: xmlrunner>=1.7.7 in c:\users\blah\.espressif\python_env\idf4.3_py3.8_env\lib\site-packages (from -r c:\Users\blah\.vscode\extensions\espressif.esp-idf-extension-1.1.1\esp_debug_adapter\requirements.txt (line 9)) (1.7.7)
Requirement already satisfied: certifi>=2017.4.17 in c:\users\blah\.espressif\python_env\idf4.3_py3.8_env\lib\site-packages (from requests>=2.21.0->-r c:\Users\blah\.vscode\extensions\espressif.esp-idf-extension-1.1.1\esp_debug_adapter\requirements.txt (line 7)) (2021.5.30)
Requirement already satisfied: charset-normalizer~=2.0.0 in c:\users\blah\.espressif\python_env\idf4.3_py3.8_env\lib\site-packages (from requests>=2.21.0->-r c:\Users\blah\.vscode\extensions\espressif.esp-idf-extension-1.1.1\esp_debug_adapter\requirements.txt (line 7)) (2.0.1)
Requirement already satisfied: idna<4,>=2.5 in c:\users\blah\.espressif\python_env\idf4.3_py3.8_env\lib\site-packages (from requests>=2.21.0->-r c:\Users\blah\.vscode\extensions\espressif.esp-idf-extension-1.1.1\esp_debug_adapter\requirements.txt (line 7)) (3.2)
Requirement already satisfied: urllib3<1.27,>=1.21.1 in c:\users\blah\.espressif\python_env\idf4.3_py3.8_env\lib\site-packages (from requests>=2.21.0->-r c:\Users\blah\.vscode\extensions\espressif.esp-idf-extension-1.1.1\esp_debug_adapter\requirements.txt (line 7)) (1.26.6)
Requirement already satisfied: colorama in c:\users\blah\.espressif\python_env\idf4.3_py3.8_env\lib\site-packages (from click->-r c:\Users\blah\.vscode\extensions\espressif.esp-idf-extension-1.1.1\esp_debug_adapter\requirements.txt (line 3)) (0.4.4)
WARNING: You are using pip version 21.1.2; however, version 21.1.3 is available.
You should consider upgrading via the 'C:\Users\blah\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe -m pip install --upgrade pip' command.
```

**Logs**
```
{"message":"Open On-Chip Debugger  v0.10.0-esp32-20210401 (2021-04-01-15:46)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n","stack":"Error: Open On-Chip Debugger  v0.10.0-esp32-20210401 (2021-04-01-15:46)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n\n    at c:\\Users\\blah\\.vscode\\extensions\\espressif.esp-idf-extension-1.1.1\\dist\\extension.js:1:6907\n    at ChildProcess.exithandler (child_process.js:308:7)\n    at ChildProcess.emit (events.js:315:20)\n    at ChildProcess.EventEmitter.emit (domain.js:467:12)\n    at maybeClose (internal/child_process.js:1048:16)\n    at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-07-14T10:30:30.972Z"}
{"message":"Open On-Chip Debugger  v0.10.0-esp32-20210401 (2021-04-01-15:46)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n","stack":"Error: Open On-Chip Debugger  v0.10.0-esp32-20210401 (2021-04-01-15:46)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n\n    at c:\\Users\\blah\\.vscode\\extensions\\espressif.esp-idf-extension-1.1.1\\dist\\extension.js:1:6907\n    at ChildProcess.exithandler (child_process.js:308:7)\n    at ChildProcess.emit (events.js:315:20)\n    at ChildProcess.EventEmitter.emit (domain.js:467:12)\n    at maybeClose (internal/child_process.js:1048:16)\n    at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-07-14T10:30:59.022Z"}
{"message":"WARNING: You are using pip version 21.1.2; however, version 21.1.3 is available.\r\nYou should consider upgrading via the 'C:\\Users\\blah\\.espressif\\python_env\\idf4.3_py3.8_env\\Scripts\\python.exe -m pip install --upgrade pip' command.\r\n","stack":"Error: WARNING: You are using pip version 21.1.2; however, version 21.1.3 is available.\r\nYou should consider upgrading via the 'C:\\Users\\blah\\.espressif\\python_env\\idf4.3_py3.8_env\\Scripts\\python.exe -m pip install --upgrade pip' command.\r\n\n    at c:\\Users\\blah\\.vscode\\extensions\\espressif.esp-idf-extension-1.1.1\\dist\\extension.js:1:6907\n    at ChildProcess.exithandler (child_process.js:308:7)\n    at ChildProcess.emit (events.js:315:20)\n    at ChildProcess.EventEmitter.emit (domain.js:467:12)\n    at maybeClose (internal/child_process.js:1048:16)\n    at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-07-14T10:31:30.970Z"}
{"message":"WARNING: You are using pip version 21.1.2; however, version 21.1.3 is available.\r\nYou should consider upgrading via the 'C:\\Users\\blah\\.espressif\\python_env\\idf4.3_py3.8_env\\Scripts\ [... cut ...] -util\\0.9\\dfu-util-0.9-win64",
    "idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"C:\\\\Users\\\\blah\\\\.espressif\\\\tools\\\\openocd-esp32\\\\v0.10.0-esp32-20210401/openocd-esp32/share/openocd/scripts\",\"IDF_CCACHE_ENABLE\":\"1\"}",
    "idf.adapterTargetName": "esp32",
    "idf.openOcdConfigs": [
        "interface/ftdi/esp32_devkitj_v1.cfg",
        "target/esp32.cfg"
    ],
    "idf.portWin": "COM28",
    "idf.flashType": "UART",
    "idf.toolsPathWin": "C:\\Users\\blah\\.espressif"
```

brianignacio5, 2021-07-14:
Makes sense. We are using Windows style paths on Windows by default because most users will stick to cmd.exe and Powershell.exe terminals. WSL users are actually running on a Linux machine.

In the meantime If you want to use MSYS bash you can modify your extension configuration settings with forward slashes. 

We'll think of how to support this in the extension.

cilliemalan, 2021-07-14:
You can probably fix it by [using `vscode.ProcessExecution` instead of `vscode.ShellExecution`](/cilliemalan/vscode-esp-idf-extension/commit/f369b09cca68f7803cd6b22511743e42cee67c66)

in [src/flash/flashTask.ts:139](https://github.com/espressif/vscode-esp-idf-extension/blob/master/src/flash/flashTask.ts#L139-L143)
```diff
     const pythonBinPath = idfConf.readParameter("idf.pythonBinPath") as string;
-    return new vscode.ShellExecution(
-      `${pythonBinPath} ${flasherArgs.join(" ")}`,
-      options
-    );
+    return new vscode.ProcessExecution(pythonBinPath, flasherArgs, options);
   }
```

brianignacio5, 2021-07-15:
The problem with that approach is that it doesn't work for MacBook zsh, we need to make sure it works for most use cases.

brianignacio5, 2021-07-15:
The main issue I find about this approach is output formatting.

When tested using Process Execution it seems that output gets truncated which makes output hard to read. Doesn't this happens to you ?

cilliemalan, 2021-07-15:
This is what I see when I change it to `vscode.ProcessExecution` and I run the *ESP-IDF: Flash (UART) your project* command:
[image]

PR #469 (2021-07-16): use process execution for flash task
Fix #466 

Use ProcessExecution for flash task instead of ShellExecution to avoid path separator issues.
closed 2021-07-16 by brianignacio5