espressif/vscode-esp-idf-extension #471: [Feature Request]: Please support rffc2217 in device port select menu.

innovak, 2021-07-15:
I use wsl2 and vscode as the development environment on win10, and I can use some like -p rffc2217://xxx.xxx.xxx.xxx:1234 in the command line mode to flash the esp32 chip. But in vscode, the idf plug-in only let me choose an existing port instead of input the string of rffc2217. I hope I can input the string to use rffc2217 mode. Or maybe the idf plugin can directly do the rffc2217 server function in the vscode remote mode?

brianignacio5, 2021-07-16:
While the command `ESP-IDF: Select port to use` only shows available serial ports in your system, you can still manually edit the `idf.port` (`idf.portWin` for Windows users) with any value you want by using the `Preferences: Open Settings (JSON)`, `Preferences: Open Settings (UI)` or directly in your `<project-directory-path>/.vscode/settings.json`

More about this extension settings in [here](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/SETTINGS.md#esp-idf-settings)

brianignacio5, 2021-07-19:
Closing this issue since the user can currently modify the value as desired.

closed 2021-07-19 by brianignacio5