espressif/vscode-esp-idf-extension #475: [Bug Report]:

MiladMohammadi72, 2021-07-18:
Hello 
It's ok and work good I mean I can compile and run and test the code 
But !!! 
I can't use intellisense engine for includes library and variable than I define
 even after compile It work without error but auto complete not work 

Look at this photos plz 

.

[image]

.

[image]

.

[image]

.

- OS Name & Version: win 10 x64
- VSCode Version: 1.58.2
- ESP-IDF Version: 1.1.1
- Python Version: 3.8 above

Thanks

brianignacio5, 2021-07-19:
Hi @MiladMohammadi72 have you look into [C_CPP configuration ](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/C_CPP_CONFIGURATION.md) documentation ?

We rely on the Microsoft C/C++ extension to provide auto-complete, syntax highlight and code navigation (i.e. Intellisense). IF you can build successfully, I would suggest to add the `compile_commands.json` to `c_cpp_properties.json` file for better results.

MiladMohammadi72, 2021-07-19:
> Hi @MiladMohammadi72 have you look into [C_CPP configuration ](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/C_CPP_CONFIGURATION.md) documentation ?
>
> We rely on the [...]

Thanks
yeah [C_CPP configuration ](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/C_CPP_CONFIGURATION.md) it's just work for include and header
But not for struc variable in first Picture...

brianignacio5, 2021-07-19:
yes it should also work for autocomplete too. If you put the `workspaceFolder` path first in the include array it should give priority for source defined variables.

This is not yet optimized since we do not maintain the Microsoft C/C++ extension. We have yet to optimize this somehow. Very sorry for the inconvenience.

brianignacio5, 2021-07-28:
Closing this since it is not an issue but user misconfiguration. Please comment here if more help is needed.

closed 2021-07-28 by brianignacio5