espressif/vscode-esp-idf-extension #476: [Feature Request]: Project example that that includes external library

RudyFiero, 2021-07-19:
**Is your feature request related to a problem? Please describe.**

I want to run the series of progressive examples from the espressif/esp-jumpstart repo. 
[esp-jumpstart](https://github.com/espressif/esp-jumpstart)

I was able to compile the 1_hello_world example. I then tried the 2_drivers example. As far as I can tell this second example needs routines from espressif/esp-iot-solution  [esp-iot-solution](https://github.com/espressif/esp-iot-solution)
Specifically esp-iot-solution/components/**button**/

I looked through the examples with **ESP-IDF: Show Example Projects**  and could not find an example with external libraries. I found no structure/ Cmake examples. 

There is a lot of information regarding Build System documentation, but not much in the way of examples. From my searches on the internet there seems to be little information available and what there is is difficult to understand, or doesn't match my described use case.
 

**Describe the solution you'd like**
A clear and concise description of what is required to include external libraries. In the project I am working on right now I want to include a character LCD display communicated via I2C. I want it there for development diagnostics. It would not be in the final product. 

I have found a couple of IDF based libraries, but I have not been able to successfully include them in a minimal project yet. So far I have been working on this project for a week and most of the issues I have had has been with getting working code to run under this framework. (not just vscode-esp-idf-extension issues)

**Describe alternatives you've considered**
Telling my manager that I can't handle the coding issues and that I will instead design in a Microchip component since the firmware programmers are already familiar with the Harmony framework. (I'm a hardware developer, not a programmer)

**Additional context**
I asked for help on the ESP32 forum.  [VS Code Extension - Cmake help needed. (or other direction)](https://www.esp32.com/viewtopic.php?f=40&p=80333#p80333) 
Hopefully that will get me moving forward. 
But even if I get past this hurdle, I believe that what I asked for here would be helpful to others in the future.

brianignacio5, 2021-07-19:
HI Rudy,

As shown in the ESP-IOT project in [here](https://github.com/espressif/esp-iot-solution#create-your-project), you need to add an include in your project's root CMakeLists.txt as shown below:

```
cmake_minimum_required(VERSION 3.5)

include($ENV{IOT_SOLUTION_PATH}/component.cmake)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)

project(project-name)
```

in the previous case assuming you set an environment variable in your system called **IOT_SOLUTION_PATH** with the path to **esp-iot-solution** or just use the full path of component.cmake instead of $ENV{IOT_SOLUTION_PATH}/component.cmake.

After adding this, you should be able to include source code from ESP-IOT-Solution as you do in the ESP-IOT-SOLUTION examples.

RudyFiero, 2021-07-19:
Hi Brian,

I thought I had to do something like
**set(EXTRA_COMPONENT_DIRS ${CMAKE_CURRENT_LIST_DIR}/../components)**

I did try
include($ENV{IOT_SOLUTION_PATH}/component.cmake)
but it didn't work.  

But it did find it when I changed it to, 
**include(C:/Users/Rudy/esp/esp-iot-solution/component.cmake)**
====================================================

While it now included the library I did get an error. 
**../main/app_driver.c:12:10: fatal error: iot_button.h: No such file or directory
 #include <iot_button.h>**

I don't understand why it did not find it. 
esp-iot-solution/components/button/include/iot_button.h
[iot_button.h](https://github.com/espressif/esp-iot-solution/blob/4787f174ba4d3c50f61dc560fc15b1d9b7d8f415/components/button/include/iot_button.h)

brianignacio5, 2021-07-19:
It makes sense it doesn't work. Look at https://github.com/espressif/esp-iot-solution/blob/4787f174ba4d3c50f61dc560fc15b1d9b7d8f415/component.cmake

When you do `include($ENV{IOT_SOLUTION_PATH}/component.cmake)` it doesn't work because you don't have an environment variable in your system called **IOT_SOLUTION_PATH**.  Same reason it is happening inside `C:/Users/Rudy/esp/esp-iot-solution/component.cmake`. On your project you used `include(C:/Users/Rudy/esp/esp-iot-solution/component.cmake)` which made it partially work.

You could:
1) Modify `C:/Users/Rudy/esp/esp-iot-solution/component.cmake` changing $ENV{IOT_SOLUTION_PATH} for `C:/Users/Rudy/esp/esp-iot-solution`.
2) Set IOT_SOLUTION_PATH as environment variable in this extension by modifying:

```
"idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"C:\\\\Users\\\\Rudy\\\\.espressif\\\\tools\\\\openocd-esp32\\\\v0.10.0-esp32-20210401/openocd-esp32/share/openocd/scripts\",\"IDF_CCACHE_ENABLE\":\"1\"}"
```
to 
```
"idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"C:\\\\Users\\\\Rudy\\\\.espressif\\\\tools\\\\openocd-esp32\\\\v0.10.0-esp32-20210401/openocd-esp32/share/openocd/scripts\",\"IDF_CCACHE_ENABLE\":\"1\",\"IOT_SOLUTION_PATH\":\"C:\\\\Users\\\\Rudy\\\\esp\\\\esp-iot-solution\"}"
```

3. Set `IOT_SOLUTION_PATH ` in your system environment variables.

* Open System (Control Panel)
* Click Advanced system settings.
* Click **Environment Variables**. In the section **System Variables** find or add an environment variable and select it. 
* In the Edit System Variable (or New System Variable) window, specify the value of the environment variable.
*  Click OK. Close all remaining windows by clicking OK.

RudyFiero, 2021-07-20:
When I saw your reply to the issue I had to try it right away. Unfortunately this was just after 4 am. I then sent you off my reply with the information I had. As I was in the shower I thought about the path variable. I realized that I had put the path in the Rudy User variable rather than the system as I should have. I believe that was why it didn't work for me. Then I had done a direct path, and that worked partially, for the reasons you have described.

 Now I have the path set the way you said. 

```
# The following lines of boilerplate have to be in your project's
# CMakeLists in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.5)

# set(EXTRA_COMPONENT_DIRS ${CMAKE_CURRENT_LIST_DIR}/../components)

include($ENV{IOT_SOLUTION_PATH}/component.cmake)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(2_drivers)
```
[image]

On my home machine I am still getting the, file not found error.

[image]

On my work machine I think it was working. At least for the **2_drivers** file compile. I was able to use the push button to turn on and off the output. But when I compiled the **3_wifi_connection** (button control plus connect to WiFi) the button control stopped working. I wanted to try it at home to see if I got the same response, because it should have worked, based on the code I looked at. So I don't know what is going on.

On my home machine, now with the missing include, I thought that something didn't get cleaned out. I tried to clean things up, to do a complete build. 

I want to make a copy of the original **2_drivers** code and start a new project. But it is too late to do that tonight.

brianignacio5, 2021-07-20:
Hi Rudy, this late issue is just a lint issue. Your project should have a `.vscode/c_cpp_properties.json` which is used by the Microsoft C/C++ to resolve include headers, auto complete, etc.

Take a look at [C/C++ Configuration](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/C_CPP_CONFIGURATION.md) documentation.

If you build successfully already, consider using `"compileCommands": "${workspaceFolder}/build/compile_commands.json"` or add it to the include section like:

```json
"includePath": [
        "${config:idf.espIdfPath}/components/**",
        "${env:IOT_SOLUTION_PATH}/components/**",
        "${config:idf.espIdfPathWin}/components/**",
        "${workspaceFolder}/**"
      ],
```

brianignacio5, 2021-07-28:
Closing this issue since it is more of a project configuration issue than extension issue.

closed 2021-07-28 by brianignacio5