espressif/vscode-esp-idf-extension #477: Won't switch between usb ports

rickAllDev, 2021-07-19:
I have 2 esp32's attached to linux box (ubuntu 20.04). switching between usb ports using the extension has always worked fine until today. 
Now, when I select anything other than usb0, I get a message that says the port has been updated but when I connect (monitor or flash) it connects to usb0.
I have tested the connection with putty and everything works fine, swapped to different usb ports, switch the chips.

I am using espidf v4.2, extension v1.1.1

brianignacio5, 2021-07-19:
Maybe there is an `idf.port` in your project `.vscode/settings.json` overriding the User settings `idf.port` ?

Visual Studio Code has many places where to set configuration settings. This extension uses the `idf.saveScope` configuration setting to determine where to save settings, Global (User Settings), Workspace and WorkspaceFolder. Please review the [visual studio code settings precedence](https://code.visualstudio.com/docs/getstarted/settings#_settings-precedence).

rickAllDev, 2021-07-19:
thanks, there was indeed an idf.port in settings. funny, I've been working on the same codebase for awhile and have never had a problem switching ports. Deleted the setting and everything back to normal.

closed 2021-07-19 by rickAllDev