espressif/vscode-esp-idf-extension #480: [Bug Report]: New project template selector ONLY shows extension projects

nickbits1024, 2021-07-27:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [ ] Make sure you have searched for existing bugs and features request before you post an issue.
- [ ] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [ ] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
After selecting "Choose template" from the new project dialog, only extension templates populate.  If ESP-IDF is selected from the dropdown, the page is blank.
This issue only occurs in v1.1.1 of the plugin.  v1.1.0 works, so I suspect it was this PR https://github.com/espressif/vscode-esp-idf-extension/pull/461
vscode is finding the examples (as shown in the last pic) but doesn't display them
It also seems weird that Extension is the default selection. 

**To Reproduce**
Steps to reproduce the behavior (including the full log).
Execute new project command
Select Choose Template
Toggle between Extension and ESP-IDF project types

**Expected behavior**
Selecting ESP-IDF should populate the ESP-IDF examples

**Screenshots**
working v1.1.0:
[image]
broken v1.1.1:
[image]
[image]
[image]
[image]

**Environment (please complete the following information):**

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19042 
System environment variable PATH 
 C:\Program Files (x86)\Microsoft SDKs\Azure\CLI2\wbin;C:\Program Files\Python39\Scripts\;C:\Program Files\Python39\;C:\Program Files\ImageMagick-7.0.9-Q16;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\WinMerge;C:\Program Files (x86)\Common Files\Intuit\QBPOSSDKRuntime;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\170\Tools\Binn\;C:\Program Files (x86)\Java\jdk1.8.0_161\bin;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files (x86)\Microsoft SQL Server\150\Tools\Binn\;C:\Program Files\Microsoft SQL Server\150\Tools\Binn\;C:\Program Files\Microsoft SQL Server\150\DTS\Binn\;C:\Program Files (x86)\Microsoft SQL Server\150\DTS\Binn\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Utility;C:\Program Files\dotnet\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\nodejs\;C:\Program Files\Azure Data Studio\bin;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\TortoiseGit\bin;C:\Program Files (x86)\dotnet\;C:\Program Files (x86)\Bluetooth Command Line Tools\bin;C:\Program Files\Microsoft\Azure Functions Core Tools\;C:\Program Files\Git\cmd;C:\Users\nickb\AppData\Local\Microsoft\WindowsApps;C:\Users\nickb\AppData\Local\Programs\Fiddler;C:\Users\nickb\AppData\Local\Microsoft\WindowsApps;C:\Users\nickb\.dotnet\tools;C:\Users\nickb\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\nickb\.dotnet\tools 
Visual Studio Code version 1.58.2 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.1.1 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:\Users\nickb\esp\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
Custom extra paths (idf.customExtraPaths) C:\Users\nickb\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin;C:\Users\nickb\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin;C:\Users\nickb\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Users\nickb\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Users\nickb\.espressif\tools\cmake\3.16.4\bin;C:\Users\nickb\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin;C:\Users\nickb\.espressif\tools\ninja\1.10.0;C:\Users\nickb\.espressif\tools\idf-exe\1.0.1;C:\Users\nickb\.espressif\tools\ccache\3.7;C:\Users\nickb\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"C:\\Users\\nickb\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20200709/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) C:\Users\nickb\.espressif\python_env\idf4.2_py3.9_env\Scripts\python.exe
Serial port (idf.port) COM3
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) C:\Users\nickb\.espressif
Git Path (idf.gitPath) C:\Users\nickb\.espressif\tools\idf-git\2.30.1\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espIdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espIdfPath) false
Access to ESP-IDF Custom extra paths
Access to C:\Users\nickb\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin: true
Access to C:\Users\nickb\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin: true
Access to C:\Users\nickb\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin: true
Access to C:\Users\nickb\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin: true
Access to C:\Users\nickb\.espressif\tools\cmake\3.16.4\bin: true
Access to C:\Users\nickb\.espressif\tools\openocd-esp32\v0.10.0-esp32-20200709\openocd-esp32\bin: true
Access to C:\Users\nickb\.espressif\tools\ninja\1.10.0: true
Access to C:\Users\nickb\.espressif\tools\idf-exe\1.0.1: true
Access to C:\Users\nickb\.espressif\tools\ccache\3.7: true
Access to C:\Users\nickb\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
[... 92 lines cut ...]
**Output**

Click the menu `View` and select `Output`. This extension uses several output channels: ESP-IDF, ESP-IDF Debug Adapter, Heap Trace, OpenOCD and SDK Configuration Editor.

Please share the output of the channel related to your issue. For example ESP-IDF is used mostly for the extension setup process.

**Logs**
If applicable, please share the log file which can be obtained from

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

**Extension Configuration settings**
Open Command Palette (F1 or View Menu -> Command Palette) and type `Preferences: Open Settings (JSON)` or use `{CURRENT_PROJECT}/.vscode/settings.json` to get the following extension configuration settings value:

- `idf.espIdfPath` (MacOS or Linux) or `idf.espIdfPathWin` (Windows) also called `Path to locate ESP-IDF framework (IDF_PATH)` in `Preferences: Open Settings (UI)`.
- `idf.pythonBinPath` (MacOS or Linux) or `idf.pythonBinPathWin` (Windows) also called `Python absolute binary path used to execute ESP-IDF Python Scripts` in `Preferences: Open Settings (UI)`.
- `idf.openOcdConfigs` also called `List of configuration files inside OpenOCD Scripts directory` in `Preferences: Open Settings (UI)`.
- `idf.customExtraPaths` also called `Paths to be appended to PATH` in `Preferences: Open Settings (UI)`.
- `idf.customExtraVars` also called `Variables to be added to system environment variables` in `Preferences: Open Settings (UI)`.
- `idf.adapterTargetName` also called `Target name for ESP-IDF Debug Adapter` in `Preferences: Open Settings (UI)`.
- `idf.customAdapterTargetName` is used when `idf.adapterTargetName` is set to `custom`.

**Additional context**
Add any other context about the problem here.

brianignacio5, 2021-07-28:
Can you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/75705023.zip) ? I tested with GitHub master and seems to work on Windows.

Based on your doctor command output you should run `Install ESP-IDF Python Packages` command for missing python packages in your virtual environment.

nickbits1024, 2021-07-28:
No difference.  Literally everything works perfect (build/flash/etc) except for that.  I did try the cmd you suggested and it said "ESP-IDF Python python Requirements has been installed" but no change in behavior.
In vscode, if I revert back to 1.1.0 (using "Install Another Version...") it works and if I upgrade it breaks.  That cycle is reproducible.  I did spend a while trying to "fix" it by uninstalling/reinstalling everything to no avail and that could have messed up the doctor report (reinstall was not smooth like my initial experience) but everything else does work.
If I get some time, I'll trying debugging the extension from the source and maybe make a PR.

brianignacio5, 2021-07-28:
Maybe you can share the logs ?

> Windows: %USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log
> Linux & MacOSX: $HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log

There could also be an error in the New project wizard UI also. There is a `Developer: Open Webview Developer tools` command and there might be an error in the Chrome console output.

I can reproduce the issue on Windows with marketplace version but not in the GitHub master . Don't seem to be any diff between code in new project wizard as far as a I see.

nickbits1024, 2021-07-28:
I will get you those after work today.  I'll also retry the visx you sent and see how it behaves on my mac.

nickbits1024, 2021-07-29:
Tried the vsix you provided again and no change.  Also tried it on my mac (new install) and it did the same thing.
pc log:
[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/6896836/esp_idf_vsc_ext.log)
mac log:
[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/6897124/esp_idf_vsc_ext.log)
but they don't seem to have anything interesting plus there's not much there since the reinstall.   There wasn't anything in the dev console.
I was going to try and debug it from source but it doesn't seem to build and I'm not sure it's worth it for me to work though all the errors.  Let me know if you want to try anything else.

brianignacio5, 2021-07-29:
When you reinstall the extension, make sure to delete the extension folder in `%USERPROFILE%.vscode\extensions\espressif.esp-idf-extension-VERSION` (Windows) or `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION` (MacOS)

nickbits1024, 2021-07-29:
Nuked those folders but it didn't make a difference.  
I was able to get the extension to build and found something odd.  When I "run" the extension from source in the "Extension Development Host", it works.  I made a package from that and installed it but then it quits working?!

PR #486 (2021-08-03): fix recursive reference of component itself
Fix #480 

Calling Vue component recursively doesn't know how to resolve itself. Need to dynamically import himself.
closed 2021-08-03 by brianignacio5