espressif/vscode-esp-idf-extension #482: [Bug Report]: Command-E is a standard shortcut on Mac OS.

gordonklaus, 2021-07-28:
On Mac OS, the keyboard shortcut Command-E is a system-wide standard text-editing shortcut which has the function "Use Selection for Find".  The user selects text, presses Command-E, then presses Command-G to find the next instance of the selected text (or presses Command-F to open a Find pane with that text already entered in the text field).

This VSCode ESP IDF extension conflicts with that shortcut, disrupting a common text-editing procedure.

brianignacio5, 2021-07-29:
I'm not sure why it is disrupting. All commands **need another letter** after CMD+E for example build is CMD+E (pressed at same time) and then pressing B. If you select a text then do CMD+E, you are indeed doing the MacOS selection command but you will need to press another letter to do any of this extension commands, which you probably won't.

But if this behavior is not desired, you can always modify your keyboard shortcuts in Visual Studio Code as shown [here](https://code.visualstudio.com/docs/getstarted/keybindings#_advanced-customization). 

We have thought about default keyboard shortcuts and if you have a suggestion for different shortcuts, please let us know.

gordonklaus, 2021-07-29:
It is indeed disrupting.  I've tried again now and here is the behavior I see:
- Select some text and press Command-E.  The status bar now says "(Command-E) was pressed.  Waiting for second key of chord..."
- Press, e.g., Command-G, intending to find the next instance of the selected text.  Nothing happens, and the status bar says "The key combination (Command-E, Command-G) is not a command."
- Now, it is clear that the initial Command-E did **not** in fact work, as, if I press Command-G now again, it does not find the next instance of the selected text.

The fact is, I always do the sequence (Command-E, Command-G) or (Command-E, Command-F), never just Command-E alone.

Yes, I have previously worked around this by deleting the ESP-IDF keyboard shortcuts.  However, the problem resurfaced when the extension updated and restored (added?) some shortcuts.

I'm not sure what to suggest for a shortcut except anything that is not already assigned :)   Perhaps Control-E?

brianignacio5, 2021-08-06:
CTRL+E in Mac is the go to the end of line shortcut. Hard to figure an alternative that makes sense.

If others have some alternatives please share

bpietsch, 2021-08-07:
I've had this issue as well. I modified all of the esp shortcuts to start with Cmd+I instead of Cmd+E... of course every time a new one is added I have to modify them too. I agree that a default other than E would make sense (at least on Mac)

closed 2021-08-24 by brianignacio5

ankostis, 2022-04-15:
Actually `Ctrl+E` is one of the most used default key-binding, bound to "Got to File...", to move across files in a project.

Is there a way to cancel all shortcuts re-defined by the extension, en-mass?

NOTE: i'm strongly against *SILENTLY* *RE* -defining keyboard shortcuts,
not at least without giving a choice to the user whether to adopt the new shortcuts.

brianignacio5, 2022-04-16:
if this behavior is not desired, you can always modify your keyboard shortcuts in Visual Studio Code as shown [here](https://code.visualstudio.com/docs/getstarted/keybindings#_advanced-customization).

PR #493 (2021-08-24): change default keys for macOS
Fix #482

Use CMD+ I **KEY** chords as extension default keyboard shortcuts in macOS