espressif/vscode-esp-idf-extension #488: My ESP-IDF installed in VS Code can not initialize or configure after running some official examples :( (IDFGH-5654)

johnnyguoyy, 2021-08-06:
Hi dude, 
As I said in title, at first I can test those examples successfully, but later, every time I started the VSCode, the ESP-IDF  failed to initialize, and reported some errors which I cannot find way to solve. And as the initialization failed, a page of "extention setup"
occurred.
errors below :
 'xtensa-esp32s3-elf-gcc' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Command failed: xtensa-esp32s3-elf-gcc --version
'xtensa-esp32s3-elf-gcc' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Error checking xtensa-esp32s3-elf version
Error: Command failed: xtensa-esp32s3-elf-gcc --version
'xtensa-esp32s3-elf-gcc' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���

'xtensa-esp32-elf-gcc' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Command failed: xtensa-esp32-elf-gcc --version
'xtensa-esp32-elf-gcc' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Error checking xtensa-esp32-elf version
Error: Command failed: xtensa-esp32-elf-gcc --version
'xtensa-esp32-elf-gcc' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���

'xtensa-esp32s2-elf-gcc' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Command failed: xtensa-esp32s2-elf-gcc --version
'xtensa-esp32s2-elf-gcc' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Error checking xtensa-esp32s2-elf version
Error: Command failed: xtensa-esp32s2-elf-gcc --version
'xtensa-esp32s2-elf-gcc' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���

'riscv32-esp-elf-gcc' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Command failed: riscv32-esp-elf-gcc --version
'riscv32-esp-elf-gcc' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Error checking riscv32-esp-elf version
Error: Command failed: riscv32-esp-elf-gcc --version
'riscv32-esp-elf-gcc' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���

'ninja' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Command failed: ninja --version
'ninja' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Error checking ninja version
Error: Command failed: ninja --version
'ninja' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���

'openocd' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Command failed: openocd --version
'openocd' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Error checking openocd-esp32 version
Error: Command failed: openocd --version
'openocd' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���

'ccache.exe' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Command failed: ccache.exe --version
'ccache.exe' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Error checking ccache version
Error: Command failed: ccache.exe --version
'ccache.exe' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���

'esp32s2ulp-elf-as' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Command failed: esp32s2ulp-elf-as --version
'esp32s2ulp-elf-as' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Error checking esp32s2ulp-elf version
Error: Command failed: esp32s2ulp-elf-as --version
'esp32s2ulp-elf-as' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���

'esp32ulp-elf-as' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Command failed: esp32ulp-elf-as --version
'esp32ulp-elf-as' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Error checking esp32ulp-elf version
Error: Command failed: esp32ulp-elf-as --version
'esp32ulp-elf-as' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���

'cmake' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Command failed: cmake --version
'cmake' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Error checking cmake version
Error: Command failed: cmake --version
'cmake' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���

'dfu-util' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Command failed: dfu-util --version
'dfu-util' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Error checking dfu-util version
Error: Command failed: dfu-util --version
'dfu-util' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���

'idf.py.exe' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Command failed: idf.py.exe -v
'idf.py.exe' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Error checking idf-exe version
Error: Command failed: idf.py.exe -v
'idf.py.exe' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���

idf_versions.txt has (1) KB
idf_versions.txt progress: 100.00% (0.10 / 0.10) KB
'xtensa-esp32s2-elf-gcc' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Command failed: xtensa-esp32s2-elf-gcc --version
'xtensa-esp32s2-elf-gcc' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Error checking xtensa-esp32s2-elf version
Error: Command failed: xtensa-esp32s2-elf-gcc --version
'xtensa-esp32s2-elf-gcc' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���

'xtensa-esp32-elf-gcc' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Command failed: xtensa-esp32-elf-gcc --version
'xtensa-esp32-elf-gcc' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Error checking xtensa-esp32-elf version
Error: Command failed: xtensa-esp32-elf-gcc --version
'xtensa-esp32-elf-gcc' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���

'esp32ulp-elf-as' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Command failed: esp32ulp-elf-as --version
'esp32ulp-elf-as' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Error checking esp32ulp-elf version
Error: Command failed: esp32ulp-elf-as --version
'esp32ulp-elf-as' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���

'esp32s2ulp-elf-as' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Command failed: esp32s2ulp-elf-as --version
'esp32s2ulp-elf-as' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Error checking esp32s2ulp-elf version
Error: Command failed: esp32s2ulp-elf-as --version
'esp32s2ulp-elf-as' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���

'openocd' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Command failed: openocd --version
'openocd' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Error checking openocd-esp32 version
Error: Command failed: openocd --version
'openocd' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���

'idf.py.exe' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Command failed: idf.py.exe -v
'idf.py.exe' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Error checking idf-exe version
Error: Command failed: idf.py.exe -v
'idf.py.exe' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���

'cmake' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Command failed: cmake --version
'cmake' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Error checking cmake version
Error: Command failed: cmake --version
'cmake' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���

'ccache.exe' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Command failed: ccache.exe --version
'ccache.exe' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Error checking ccache version
Error: Command failed: ccache.exe --version
'ccache.exe' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���

'xtensa-esp32s3-elf-gcc' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Command failed: xtensa-esp32s3-elf-gcc --version
'xtensa-esp32s3-elf-gcc' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Error checking xtensa-esp32s3-elf version
Error: Command failed: xtensa-esp32s3-elf-gcc --version
'xtensa-esp32s3-elf-gcc' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���

'riscv32-esp-elf-gcc' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Command failed: riscv32-esp-elf-gcc --version
'riscv32-esp-elf-gcc' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Error checking riscv32-esp-elf version
Error: Command failed: riscv32-esp-elf-gcc --version
'riscv32-esp-elf-gcc' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���

'ninja' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Command failed: ninja --version
'ninja' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Error checking ninja version
Error: Command failed: ninja --version
'ninja' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���

'dfu-util' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Command failed: dfu-util --version
'dfu-util' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���
Error checking dfu-util version
Error: Command failed: dfu-util --version
'dfu-util' �����ڲ����ⲿ���Ҳ���ǿ����еĳ���
���������ļ���

johnnyguoyy, 2021-08-06:
Thanks Brian, I'll try.

brianignacio5, 2021-08-17:
Closing this issue from lack of response. Please comment here if you still face any issues.

closed 2021-08-17 by brianignacio5