espressif/vscode-esp-idf-extension #490: [Bug Report]: Ninja not found

MarkusvonStaden, 2021-08-06:
**Describe the bug**
I installed the extension on a fresh install of Windows 10 on two machines.
When building I get the following error on both machines
`
> Executing task: cmake --build . <

ninja: error: loading 'build.ninja': the system cannot find the file specified.

The terminal process "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command cmake --build ." terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
`

**To Reproduce**
I installed the extension from the vscode marketplace. Then i chose express install. No errors showed up.

**Expected behavior**
Building and not getting an error.

**Environment (please complete the following information):**

- OS Name & Version: Windows 10  10.0.19043 Build 19043
- VSCode Version: 1.59.0
- ESP-IDF Version: 4.3
- Python Version: 3.8.7 (installed by esp-idf extension)

Doctor Command:
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19043 
System environment variable PATH 
 C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Users\marku\AppData\Local\Microsoft\WindowsApps;;C:\Users\marku\AppData\Local\Programs\Microsoft VS Code\bin 
Visual Studio Code version 1.59.0 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.1.1 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:\Users\marku\esp\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
Custom extra paths (idf.customExtraPaths) C:\Users\marku\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin;C:\Users\marku\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin;C:\Users\marku\.espressif\tools\xtensa-esp32s3-elf\esp-2020r3-8.4.0\xtensa-esp32s3-elf\bin;C:\Users\marku\.espressif\tools\riscv32-esp-elf\1.24.0.123_64eb9ff-8.4.0\riscv32-esp-elf\bin;C:\Users\marku\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Users\marku\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Users\marku\.espressif\tools\cmake\3.16.4\bin;C:\Users\marku\.espressif\tools\openocd-esp32\v0.10.0-esp32-20210401\openocd-esp32\bin;C:\Users\marku\.espressif\tools\ninja\1.10.2;C:\Users\marku\.espressif\tools\idf-exe\1.0.1;C:\Users\marku\.espressif\tools\ccache\3.7;C:\Users\marku\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64;C:\Users\marku\.espressif\tools\idf-python-wheels\idf4.3_py3.9_2021-01-07
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"C:\\Users\\marku\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20210401/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) C:\Users\marku\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe
Serial port (idf.port) COM1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) C:\Users\marku\.espressif
Git Path (idf.gitPath) C:\Users\marku\.espressif\tools\idf-git\2.30.1\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espIdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espIdfPath) false
Access to ESP-IDF Custom extra paths
Access to C:\Users\marku\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin: true
Access to C:\Users\marku\.espressif\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin: true
Access to C:\Users\marku\.espressif\tools\xtensa-esp32s3-elf\esp-2020r3-8.4.0\xtensa-esp32s3-elf\bin: true
Access to C:\Users\marku\.espressif\tools\riscv32-esp-elf\1.24.0.123_64eb9ff-8.4.0\riscv32-esp-elf\bin: true
Access to C:\Users\marku\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin: true
Access to C:\Users\marku\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin: true
Access to C:\Users\marku\.espressif\tools\cmake\3.16.4\bin: true
Access to C:\Users\marku\.espressif\tools\openocd-esp32\v0.10.0-esp32-20210401\openocd-esp32\bin: true
Access to C:\Users\marku\.espressif\tools\ninja\1.10.2: true
Access to C:\Users\marku\.espressif\tools\idf-exe\1.0.1: true
Access to C:\Users\marku\.espressif\tools\ccache\3.7: true
Access to C:\Users\marku\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to C:\Users\marku\.espressif\tools\idf-python-wheels\idf4.3_py3.9_2021-01-07: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) true
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version 2.30.1.windows.1
[... 80 lines cut ...]
            "cStandard": "c11",
            "cppStandard": "c++17",
            "includePath": [
                "${config:idf.espIdfPath}/components/**",
                "${config:idf.espIdfPathWin}/components/**",
                "${config:idf.espAdfPath}/components/**",
                "${config:idf.espAdfPathWin}/components/**",
                "${workspaceFolder}/**"
            ],
            "browse": {
                "path": [
                    "${config:idf.espIdfPath}/components",
                    "${config:idf.espIdfPathWin}/components",
                    "${config:idf.espAdfPath}/components/**",
                    "${config:idf.espAdfPathWin}/components/**",
                    "${workspaceFolder}"
                ],
                "limitSymbolsToIncludedHeaders": false
            }
        }
    ],
    "version": 4
}
 
--------------------------------------------------------------------------------------------------------------------------------------------

MarkusvonStaden, 2021-08-07:
Update:
It seems like the installer from the extension does not install esp-idf right.
I removed the extension and the leftovers from the idf installation and installed the idf manually, in the exact location where the extension installed it. 
Then I installed the extension again and that worked for me.
But that is a very annoing way to setup esp-idf for VSCode.

closed 2021-08-07 by MarkusvonStaden

brianignacio5, 2021-08-09:
The extension does install ESP-IDF and ESP-IDF Tools, maybe something went wrong while installing the tools or something. If you still have the log or the error message we could see what went wrong and hopefully fix any issues.

Here is the [troubleshooting guide](https://github.com/espressif/vscode-esp-idf-extension#troubleshooting) which specifies where errors might be.

dobairoland, 2021-08-10:
@brianignacio5 This is another ninja (s3 toolchain) download issue from VSCODE: https://github.com/espressif/esp-idf/issues/7391

I don't know if they are related but according to the report, download got stuck from Vscode but was successful directly from Windows tool installer.

cc @UnoCube