espressif/vscode-esp-idf-extension #494: [Bug Report]: EPS32-C3 "Failed to init Debug Adapter"

SilentButeo2, 2021-08-13:
I try to debug an esp32c3 unit with the integrated serail/jtag functionality over usb.
Building and running with "Build, flash and start monitor on your device" seems to work fine. (using the default hello_world example)
But if I want to debug with F5, I get "Failed to init Debug Adapter".
Don't know if the default launch.json is the correct one for debugging the esp32c3.
I came across several other launch.json files but some of them stated there were out dated.

**HW:**
Using a ESP32-C3-DevKitM-1. Changed it to use the usb serial/jtag functionality.
Running 

**Expected behavior**
Possibility to step through the code

- OS Name & Version :Win10 20H2 (19042.985)
- VSCode Version: 1.59.0
- Python Version: 3.8.7
- ESP-IDF Version: 4.4 (current master build)
- ESP-IDF Extension version 1.1.1

**Logs**
debug.log:
```
2021-08-13 11:16:51,210 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STARTED
2021-08-13 11:16:51,215 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT
2021-08-13 11:16:51,278 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STOPPED
```

**Settings**
launch.json
```
{
  "version": "0.2.0",
  "configurations": [
    {
      "type": "espidf",
      "name": "Launch",
      "request": "launch"
    }
  ]
}
```

brianignacio5, 2021-08-16:
Use the `ESP-IDF: Doctor command` to generate a report of your configuration.

**Output**

Click the menu `View` and select `Output`. Check the output channels: ESP-IDF, ESP-IDF Debug Adapter, OpenOCD

Please share the output of the channel related to your issue. For example ESP-IDF is used mostly for the extension setup process.

**Logs**
If applicable, please share the log file which can be obtained from

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

SilentButeo2, 2021-08-17:
***ESP-IDF: Doctor command***
```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19042 
System environment variable PATH 
 C:\Program Files\AdoptOpenJDK\jdk-11.0.10.9-hotspot\bin;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Program Files\Microsoft MPI\Bin\;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\nodejs\;C:\Program Files (x86)\Microsoft SQL Server\150\DTS\Binn\;C:\Program Files (x86)\Microsoft SQL Server\140\Tools\Binn\;C:\Program Files\Microsoft SQL Server\140\Tools\Binn\;C:\Program Files (x86)\Microsoft SQL Server\140\DTS\Binn\;C:\Program Files\Microsoft SQL Server\140\DTS\Binn\;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\130\Tools\Binn\;C:\Program Files\PuTTY\;C:\Program Files\dotnet\;C:\Program Files (x86)\Pico Technology\PicoScope6\;C:\Program Files (x86)\Perforce;C:\Program Files (x86)\dotnet\;C:\Program Files\Git\cmd;C:\Program Files\TortoiseGit\bin;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\Docker\Docker\resources\bin;C:\ProgramData\DockerDesktop\version-bin;D:\progs\Ruby27-x64\bin;C:\Users\demouser\AppData\Local\Programs\Python\Python39\Scripts\;C:\Users\demouser\AppData\Local\Programs\Python\Python39\;C:\Users\demouser\AppData\Local\Microsoft\WindowsApps;C:\Users\demouser\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\demouser\AppData\Roaming\npm;P:\scripting\pc-lint\bin;C:\Users\demouser\.dotnet\tools;C:\Users\demouser\AppData\Local\Pandoc\;D:\progs\cygwin64\bin; 
Visual Studio Code version 1.59.0 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.1.1 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) D:/progs/esp/master/esp-idf/
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
Custom extra paths (idf.customExtraPaths) C:\Users\demouser\.espressif\tools\xtensa-esp32-elf\esp-2021r1-8.4.0\xtensa-esp32-elf\bin;C:\Users\demouser\.espressif\tools\xtensa-esp32s2-elf\esp-2021r1-8.4.0\xtensa-esp32s2-elf\bin;C:\Users\demouser\.espressif\tools\xtensa-esp32s3-elf\esp-2021r1-8.4.0\xtensa-esp32s3-elf\bin;C:\Users\demouser\.espressif\tools\riscv32-esp-elf\esp-2021r1-8.4.0\riscv32-esp-elf\bin;C:\Users\demouser\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Users\demouser\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Users\demouser\.espressif\tools\cmake\3.20.3\bin;C:\Users\demouser\.espressif\tools\openocd-esp32\v0.10.0-esp32-20210721\openocd-esp32\bin;C:\Users\demouser\.espressif\tools\ninja\1.10.2;C:\Users\demouser\.espressif\tools\idf-exe\1.0.1;C:\Users\demouser\.espressif\tools\ccache\4.3\ccache-4.3-windows-64;C:\Users\demouser\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"C:\\Users\\demouser\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20210721/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) C:\Users\demouser\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe
Serial port (idf.port) COM30
OpenOCD Configs (idf.openOcdConfigs) board/esp32c3-builtin.cfg
ESP-IDF Tools Path (idf.toolsPath) C:\Users\demouser\.espressif
Git Path (idf.gitPath) C:/Users/demouser/.espressif/tools/idf-git/2.30.1/cmd/git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espIdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espIdfPath) false
Access to ESP-IDF Custom extra paths
Access to C:\Users\demouser\.espressif\tools\xtensa-esp32-elf\esp-2021r1-8.4.0\xtensa-esp32-elf\bin: true
Access to C:\Users\demouser\.espressif\tools\xtensa-esp32s2-elf\esp-2021r1-8.4.0\xtensa-esp32s2-elf\bin: true
Access to C:\Users\demouser\.espressif\tools\xtensa-esp32s3-elf\esp-2021r1-8.4.0\xtensa-esp32s3-elf\bin: true
Access to C:\Users\demouser\.espressif\tools\riscv32-esp-elf\esp-2021r1-8.4.0\riscv32-esp-elf\bin: true
Access to C:\Users\demouser\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin: true
Access to C:\Users\demouser\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin: true
Access to C:\Users\demouser\.espressif\tools\cmake\3.20.3\bin: true
Access to C:\Users\demouser\.espressif\tools\openocd-esp32\v0.10.0-esp32-20210721\openocd-esp32\bin: true
Access to C:\Users\demouser\.espressif\tools\ninja\1.10.2: true
Access to C:\Users\demouser\.espressif\tools\idf-exe\1.0.1: true
Access to C:\Users\demouser\.espressif\tools\ccache\4.3\ccache-4.3-windows-64: true
Access to C:\Users\demouser\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64: true
[... cut ...] 99Z"}
{"user":true,"message":"OpenOCD Exit with non-zero error code 3221225477","stack":"Error: Spawn exit with non-zero3221225477\n\tat ChildProcess.<anonymous> (c:\\Users\\demouser\\.vscode\\extensions\\espressif.esp-idf-extension-1.1.1\\dist\\extension.js:1:81919)\n\tat ChildProcess.emit (events.js:315:20)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-08-17T07:54:01.576Z"}
```

gerekon, 2021-08-24:
Hi @SilentButeo2 
Have you connected separate USB cable to USB D+/- pins of the chip as described [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/usb-serial-jtag-console.html#hardware-requirements)?

pdecostervgls, 2021-08-24:
I got the same issue, although not with a DevKit but a custom build board. It's connected via a Tag-Connect (10 pin) to a Segger J-Link over USB.

gerekon, 2021-08-24:
@pdecostervgls If you want to use external Debug Adapter, please follow this steps https://github.com/espressif/openocd-esp32/issues/165#issuecomment-878397446

pdecostervgls, 2021-08-24:
@gerekon Are you referring to burning the (e)fuses? I am using a prototype board made by our electrical engineer, and I believe its a ESP32, but he's on vacation and I'm not sure i'd be comfortable with doing something irreversible.

Mind you, the programming and debugging used to work a few weeks ago with Eclipse, so I am not sure if its neccesary

SilentButeo2, 2021-08-24:
Yes, I've changed the 0 ohm resistors to switch from UART to GPIO18/19.

Currently I've switch to the Eclipse debug setup and there, I have made (after a lot of trial and error) this to work (not yet stable, but it seems to work)

gerekon, 2021-08-24:
> Yes, I've changed the 0 ohm resistors to switch from UART to GPIO18/19.

@SilentButeo2 Do I understand correctly that you use USB Serial/JTAG Controller Console instead of the default chip's HW UART? And flashing via USB Serial/JTAG Controller Console works ok?

BTW, in order to narrow down possible problem's causes. Can I ask you to reproduce the problem w/o VsCode?
1. Run OpenOCD `openocd -f board/esp32c3-builtin.cfg`
2. Create `gdbinit` file with the following contents
```
target remote :3333
mon reset halt
flushregs
b app_main
c
```
3. Run GDB `riscv32-esp-elf-gdb -x gdbinit build/blink.elf`
4. You should stop at `app_main`

gerekon, 2021-08-24:
>  Tag-Connect (10 pin) to a Segger J-Link over USB.
> and I believe its a ESP32

@pdecostervgls According to your message you use SEGGER JLink as debug adapter with ESP32 (not ESP32-C3). Am I right? Here we are talking about ESP32-C3 builltin JTAG adapter. So I do not understand how you can see the problem described at the beginning of the topic.

gerekon, 2021-08-24:
@pdecostervgls If you have a problem with ESP32 debugging you can create separate ticket in OpenOCD repo. https://github.com/espressif/openocd-esp32