espressif/vscode-esp-idf-extension #495: panic_abort for ESP IDF 4.3 master with example : esp-idf-release-v4.3_master\examples\peripherals\i2s\i2s_adc_dac

nitinbhosaleemb, 2021-08-16:
Executing action: monitor
Running idf_monitor in directory c:\users\e0636298\documents\git\hybrid_test_esp
32\simpleone
Executing "C:\Users\E0636298\.espressif\python_env\idf4.3_py3.9_env\Scripts\pyth
on.exe C:\Users\E0636298\Documents\esp-idf-v4.3\tools/idf_monitor.py -p COM6 -b
115200 --toolchain-prefix xtensa-esp32-elf- --print_filter  c:\users\e0636298\do
cuments\git\hybrid_test_esp32\simpleone\build\simpleone.elf -m 'C:\Users\E063629
8\.espressif\python_env\idf4.3_py3.9_env\Scripts\python.exe' 'C:\Users\E0636298\
Documents\esp-idf-v4.3\tools\idf.py'"...
?[0;33m--- WARNING: GDB cannot open serial ports accessed as COMx?[0m
?[0;33m--- Using \\.\COM6 instead...?[0m
--- idf_monitor on \\.\COM6 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7084
ho 0 tail 12 room 4
load:0x40078000,len:14292
load:0x40080400,len:3688
0x40080400: _init at ??:?

entry 0x40080678
I (30) boot: ESP-IDF v4.3-dirty 2nd stage bootloader
I (30) boot: compile time 08:37:38
I (30) boot: chip revision: 3
I (33) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 2MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (97) boot_comm: chip revision: 3, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=09a08h ( 39432)
 map
I (127) esp_image: segment 1: paddr=00019a30 vaddr=3ffb0000 size=0227ch (  8828)
 load
I (131) esp_image: segment 2: paddr=0001bcb4 vaddr=40080000 size=04364h ( 17252)
 load
I (140) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=188d8h (100568)
 map
I (179) esp_image: segment 4: paddr=00038900 vaddr=40084364 size=08de8h ( 36328)
 load
I (195) esp_image: segment 5: paddr=000416f0 vaddr=50000000 size=00010h (    16)
 load
I (203) boot: Loaded app from partition at offset 0x10000
I (203) boot: Disabling RNG early entropy source...
I (215) cpu_start: Pro cpu up.
I (215) cpu_start: Single core mode
I (224) cpu_start: Pro cpu start user code
I (224) cpu_start: cpu freq: 160000000
I (224) cpu_start: Application information:
I (228) cpu_start: Project name:     simpleone
I (233) cpu_start: App version:      6b28b94-dirty
I (239) cpu_start: Compile time:     Aug 16 2021 11:30:02
I (245) cpu_start: ELF file SHA256:  d9197ab2ffb89157...
I (251) cpu_start: ESP-IDF:          v4.3-dirty
I (256) heap_init: Initializing. RAM available for dynamic allocation:
I (263) heap_init: At 3FF80000 len 00002000 (8 KiB): RTCRAM
I (270) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (276) heap_init: At 3FFB2F18 len 0002D0E8 (180 KiB): DRAM
I (282) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM
[... 147 lines cut ...]
I (294) heap_init: At 4008D14C len 00012EB4 (75 KiB): IRAM
I (301) spi_flash: detected chip: generic
I (305) spi_flash: flash io: dio
W (309) spi_flash: Detected size(4096k) larger than the size in the binary image
 header(2048k). Using the size in the binary image header.
I (323) cpu_start: Starting scheduler on PRO CPU.
I (328) gpio: GPIO[21]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldo
wn: 0| Intr:1
I (328) gpio: GPIO[22]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldo
wn: 0| Intr:1
I (338) gpio: GPIO[23]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldo
wn: 0| Intr:1
eFuse Two Point: NOT supported
eFuse Vref: Supported
I (358) i2c_bus: i2c1 bus inited
device ID is: bc
I (368) spi_bus: SPI2 bus created
I (368) spi_bus: SPI2 bus device added, CS=10 Mode=0 Speed=10000000
in=0
in=0
*******************
TASK: MASTER READ SENSOR(HTS221 )
*******************
humidity: 53.700001
temperature: 21.799999

nitinbhosaleemb, 2021-08-16:
Link for 4.0 code which works for esp idf 4.2 : https://www.esp32.com/viewtopic.php?t=2346#p37378

the multi channel code with i2s adc crashes with esp idf 4.3

brianignacio5, 2021-08-17:
I don't think this is the right repository for your issue. 

Could re-open it in the [ESP-IDF Repository](https://github.com/espressif/esp-idf/issues) ?

closed 2021-08-17 by brianignacio5