espressif/vscode-esp-idf-extension #496: [Bug Report]: Can't find git in WSL2

Flordsky, 2021-08-17:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [ ] Make sure you have searched for existing bugs and features request before you post an issue.
- [ ] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [ ] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
A clear and concise description of what the bug is (current behavior).
The extension can't find git in WSL2.It can't build any example when set path by self,the example could build by terminal idf.py.

**To Reproduce**
Steps to reproduce the behavior (including the full log).

**Expected behavior**
A clear and concise description of what you expected to happen.
just work normal

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]
[image]

**Environment (please complete the following information):**
- OS Name & Version: windows19043.1165  WSL2 Ubuntu20.04
- VSCode Version: 1.59.0
- ESP-IDF Version: ESP-IDF master
- Python Version: 3.8.10
- Git Version:2.33.0
- Cmake Version:3.16.3

> **NOTE**: You can use the `ESP-IDF: Doctor command` to generate a report of your configuration.

**Output**
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS linux x64 5.10.16.3-microsoft-standard-WSL2 
System environment variable PATH 
 /home/flordsky/.vscode-server/bin/379476f0e13988d90fab105c5c19e7abc8b1dea8/bin:/home/flordsky/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/mnt/d/Env_software_WJ/Python/Scripts/:/mnt/d/Env_software_WJ/Python/:/mnt/d/Env_software_WJ/VMXuNiJi/bin/:/mnt/c/WINDOWS/system32:/mnt/c/WINDOWS:/mnt/c/WINDOWS/System32/Wbem:/mnt/c/WINDOWS/System32/WindowsPowerShell/v1.0/:/mnt/c/WINDOWS/System32/OpenSSH/:/mnt/d/Env_software_WJ/ebook/:/mnt/d/Env_software_WJ/Matlab/runtime/win64:/mnt/d/Env_software_WJ/Matlab/bin:/mnt/d/Env_software_WJ/Matlab/polyspace/bin:/mnt/d/Program Files/Git/cmd:/mnt/c/Users/Administrator/AppData/Local/Microsoft/WindowsApps:/mnt/d/Env_software_WJ/Microsoft VS Code/bin:/snap/bin 
Visual Studio Code version 1.59.0 
Visual Studio Code language zh-cn 
Visual Studio Code shell bash 
ESP-IDF Extension version 1.1.1 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath)  /home/flordsky/ESP/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
Custom extra paths (idf.customExtraPaths) 
Custom extra vars (idf.customExtraVars) 
Virtual env Python Path (idf.pythonBinPath) /home/flordsky/.espressif/python_env/idf4.4_py3.8_env/bin/python
Serial port (idf.port) /dev/ttyUSB1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) /home/flordsky/.espressif
Git Path (idf.gitPath) git
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espIdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) false
Access to ESP-MDF Path (idf.espIdfPath) false
Access to ESP-IDF Custom extra paths
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH true
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) true
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version undefined
ESP-IDF version undefined
Python version undefined
Python's pip version undefined
-------------------------------------------------- Python packages in idf.pythonBinPath ----------------------------------------------------
---------------------------------------------------- Check ESP-IDF python requirements.txt -------------------------------------------------
Check ESP-IDF Python packages undefined
---------------------------------------------------- Check extension requirements.txt ------------------------------------------------------
Check Extension Python packages undefined
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages undefined
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at Command failed: which ninja

--------------------------------------------------------------------------------------------------------------------------------------------

Click the menu `View` and select `Output`. This extension uses several output channels: ESP-IDF, ESP-IDF Debug Adapter, Heap Trace, OpenOCD and SDK Configuration Editor.

Please share the output of the channel related to your issue. For example ESP-IDF is used mostly for the extension setup process.

**Logs**
If applicable, please share the log file which can be obtained from

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

**Extension Configuration settings**
Open Command Palette (F1 or View Menu -> Command Palette) and type `Preferences: Open Settings (JSON)` or use `{CURRENT_PROJECT}/.vscode/settings.json` to get the following extension configuration settings value:

- `idf.espIdfPath` (MacOS or Linux) or `idf.espIdfPathWin` (Windows) also called `Path to locate ESP-IDF framework (IDF_PATH)` in `Preferences: Open Settings (UI)`.
- `idf.pythonBinPath` (MacOS or Linux) or `idf.pythonBinPathWin` (Windows) also called `Python absolute binary path used to execute ESP-IDF Python Scripts` in `Preferences: Open Settings (UI)`.
- `idf.openOcdConfigs` also called `List of configuration files inside OpenOCD Scripts directory` in `Preferences: Open Settings (UI)`.
- `idf.customExtraPaths` also called `Paths to be appended to PATH` in `Preferences: Open Settings (UI)`.
- `idf.customExtraVars` also called `Variables to be added to system environment variables` in `Preferences: Open Settings (UI)`.
- `idf.adapterTargetName` also called `Target name for ESP-IDF Debug Adapter` in `Preferences: Open Settings (UI)`.
- `idf.customAdapterTargetName` is used when `idf.adapterTargetName` is set to `custom`.

**Additional context**
Add any other context about the problem here.

brianignacio5, 2021-08-19:
As mentioned in the prerequisites section in the README.md, have you looked into [WSL docs](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/WSL.md) contain a set of packages to install on the WSL bistro before using this extension.

Seems that ninja-build is not available as shown in doctor command output.

brianignacio5, 2021-08-23:
Closing this issue for lack of activity. Please reply here if you need more assistance.

closed 2021-08-23 by brianignacio5