espressif/vscode-esp-idf-extension #497: [Bug Report]: Install

hdpklm, 2021-08-19:
esp-idf-v4.3.zip progress: 99.97% (637966.46 / 638189.50) KB
esp-idf-v4.3.zip progress: 99.97% (637982.46 / 638189.50) KB
esp-idf-v4.3.zip progress: 99.97% (637998.46 / 638189.50) KB
esp-idf-v4.3.zip progress: 99.97% (638014.46 / 638189.50) KB
esp-idf-v4.3.zip progress: 99.98% (638030.46 / 638189.50) KB
esp-idf-v4.3.zip progress: 99.98% (638046.46 / 638189.50) KB
esp-idf-v4.3.zip progress: 99.98% (638062.46 / 638189.50) KB
esp-idf-v4.3.zip progress: 99.98% (638078.46 / 638189.50) KB
esp-idf-v4.3.zip progress: 99.99% (638094.46 / 638189.50) KB
esp-idf-v4.3.zip progress: 99.99% (638110.46 / 638189.50) KB
esp-idf-v4.3.zip progress: 99.99% (638126.46 / 638189.50) KB
esp-idf-v4.3.zip progress: 99.99% (638142.46 / 638189.50) KB
esp-idf-v4.3.zip progress: 100.00% (638158.46 / 638189.50) KB
esp-idf-v4.3.zip progress: 100.00% (638174.46 / 638189.50) KB
esp-idf-v4.3.zip progress: 100.00% (638189.50 / 638189.50) KB
Downloaded v4.3 (release version).

Extracted C:\Users\Hassan Daoud\esp3\esp-idf-v4.3.zip in C:\Users\Hassan Daoud\esp3.

Invalid ESP-IDF

brianignacio5, 2021-08-20:
Can you try to install in a directory without spaces ?

Also please follow the bug report template. Without the logs and doctor command output is hard to reproduce your issue.

hdpklm, 2021-08-20:
ok i try to install in a path without spaces. 
but it show me the problem
[image]

hdpklm, 2021-08-20:
is still not work 😢😢

[image]

esp-idf-v4.3.zip progress: 99.99% (638099.31 / 638189.50) KB
esp-idf-v4.3.zip progress: 99.99% (638115.31 / 638189.50) KB
esp-idf-v4.3.zip progress: 99.99% (638131.31 / 638189.50) KB
esp-idf-v4.3.zip progress: 99.99% (638147.31 / 638189.50) KB
esp-idf-v4.3.zip progress: 100.00% (638163.31 / 638189.50) KB
esp-idf-v4.3.zip progress: 100.00% (638179.31 / 638189.50) KB
esp-idf-v4.3.zip progress: 100.00% (638189.50 / 638189.50) KB
Downloaded v4.3 (release version).

Extracted c:\hassan\programs\esp\esp-idf-v4.3.zip in c:\hassan\programs\esp.

Invalid ESP-IDF

hdpklm, 2021-08-20:
i cant use the bug report template, because is not instaled and the must information needed is from configuratoin
but i find the log file

[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/7023358/esp_idf_vsc_ext.log)

brianignacio5, 2021-08-23:
Based on the logs you need to specify the IDF_TOOLS_PATH in place without spaces too.

In the setup, there is a field called `Enter ESP-IDF Tools directory (IDF_TOOLS_PATH)`.

hdpklm, 2021-08-23:
> Based on the logs you need to specify the IDF_TOOLS_PATH in place without spaces too.
> 
> In the setup, there is a field called `Enter ESP-IDF Tools directory (IDF_TOOLS_PATH)`.

ok thank you very much. its installed correctly 🌹❤
i think you must specify in the installation note to enter path without spaces and/or validate input box if has spaces
thank you a lot. have a nice day

closed 2021-08-24 by brianignacio5