espressif/vscode-esp-idf-extension #499: [Bug Report]:

jvanlangen, 2021-08-22:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [X] Make sure you have searched for existing bugs and features request before you post an issue.
- [X] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [X] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
After installing VS code and installing the ESP-IDF extension, i'm not able to execute the "idf.py menuconfig". not by prefixing pthon. entering the full path of python etc

- Windows 10
- VSCode Version: 1.59.1
- ESP-IDF Version: v1.1.1
- Python Version: what is installed with it

The rest of the requirements I don't understand

brianignacio5, 2021-08-23:
Have you run the setup within the extension ? press F1 and type `ESP-IDF: Configure ESP-IDF extension` in the Command Palette to open the extension configuration wizard. This will install ESP-IDF and tools and configure the extension.

jvanlangen, 2021-08-23:
Yes, i've done it. What I didn't knew is, i'm using a ESP32-A1S. And I think the problem is that the I2c is not working correctly:

_I (377) PLAY_SDCARD_MUSIC: [ 1 ] Mount sdcard
I (887) PLAY_SDCARD_MUSIC: [ 2 ] Start codec chip
E (887) gpio: gpio_install_isr_service(460): GPIO isr service already installed
E (1277) I2C_BUS: C:/Users/jvanl/.espressif/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error
---- snip similar errors
E (1277) I2C_BUS: C:/Users/jvanl/.espressif/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:101 (i2c_bus_write_bytes):I2C Bus WriteReg Error
E (1287) AUDIO_HAL: codec init failed!
E (1287) AUDIO_BOARD: C:/Users/jvanl/.espressif/esp-adf/components/audio_board/lyrat_v4_3/board.c:55 (audio_board_codec_init): Got NULL Pointer
E (1307) AUDIO_HAL: audio_hal handle is null
I (1307) PLAY_SDCARD_MUSIC: [3.0] Create audio pipeline for playback
I (1317) PLAY_SDCARD_MUSIC: [3.1] Create fatfs stream to read data from sdcard
I (1327) PLAY_SDCARD_MUSIC: [3.2] Create i2s stream to write data to codec chip
I (1357) PLAY_SDCARD_MUSIC: [3.3] Create mp3 decoder
I (1357) PLAY_SDCARD_MUSIC: [3.4] Register all elements to audio pipeline
I (1357) PLAY_SDCARD_MUSIC: [3.5] Link it together [sdcard]-->fatfs_stream-->music_decoder-->i2s_stream-->[codec_chip]
I (1367) PLAY_SDCARD_MUSIC: [3.6] Set up uri: /sdcard/test.mp3 
I (1377) PLAY_SDCARD_MUSIC: [ 4 ] Set up  event listener
I (1387) PLAY_SDCARD_MUSIC: [4.1] Listening event from all elements of pipeline
I (1387) PLAY_SDCARD_MUSIC: [4.2] Listening event from peripherals
I (1397) PLAY_SDCARD_MUSIC: [ 5 ] Start audio_pipeline
I (1407) PLAY_SDCARD_MUSIC: [ 6 ] Listen for all pipeline events
I (1417) PLAY_SDCARD_MUSIC: [ * ] Receive music info from mp3 decoder, sample_rates=44100, bits=16, ch=2_

Any thoughts?

brianignacio5, 2021-08-23:
If you figured the extension install I think we can close this issue.

For the i2C issue, maybe check the [ESP-ADF](https://github.com/espressif/esp-adf/issues) repository for help with `i2c_bus` peripheral

closed 2021-08-23 by brianignacio5