espressif/vscode-esp-idf-extension #501: [Bug Report]: can't build ESP-AT project, is it a bug of toolchain?

xiongyu0523, 2021-08-25:
After install vscode extension and ESP-IDF 4.3, I can successfully build hello world project and flash to my target. But I failed on build ESP-AT project. Would you please help to check it is a bug of toolchain?

Detailed Error logs when I select target:

```
{ error: [Error: non zero exit code 2 CMake Error at CMakeLists.txt:9 (string): string no output variable specified CMake Error at CMakeLists.txt:13 (string): string begin index: 9 is out of range 0 - 0 CMake Error at CMakeLists.txt:14 (string): string sub-command STRIP requires two arguments. CMake Error at CMakeLists.txt:15 (string): string no output variable specified silence:0 sdkconfig:C:/Users/yuxiong/Documents/GitHub/esp-at/module_config/module__default/sdkconfig.defaults CMake Error at CMakeLists.txt:49 (include): include could not find load file: C:/Users/yuxiong/Documents/GitHub/esp-at/esp-idf/tools/cmake/project.cmake -- The C compiler identification is unknown -- The CXX compiler identification is unknown CMake Error at CMakeLists.txt:51 (project): No CMAKE_C_COMPILER could be found. Tell CMake where to find the compiler by setting either the environment variable "CC" or the CMa...
```

brianignacio5, 2021-08-25:
Not all versions of ESP-IDF are supported in ESP-AT as shown in [here]( https://github.com/espressif/esp-at/blob/master/module_config/module_esp32_default/IDF_VERSION).

My suggestion for you would be to set the ESP-IDF path of the extension (using the setup wizard for example) to the one inside  ESP-AT (`C:/Users/yuxiong/Documents/GitHub/esp-at/esp-idf`). The ESP-IDF downloaded within ESP-AT is downloaded when running the build.py script as shown [here](https://docs.espressif.com/projects/esp-at/en/latest/Compile_and_Develop/How_to_clone_project_and_compile_it.html) which is different from global IDF_PATH as mentioned there.

brianignacio5, 2021-08-30:
1. As mentioned in [ESP-IDF Vscode extension configuration Settings](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/SETTINGS.md#esp-idf-settings) there is a setting called `idf.customExtraVars` which you can use to add additional env variables for the extension tasks.

For example look at [JSON Manual Configuration](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/SETUP.md#json-manual-configuration) showing you how we add OPENOCD_SCRIPTS env variable.

But for your case what you want is the extension to use the IDF_PATH from ESP-AT. What I would suggest you is to use the extension setup to find and use the existing esp-idf in ESP-AT so the extension is configured (means you don't need to run export.ps1 from terminal and just use extension commands). 

2. When your extension is configured you can just run the `ESP-IDF: SDK Configuration editor` command from the extension and the menuconfig is shown. If you want to use a terminal you can use the `ESP-IDF: Open ESP-IDF Terminal` and run the script like `./path/to/build.py menuconfig`

xiongyu0523, 2021-08-31:
Thanks for guide.

closed 2021-08-31 by xiongyu0523