espressif/vscode-esp-idf-extension #502: DOCS remark

goofy2k, 2021-08-25:
In https://github.com/espressif/vscode-esp-idf-extension/tree/2b5ae6ea7aa179a72be16c43adf2900a53deeab0#How-to-use the link to "Working with multiple projects" gives a 404.  See this:  https://github.com/espressif/vscode-esp-idf-extension/blob/master/MULTI_PROJECTS.md

brianignacio5, 2021-08-25:
Thanks for reporting this. Will fix now.

goofy2k, 2021-08-25:
OK. Thanks. For me VS Code and ESP-IDF is pretty complex. Struggling my way
through the docs ;-)

goofy2k, 2021-08-29:
> Yeah we have a lot of documentation. Sorry for that.
>
> I think onboarding documentation
> is a good entry point to follow the tutorials by order for basic
> functionality.
>
> Please let us know any suggestions to make it easier/ improve it.

Thanks for your answer Brian!

I am somewhat further with my endeavors. I now use ESP-IDF from the command
line (using idf.py).

I recognize the clear advantage of starting from examples. I also learned
how to convert, or better, wrap a C code example such that it can be
compiled as C++ that uses external C.  This can be very useful for starters
or for people that are not very into C. I found the procedure over here:

https://www.deeptronic.com/software-design/developing-esp32-application-with-esp-idf-and-c-in-windows

and wrote a summary of my findings here:

https://github.com/goofy2k/ESPIDF-Code-Conversion

Feel free to use it however you like....

Fred

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>