espressif/vscode-esp-idf-extension #506: [Bug Report]: Git not found

krishnaTORQUE, 2021-08-28:
**Describe the bug**
After install the extension I go for Setup IDF & its showing me git is not found.

**To Reproduce**
After install the extension I go for Setup IDF & its showing me git is not found.

**Expected behavior**
It suppose to found git & python3

**Screenshots**
[image]

**Environment (please complete the following information):**

- OS Name & Version:Ubuntu 20.04
- VSCode Version: 1.59.1
- ESP-IDF Version: 1.1.1
- Python Version: 3.8.10

Its a fresh install, so there is not output or log or any other details.

brianignacio5, 2021-08-30:
Do you have git and python3 in your environment PATH ? It is mentioned in [readme prerequisites](https://github.com/espressif/vscode-esp-idf-extension#prerequisites)

Are you sure there is no log in these locations?

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

Logs should be created on any error of the extension.

krishnaTORQUE, 2021-08-30:
I am using ubuntu 20.04 as mention above.
Yes git(2.33.0) is installed installed.
Ubuntu by default comes with python3 installed.

Thanks for the path

Here is the log
```
{"message":"Command failed: git describe --tags\nfatal: not a git repository (or any of the parent directories): .git\n","stack":"Error: Command failed: git describe --tags\nfatal: not a git repository (or any of the parent directories): .git\n\n\tat ChildProcess.exithandler (child_process.js:317:12)\n\tat ChildProcess.emit (events.js:315:20)\n\tat ChildProcess.EventEmitter.emit (domain.js:467:12)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2021-08-28T02:37:05.451Z"}
{"message":"spawn /bin/sh ENOENT","stack":"Error: spawn /bin/sh ENOENT\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:269:19)\n\tat onErrorNT (internal/child_process.js:465:16)\n\tat processTicksAndRejections (internal/process/task_queues.js:80:21)","level":"error","timestamp":"2021-08-28T02:37:16.149Z"}
{"user":true,"message":"Git is not found in current environment","stack":"Error: spawn /bin/sh ENOENT\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:269:19)\n\tat onErrorNT (internal/child_process.js:465:16)\n\tat processTicksAndRejections (internal/process/task_queues.js:80:21)","level":"error","timestamp":"2021-08-28T02:37:16.150Z"}
{"message":"spawn /bin/sh ENOENT","stack":"Error: spawn /bin/sh ENOENT\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:269:19)\n\tat onErrorNT (internal/child_process.js:465:16)\n\tat processTicksAndRejections (internal/process/task_queues.js:80:21)","level":"error","timestamp":"2021-08-28T02:37:16.165Z"}
{"message":"non zero exit code 1\n\n","stack":"Error: non zero exit code 1\n\n\n\tat ChildProcess.<anonymous> (/home/krishna/.vscode/extensions/espressif.esp-idf-extension-1.1.1/dist/extension.js:1:4137)\n\tat ChildProcess.emit (events.js:315:20)\n\tat ChildProcess.EventEmitter.emit (domain.js:467:12)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:277:12)","level":"error","timestamp":"2021-08-28T02:37:16.181Z"}
{"message":"Cannot access filePath: ninja","level":"error","timestamp":"2021-08-28T02:37:16.181Z"}
{"message":"spawn /bin/sh ENOENT","stack":"Error: spawn /bin/sh ENOENT\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:269:19)\n\tat onErrorNT (internal/child_process.js:465:16)\n\tat processTicksAndRejections (internal/process/task_queues.js:80:21)","level":"error","timestamp":"2021-08-28T02:37:17.059Z"}
{"user":true,"message":"Git is not found in current environment","stack":"Error: spawn /bin/sh ENOENT\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:269:19)\n\tat onErrorNT (internal/child_process.js:465:16)\n\tat processTicksAndRejections (internal/process/task_queues.js:80:21)","level":"error","timestamp":"2021-08-28T02:37:17.060Z"}
{"message":"spawn /bin/sh ENOENT","stack":"Error: spawn /bin/sh ENOENT\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:269:19)\n\tat onErrorNT (internal/child_process.js:465:16)\n\tat processTicksAndRejections (internal/process/task_queues.js:80:21)","level":"error","timestamp":"2021-08-28T02:37:17.078Z"}
{"message":"non zero exit code 1\n\n","stack":"Error: non zero exit code 1\n\n\n\tat ChildProcess.<anonymous> (/home/krishna/.vscode/extensions/espressif.esp-idf-extension-1.1.1/dist/extension.js:1:4137)\n\tat ChildProcess.emit (events.js:315:20)\n\tat ChildProcess.EventEmitter.emit (domain.js:467:12)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:277:12)","level":"error","timestamp":"2021-08-28T02:37:17.092Z"}
{"message":"Cannot access filePath: ninja","level":"error","timestamp":"2021-08-28T02:37:17.093Z"}
{"user":true,"message":"Open a folder first.","stack":"Error: PRECHECK_FAILED\n\tat /home/krishna/.vscode/extensions/espressif.esp-idf-extension-1.1.1/dist/extension.js:1:3215\n\tat Array.forEach (<anonymous>)\n\tat Function.perform (/home/krishna/.vscode/extensions/espressif.esp-idf-extension-1.1.1/dist/extension.js:1:3168)\n\tat /home/krishna/.vscode/extensions/espressif.esp-idf-extension-1.1.1/dist/extension.js:8:292631\n\tat /home/krishna/.vscode/extensions/espressif.esp-idf-extension-1.1.1/dist/extension.js:8:268512\n\tat _executeContributedCommand (/snap/code/72/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:85:109251)\n\tat _.$executeContributedCommand (/snap/code/72/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:85:109689)\n\tat h._doInvokeHandler (/snap/code/72/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:89:12819)\n\tat h._invokeHandler (/snap/code/72/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:89:12503)\n\tat h._receiveRequest (/snap/code/72/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:89:11159)\n\tat h._receiveOneMessage (/snap/code/72/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:89:9943)\n\tat /snap/code/72/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/extensionHostProcess.js:89:8039\n\tat d.fire (/snap/code/72/usr/share/code/resources/app/out/vs/workbench/services/extensions/node/ex [... cut ...]
{"message":"non zero exit code 1\n\n","stack":"Error: non zero exit code 1\n\n\n\tat ChildProcess.<anonymous> (/home/krishna/.vscode/extensions/espressif.esp-idf-extension-1.1.1/dist/extension.js:1:4137)\n\tat ChildProcess.emit (events.js:315:20)\n\tat ChildProcess.EventEmitter.emit (domain.js:467:12)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:277:12)","level":"error","timestamp":"2021-08-28T03:00:43.511Z"}
{"message":"Cannot access filePath: ninja","level":"error","timestamp":"2021-08-28T03:00:43.512Z"}

```

krishnaTORQUE, 2021-08-31:
Doctor Commend Report

```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS linux x64 5.4.0-81-generic 
System environment variable PATH 
 /home/krishna/...
Visual Studio Code version 1.59.1 
Visual Studio Code language en 
Visual Studio Code shell bash 
ESP-IDF Extension version 1.1.1 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) ${env:IDF_PATH}
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
Custom extra paths (idf.customExtraPaths) 
Custom extra vars (idf.customExtraVars) 
Virtual env Python Path (idf.pythonBinPath) python
Serial port (idf.port) /dev/ttyUSB1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) ${env:IDF_TOOLS_PATH}
Git Path (idf.gitPath) git
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espIdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) false
Access to ESP-MDF Path (idf.espIdfPath) false
Access to ESP-IDF Custom extra paths
Access to Virtual env Python Path (idf.pythonBinPath) false
Access to CMake in environment PATH true
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) false
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version undefined
ESP-IDF version undefined
Python version undefined
Python's pip version undefined
-------------------------------------------------- Python packages in idf.pythonBinPath ----------------------------------------------------
---------------------------------------------------- Check ESP-IDF python requirements.txt -------------------------------------------------
Check ESP-IDF Python packages undefined
---------------------------------------------------- Check extension requirements.txt ------------------------------------------------------
Check Extension Python packages undefined
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages undefined
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at Command failed: which ninja

--------------------------------------------------------------------------------------------------------------------------------------------

```

brianignacio5, 2021-09-06:
Hey @krishnaTORQUE sorry for late reply.

Based on the doctor command output you are missing ninja-build from your system. It is mentioned in the [readme prerequisites section](https://github.com/espressif/vscode-esp-idf-extension#prerequisites) and in [ESP-IDF](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-setup.html).

For some reason the error message being shown is "git is not found" probably because the error being captured is somehow replaced. We will fix this asap.

krishnaTORQUE, 2021-09-06:
```
sudo apt install ninja-build
```
after install `ninja buid` its working fine

closed 2021-09-06 by krishnaTORQUE