espressif/vscode-esp-idf-extension #508: ESP32-C3-DevKitC-02 flash using Internal JTAG but debug fails (OCD-401)

NicolSpies, 2021-08-30:
### Environment
- Development Kit: ESP32-C3-DevKitC-02
- Module or chip used: ESP32-C3-WROOM-02
- Debug Adapter: ESP32-C3 Internal JTAG
- OpenOCD version: v0.10.0-esp32-20210721
- Operating System: Windows 10
- Using an IDE?: VS Code
- ESP-IDF version: v4.3

### Problem Description
Steps to reproduce the behavior:
1. Build ESP-IDF blink example
2. Flash and verify ESP32-C3 using Internal JTAG
3. Disconnect Internal JTAG and verify blink running using devkit COM port.
4. Disconnect COM port, reconnect internal JTAG and use F5 to start debug.
5. Debug fails with VS Code connection refusal message: `connect ECONNREFUSED 127.0.0.1:43474`

### Debug Logs
OpenOCD log:
```
Open On-Chip Debugger  v0.10.0-esp32-20210721 (2021-07-21-13:35)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : datacount=2 progbufsize=16
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40101104
Info : Listening on port 3333 for gdb connections
```

Terminal log:
```
Windows PowerShell
Copyright (C) Microsoft Corporation. All rights reserved.

Try the new cross-platform PowerShell https://aka.ms/pscore6

PS D:\Dropbox\DATA\DEVELOPMENT\ESP32C3\blink> C:\Users\nicol\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe C:\Users\nicol\esp\esp-idf\tools\idf.py -p COM9 monitor
Executing action: monitor
Running idf_monitor in directory d:\dropbox\data\development\esp32c3\blink
Executing "C:\Users\nicol\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe C:\Users\nicol\esp\esp-idf\tools/idf_monitor.py -p COM9 -b 115200 --toolchain-prefix riscv32-esp-elf- --decode-panic backt
d:\dropbox\data\development\esp32c3\blink\build\blink.elf -m 'C:\Users\nicol\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe' 'C:\Users\nicol\esp\esp-idf\tools\idf.py' '-p' 'COM9'"...
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM9 instead...
--- idf_monitor on \\.\COM9 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x42005a5e
0x42005a5e: cpu_ll_waiti at C:/Users/nicol/esp/esp-idf/components/hal/esp32c3/include/hal/cpu_ll.h:156
 (inlined by) esp_pm_impl_waiti at C:/Users/nicol/esp/esp-idf/components/esp_pm/pm_impl.c:827

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x1778
load:0x403ce000,len:0x8d4
load:0x403d0000,len:0x293c
entry 0x403ce000
I (24) boot: ESP-IDF v4.3-dirty 2nd stage bootloader
I (24) boot: compile time 16:00:58
I (25) boot: chip revision: 3
I (27) boot.esp32c3: SPI Speed      : 80MHz
I (31) boot.esp32c3: SPI Mode       : DIO
I (36) boot.esp32c3: SPI Flash Size : 4MB
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00100000
I (79) boot: End of partition table
I (83) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=05fa8h ( 24488) map
I (96) esp_image: segment 1: paddr=00015fd0 vaddr=3fc89a00 size=01a90h (  6800) load
I (101) esp_image: segment 2: paddr=00017a68 vaddr=40380000 size=085b0h ( 34224) load
I (115) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=15b1ch ( 88860) map
I (131) esp_image: segment 4: paddr=00035b44 vaddr=403885b0 size=013a0h (  5024) load
I (132) esp_image: segment 5: paddr=00036eec vaddr=50000000 size=00010h (    16) load
I (140) boot: Loaded app from partition at offset 0x10000
I (143) boot: Disabling RNG early entropy source...
I (160) cpu_start: Pro cpu up.
I (172) cpu_start: Pro cpu start user code
I (172) cpu_start: cpu freq: 160000000
I (172) cpu_start: Application information:
I (175) cpu_start: Project name:     blink
I (207) cpu_start: App version:      1
I (207) cpu_start: Compile time:     Aug 29 2021 16:59:10
I (208) cpu_start: ELF file SHA256:  f3c5fe14c0f6c604...
I (214) cpu_start: ESP-IDF:          v4.3-dirty
I (219) heap_init: Initializing. RAM available for dynamic allocation:
I (226) heap_init: At 3FC8C2D0 len 00033D30 (207 KiB): DRAM
I (232) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (239) heap_init: At 50000010 len 00001FF0 (7 KiB): RTCRAM
```
Debug launch.json:
```
{
  "version": "0.2.0",
  "configurations": [
    {
      "type": "espidf",
      "name": "Launch",
      "request": "launch",
      "debugPort": 43474,
      "logLevel": 2,
      "mode": "manual",
      "skipVerifyAppBinBeforeDebug": false,
      "initGdbCommands": [
        "target remote :3333",
        "symbol-file D:/Dropbox/DATA/DEVELOPMENT/ESP32C3/blink/build/blink.elf",
        "mon reset halt",
        "flushregs",
        "thb app_main"
      ],
      "env": {
        "CUSTOM_ENV_VAR": "SOME_VALUE"
      }
    }
  ]
}
```
### Expected behavior
Debugging of flashed program.

brianignacio5, 2021-08-30:
Based on your launch.json you are using the manual mode which as stated in [debugging documentation](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/DEBUGGING.md) when you set `"mode": "manual"` means the vscode extension will try to connect to an existing debug adapter server executed by yourself (i.e it will not run it automatically as done in `mode: "auto"`)

NicolSpies, 2021-08-30:
> Hi @NicolSpies
> 
> `connect ECONNREFUSED 127.0.0.1:43474` is related to DebugAdapter script used by VsCode. Looks like it was not terminated after step 3 `Disconnect Internal JTAG and verify blink running using devkit COM port.`. Could you describe what do you do exactly in that step? What does mean `Disconnect Internal JTAG`?
> Also details on step 4 `Disconnect COM port, reconnect internal JTAG and use F5 to start debug.` could also help to diagnose the problem. What does mean `reconnect internal JTAG`?
> 
> cc @brianignacio5

Hi @gerekon, I build successfully, I then successfully flash and verify the ESP32-C3 via the Internal JTAG (wired in on GPIO 18,19, 5V and Gnd of ESP32-C2-DevkitC-02) of the ESP32-C3. I then disconnected this connection and connect the Micro-USB on the devkit and confirm the blink example running on the terminal COM port. I then disconnect the Micro-USB, reconnect the ESP internal JTAG again to debug. I confirm that OpenOCD finds the device and listen on port 3333.
It almost appear as if the localhost IP 127.0.0.1 is being refused and not the port because when I change the port the same happens.

NicolSpies, 2021-08-30:
> Based on your launch.json you are using the manual mode which as stated in [debugging documentation](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/DEBUGGING.md) when you set `"mode": "manual"` means the vscode extension will try to connect to an existing debug adapter server executed by yourself (i.e it will not run it automatically as done in `mode: "auto"`).
> 
> The default launch.json should work without modification.
>
Hi @brianignacio5, I changed the mode to manual to be able to start the OpenOCD Server manually, with auto mode everything happens similar as before. 
I when I use the default launch.json, the debug fails earlier with `Failed to init Debug Adapter!`

Debug Adapter Log:
```
DEBUG_ADAPTER_STARTED
2021-08-30 11:49:32,928 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STARTED
DEBUG_ADAPTER_READY2CONNECT
2021-08-30 11:49:32,976 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT
DEBUG_ADAPTER_STOPPED
2021-08-30 11:49:33,237 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STOPPED
[Stopped] : ESP-IDF Debug Adapter
```

gerekon, 2021-08-30:
@NicolSpies In manual mode you need to start both DebugAdapter and OpenOCD yourself.

NicolSpies, 2021-08-30:
@gerekon, from [https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/DEBUGGING.md](url)

> mode: Can be either auto, to start the debug adapter and openOCD server within the extension or manual, to connect to existing debug adapter and openOCD session. Default: auto.
NOTE: If set to manual, openOCD and ESP-IDF Debug Adapter have to be manually executed by the user and the extension will just try to connect to existing servers at configured ports.

I manually started openOCD from the command palette in VS Code. 
But as I have stated, even in auto mode the problem is still the same.

I have not run the DebugAdapter from command prompt yet, I relied on ESP-IDF extension up to now.

NicolSpies, 2021-08-30:
@brianignacio5, @gerekon something else I have spotted that could be related, in terminal mode using the internal JTAG and micro-USB COM port for monitoring the blink program running:

Using the micro-USB COM port the flashed blink example runs correct, echoing the `printf("Turning off the LED\n");` statements to terminal.

```
...
I (181) cpu_start: Project name:     blink
I (185) cpu_start: App version:      1
I (190) cpu_start: Compile time:     Aug 29 2021 16:59:10
I (196) cpu_start: ELF file SHA256:  f3c5fe14c0f6c604...
I (202) cpu_start: ESP-IDF:          v4.3-dirty
I (207) heap_init: Initializing. RAM available for dynamic allocation:
I (214) heap_init: At 3FC8C2D0 len 00033D30 (207 KiB): DRAM
I (221) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (227) heap_init: At 50000010 len 00001FF0 (7 KiB): RTCRAM
I (234) spi_flash: detected chip: generic
I (238) spi_flash: flash io: dio
I (243) sleep: Configure to isolate all GPIO pins in sleep state
I (249) sleep: Enable automatic switching of GPIO sleep configuration
I (256) cpu_start: Starting scheduler.
I (261) gpio: GPIO[5]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 

Turning off the LED
...
```

Using the internal JTAG in terminal mode the process stops at the following line: 
```
...
I (200) cpu_start: Project name:     blink
I (204) cpu_start: App version:      1
I (209) cpu_start: Compile time:     Aug 29 2021 16:59:10
I (215) cpu_start: ELF file SHA256:  f3c5fe14c0f6c604...
I (221) cpu_start: ESP-IDF:          v4.3-dirty
I (226) heap_init: Initializing. RAM available for dynamic allocation:
I (233) heap_init: At 3FC8C2D0 len 00033D30 (207 KiB): DRAM
I (239) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (246) heap_init: At 50000010 len 00001FF0 (7 KiB): RTCRAM
```

NicolSpies, 2021-08-30:
@brianignacio5, @gerekon, I uninstalled ESP-IDF v4.3 rc, removed ESP and  .espressif folders. Reinstalled ESP-IDF v4.3 release branch from Github. Run throught the process again.

Debug now fails with VS Code message: `Failed to init Debug Adapter!` with the default and expanded launch.json. The localhost connection refusal does not happen anymore.

In the internal JTAG monitor COM port of the device an extra line not present before is added after the line ending with RTCRAM shown earlier:

```
W (236) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
```

NicolSpies, 2021-08-30:
@brianignacio5, I captured feedback flashing in the Terminal Console using video and extracted a screen shot. 
The error messages originates from the serialwin321.py file where the COM5 port, apparently used by the debugging, cannot be opened. 
There is no COM5 in the device manager. When the Internal JTAG USB cable is connected, COM9 is present as the USB serial device. I believe this port should be used ?

[image]

xiaolongba, 2021-08-31:
Hi @NicolSpies Bro.

I also encountered a similar issue before, pls refer to this [ticket](https://github.com/espressif/vscode-esp-idf-extension/issues/437), it seems that this is a bug about debug adaptor, but I don't know if it is fixed or not.

 @brianignacio5 Bro, could you share with us the preprimary version about it first if this issue is fixed already?

thanks.

NicolSpies, 2021-08-31:
Hi @xiaolongba, thanks for the feedback, I read your ticket, you went further than me, thanks, I will wait for solution.

xiaolongba, 2021-08-31:
@NicolSpies OK, I thought this issue is general for users when using the JTAG built-in interface. Hope the official guys can resolve it asap.

gerekon, 2021-08-31:
Hi @NicolSpies 

> Using the internal JTAG in terminal mode the process stops at the following line:

There is a bug in IDF console driver that leads to system freeze during boot when USB Serial/JTAG Controller Console is used. The fix is not merged yet into `master` and `4.3 release` branches. So I recommend you to use chip's HW UART instead

Let's do it step-by-step. At first, I would ask you to check that debugging works without VsCode.
1) Connect USB cable to microUSB port for using chip's HW UART.
2) Connect USB cable to IO18/19/GND/5V pins as described [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/jtag-debugging/configure-builtin-jtag.html). For using builtin JTAG.
3) Build and flash `blink` example `idf.py -p COMxx flash monitor`
4) Run OpenOCD `openocd  -f board/esp32c3-builtin.cfg -d3 -l oocd.log`
5) Create `gdbinit` file with the following contents
```
target remote localhost:3333
mon reset halt
flushregs
b app_main
c
```
6) Run GDB `riscv32-esp-elf-gdb -x gdbinit build/blink.elf`

You should see that GDB starts, connects to target and stops at `app_main`. If not, please, attach `oocd.log` from step 4 here.

xiaolongba, 2021-09-01:
Hi @gerekon Bro,

as I said above, I encountered this issue 2 months ago. the situation is the same, the details pls refer to this [ticket](https://github.com/espressif/vscode-esp-idf-extension/issues/437#issuecomment-885031196). At that moment, @brianignacio5  Bro told me that this is a bug and needs to fix, but I am not sure if it is fixed or not.

also, I tried to follow the instructions you mentioned above, the results are still the same as I did before. the details and log as follows:

> 1. Connect USB cable to microUSB port for using chip's HW UART
I used the JTAG built-in instead as it also supports the uart

> 2\. Connect USB cable to IO18/19/GND/5V pins as described [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/jtag-debugging/configure-builtin-jtag.html). For using builtin JTAG.
  yes, I used the ESP32-C3 Mini Devkit and switched the resistance accordingly.

> 3\. Build and flash `blink` example `idf.py -p COMxx flash monitor`

> 4\. Run OpenOCD `openocd  -f board/esp32c3-builtin.cfg -d3 -l oocd.log`
[image]
[oocd.log](https://github.com/espressif/vscode-esp-idf-extension/files/7087938/oocd.log)

> 5\. Create `gdbinit` file with the following contents
> 
> ```
> target remote localhost:3333
> mon reset halt
> flushregs
> b app_main
> c
> ```

[image]

> 6. Run GDB `riscv32-esp-elf-gdb -x gdbinit build/blink.elf`

nothing happens.
[image]

xiaolongba, 2021-09-01:
However, I tried to type the same commands  from powershell to git(bash), it seems that there is some difference, as below:

For OpenOCD
```c
wireless_tech@LAPTOP-GUPHVV02 MINGW64 /f/BLE_WIFI/Espressif/SDK/blink/blink
$  F:\\Tools\\Espressif\\IDF_Tools\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20210721\\openocd-esp32\\bin\\openocd.exe -s tcl -f board/esp32c3-builtin.cfg
Open On-Chip Debugger  v0.10.0-esp32-20210721 (2021-07-21-13:35)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : datacount=2 progbufsize=16
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40101104
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Warn : No symbols for FreeRTOS!
Error: Algorithm timed out after 40001 ms.
Error: ra = 0x0
Error: sp = 0x3fc84890
Error: gp = 0x3fc89c00
Error: tp = 0x3fc8879c
Error: t0 = 0x80000000
Error: t1 = 0x0
Error: t2 = 0x0
Error: s0 = 0x600d0170
Error: s1 = 0x46a303
Error: a0 = 0x5
Error: a1 = 0xffffffff
Error: a2 = 0x3fc848ac
Error: a3 = 0x3fc8eb28
Error: a4 = 0x3fc8b470
Error: a5 = 0x600c4000
Error: a6 = 0x0
Error: a7 = 0x403816d6
Error: s2 = 0xbfe1c34c
Error: s3 = 0x0
Error: s4 = 0x0
Error: s5 = 0x0
Error: s6 = 0x0
Error: s7 = 0x0
Error: s8 = 0x0
Error: s9 = 0x0
Error: s10 = 0x0
Error: s11 = 0x0
Error: t3 = 0x0
Error: t4 = 0x0
Error: t5 = 0x0
Error: t6 = 0x0
Error: pc = 0x40380000
Error: mstatus = 0x1800
Error: mepc = 0x40380000
Error: mcause = 0x2
Error: Failed to wait algorithm (-302)!
Error: Algorithm run failed (-302)!
Error: Failed to run flasher stub (-302)!
Warn : Failed to get flash mappings (-302)!
Error: Target is already running an algorithm
Error: Failed to start algorithm (-4)!
Error: Failed to run flasher stub (-4)!
Warn : not enough working area available(requested 7640)
Error: no working area available, can't alloc space for stub code!
Error: Failed to run flasher stub (-308)!
Error: Failed to probe flash, size 0 KB
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected
···

For GDB
```c
wireless_tech@LAPTOP-GUPHVV02 MINGW64 /f/BLE_WIFI/Espressif/SDK/blink/blink
$ F:\\Tools\\Espressif\\IDF_Tools\\.espressif\\tools\\riscv32-esp-elf\\1.24.0.123_64eb9ff-8.4.0\\riscv32-esp-elf\\bin\\riscv32-esp-elf-gdb.exe -x gdbinit build/blink.elf
GNU gdb (crosstool-NG 1.24.0.123_64eb9ff) 8.3.1.20191211-git
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-host_w64-mingw32 --target=riscv32-esp-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from build/blink.elf...
Ignoring packet error, continuing...
warning: unrecognized item "timeout" in "qSupported" response
Ignoring packet error, continuing...
gdbinit:1: Error in sourced command file:
Remote replied unexpectedly to 'vMustReplyEmpty': timeout
(gdb)
```

while nothing happens when using the powershell with the same commands.

xiaolongba, 2021-09-02:
Appended more information here, I tried to return back from `v0.10.0-esp32-20210721` to `v0.10.0-esp32-20210401`and type in the same commands you mentioned above with **git(bash)**, not **powershell(windows 10)**, it works well and can stop at `app_main()` , the details as follows:

1. OpenOCD
```c
PS F:\BLE_WIFI\Espressif\SDK\blink\blink> F:\Tools\Espressif\IDF_Tools\.espressif\tools\openocd-esp32\v0.10.0-esp32-20210401\openocd-esp32\bin\openocd.exe -f board/esp32c3-builtin.cfg -d3 -l oocd.log
Open On-Chip Debugger  v0.10.0-esp32-20210401 (2021-04-01-15:46)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
User : 13 4 options.c:57 configuration_output_handler(): debug_level: 3
User : 14 10 options.c:57 configuration_output_handler():
Debug: 15 17 command.c:143 script_debug(): command - log_output log_output oocd.log
```
2. GDB
```c
$ F:\\Tools\\Espressif\\IDF_Tools\\.espressif\\tools\\riscv32-esp-elf\\1.24.0.123_64eb9ff-8.4.0\\riscv32-esp-elf\\bin\\riscv32-esp-elf-gdb.exe -x gdbinit build/blink.elf
GNU gdb (crosstool-NG 1.24.0.123_64eb9ff) 8.3.1.20191211-git
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-host_w64-mingw32 --target=riscv32-esp-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from build/blink.elf...
app_main () at ../main/blink.c:74
74          configure_led();
JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Hardware assisted breakpoint 1 at 0x420059fc: file ../main/blink.c, line 74.

Thread 2 hit Temporary breakpoint 1, app_main () at ../main/blink.c:74
74          configure_led();
(gdb)
```
However, it still cannot work when using the esp-idf-extension even if using the `v0.10.0-esp32-20210401`, also it is the same as I mentioned above when typing in the same commands in the `powershell(windows 10)` that is nothing happens.

gerekon, 2021-09-02:
@xiaolongba 

> as I said above, I encountered this issue 2 months ago. the situation is the same, the details pls refer to this ticket. At that moment, @brianignacio5 Bro told me that this is a bug and needs to fix, but I am not sure if it is fixed or not.

As far as I understand the problem was that DebugAdapter did not supported `esp32c3`. It is fixed and released. @brianignacio5 can you confirm?

> I used the JTAG built-in instead as it also supports the uart

As I wrote there is a bug in IDF which can lead to system freeze during boot when JTAG built-in UART i used.
```
There is a bug in IDF console driver that leads to system freeze during boot when USB Serial/JTAG Controller Console is used. The fix is not merged yet into master and 4.3 release branches. So I recommend you to use chip's HW UART instead
```

> nothing happens.

Your OpenOCD log shows that GDB does not connect to OpenOCD at all. Also I do not see any GDB output on your screenshot. Please, check how you run GDB, maybe it silently crashes.
```
Debug: 365 140 gdb_server.c:3584 gdb_target_start(): starting gdb server for esp32c3 on 3333
Info : 366 140 server.c:311 add_service(): Listening on port 3333 for gdb connections
Debug: 367 466973 server.c:609 sig_handler(): Terminating on Signal 2
Debug: 368 467040 command.c:143 script_debug(): command - shutdown shutdown
```

gerekon, 2021-09-02:
@xiaolongba 

> PS F:\BLE_WIFI\Espressif\SDK\blink\blink> F:\Tools\Espressif\IDF_Tools\.espressif\tools\openocd-esp32\v0.10.0-esp32-20210401\openocd-esp32\bin\openocd.exe -f board/esp32c3-builtin.cfg -d3 -l oocd.log

BTW. How do you switch between OpenOCD versions? Do you also change `OOCD_SCRIPTS` variable? It should point to tcl scripts directory of the correspodning OpenOCD distro. Another option is to override that dir via command line option -s. Like `F:\Tools\Espressif\IDF_Tools\.espressif\tools\openocd-esp32\v0.10.0-esp32-20210721\openocd-esp32\bin\openocd.exe -s  F:\Tools\Espressif\IDF_Tools\.espressif\tools\openocd-esp32\v0.10.0-esp32-20210721\openocd-esp32\share\openocd\scripts -f board/esp32c3-builtin.cfg -d3 -l oocd.log`

xiaolongba, 2021-09-03:
> As far as I understand the problem was that DebugAdapter did not supported `esp32c3`. It is fixed and released. @brianignacio5 can you confirm?

Hi @gerekon Bro

I tried to build the esp-idf-extension from the source from the master branch last night and then install this add-on,  the issue still exists, so it seems that this bug has not been fixed yet. 

> As I wrote there is a bug in IDF which can lead to system freeze during boot when JTAG built-in UART i used.

Got it, is it fixed in the latest master branch?

> Your OpenOCD log shows that GDB does not connect to OpenOCD at all. Also I do not see any GDB output on your screenshot. Please, check how you run GDB, maybe it silently crashes.

I don't know, I just typed in the commands with powershell as you seen above, but the same commands can work in git(bash). I also have some doubts about it. could you reproduce it on your side when using the win10 OS, is it the same? as you have known, all configurations are running in win10 OS.

> BTW. How do you switch between OpenOCD versions?

as you seen, I just typed in the commands `F:\Tools\Espressif\IDF_Tools.espressif\tools\openocd-esp32\v0.10.0-esp32-20210401\openocd-esp32\bin\openocd.exe -f board/esp32c3-builtin.cfg -d3 -l oocd.log`with git(bash), where using the different OpenOCD version, no specific setting.

> Do you also change `OPENOCD_SCRIPTS` variable?

No, as I said above, I don't use the esp-idf-extension for testing when using the commands. all steps are using the commands , the only difference is that it works well in git(bash), while cannot in powershell. since using the esp-idf-extension, it cannot debug no matter which OpenOCD versions I used. so I tried to use the command line first,  to make sure the command line can work first.

regarding the issues i mentioned above, you can easy to reproduce when using win10 + vscode+ power shell + esp-idf-extension.May be it works well when using Linux or Mac os, but actually, there are lots of issues occurred when using win10

xiaolongba, 2021-09-04:
Hi @gerekon Bro,

I have tested with different conditions and searched lots of information online this Saturday. there is a bit further progress and I can confirm the root is caused by `ESP-IDF/tools/tools.json` and `esp-idf-extension(v1.1.1)`. let me show you the evidences, as follows:

1. regarding no output when using the `F:\\Tools\\Espressif\\IDF_Tools\\.espressif\\tools\\riscv32-esp-elf\\esp-2021r1-8.4.0\\riscv32-esp-elf\\bin\\riscv32-esp-elf-gdb.exe build/blink.elf -x .\gdbinit` commands in the powershell.

    > it is caused by the `riscv32-esp-elf-gdb.exe` the current `ESP-IDF/tools/tools.json` provided, the current version cannot support the output on the windows OS, we should use the patch version about it, pls refer to https://github.com/espressif/crosstool-NG/releases/download/esp-2021r1/riscv32-esp-elf-gcc8_4_0-esp-2021r1-patch1-win64.zip. it works well after using this version, the details as below:
    ```c
    PS F:\BLE_WIFI\Espressif\SDK\blink\blink> F:\\Tools\\Espressif\\IDF_Tools\\.espressif\\tools\\riscv32-esp-elf\\esp-2021r1-8.4.0\\riscv32-esp-elf\\bin\\riscv32-esp-elf-gdb.exe build/blink.elf -x .\gdbinit
    F:\Tools\Espressif\IDF_Tools\.espressif\tools\riscv32-esp-elf\esp-2021r1-8.4.0\riscv32-esp-elf\bin\riscv32-esp-elf-gdb.exe: warning: Couldn't determine a path for the index cache directory.
    GNU gdb (crosstool-NG esp-2021r1) 9.2.90.20200913-git
    Copyright (C) 2020 Free Software Foundation, Inc.
    License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
    This is free software: you are free to change and redistribute it.
    There is NO WARRANTY, to the extent permitted by law.
    Type "show copying" and "show warranty" for details.
    This GDB was configured as "--host=x86_64-host_w64-mingw32 --target=riscv32-esp-elf".
    Type "show configuration" for configuration details.
    For bug reporting instructions, please see:
    <http://www.gnu.org/software/gdb/bugs/>.
    Find the GDB manual and other documentation resources online at:
    --Type <RET> for more, q to quit, c to continue without paging--
        <http://www.gnu.org/software/gdb/documentation/>.
    
    For help, type "help".
    Type "apropos word" to search for commands related to "word"...
    Reading symbols from build/blink.elf...
    0x40000000 in ?? ()
    JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
    Hardware assisted breakpoint 1 at 0x42005332: file ../main/blink.c, line 74.
    [New Thread 1070134152]
    [New Thread 1070127760]
    [Switching to Thread 1070132268]
    
    --Type <RET> for more, q to quit, c to continue without paging--
    Thread 1 hit Temporary breakpoint 1, app_main () at ../main/blink.c:74
    74          configure_led();
    (gdb)
    ```

2. the debug functionality cannot work well when using the following setups:
    - vscode
    - esp-idf-extension(v1.1.1)
    - launch.json
     ```c
     {
      "version": "0.2.0",
      "configurations": [
        {
          "type": "espidf",
          "name": "Launch",
          "request": "launch",
          "debugPort": 9998,
          "logLevel": 5,
          "mode": "auto",
          "initGdbCommands": [
            "target remote :3333",
            "symbol-file /path/to/program.elf",
            "mon reset halt",
            "flushregs",
            "thb app_main"
          ],
          "env": {
            "CUSTOM_ENV_VAR": "SOME_VALUE"
          }
        }
      ]
    }
   ```

[... 240 lines cut ...]
          1: (2501) ->~"Remote debugging using :3333\n"
          1: (8529) ->~"Ignoring packet error, continuing...\n"
          1: (8529) ->&"warning: unrecognized item \"timeout\" in \"qSupported\" response\n"
          1: (22644) ->~"Ignoring packet error, continuing...\n"
          1: (79) LaunchOptions{"name":"GDB","type":"cppdbg","request":"launch","MIMode":"gdb","miDebuggerPath":"F:\\Tools\\Espressif\\IDF_Tools\\.espressif\\tools\\riscv32-esp-elf\\esp-2021r1-8.4.0\\riscv32-esp-elf\\bin\\riscv32-esp-elf-gdb.exe","program":"F:\\BLE_WIFI\\Espressif\\SDK\\blink\\blink\\build\\blink.elf","cwd":"F:\\BLE_WIFI\\Espressif\\SDK\\blink\\blink","environment":[{"name":"PATH","value":"F:\\BLE_WIFI\\Espressif\\SDK\\esp-idf-c3\\components\\esptool_py\\esptool;F:\\BLE_WIFI\\Espressif\\SDK\\esp-idf-c3\\components\\app_update;F:\\BLE_WIFI\\Espressif\\SDK\\esp-idf-c3\\components\\espcoredump;F:\\BLE_WIFI\\Espressif\\SDK\\esp-idf-c3\\components\\partition_table;F:\\Tools\\Espressif\\IDF_Tools\\.espressif\\tools\\riscv32-esp-elf\\esp-2021r1-8.4.0\\riscv32-esp-elf\\bin;F:\\Tools\\Espressif\\IDF_Tools\\.espressif\\tools\\cmake\\3.20.3\\bin;F:\\Tools\\Espressif\\IDF_Tools\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20210721\\openocd-esp32\\bin;F:\\Tools\\Espressif\\IDF_Tools\\.espressif\\tools\\ninja\\1.10.2\\;F:\\Tools\\Espressif\\IDF_Tools\\.espressif\\tools\\idf-exe\\1.0.1\\;F:\\Tools\\Espressif\\IDF_Tools\\.espressif\\tools\\ccache\\4.3\\ccache-4.3-windows-64;F:\\BLE_WIFI\\Espressif\\SDK\\esp-idf-c3\\tools;"}],"setupCommands":[{"text":"target remote :3333"},{"text":"mon reset halt"},{"text":"thb app_main"},{"text":"flushregs"}],"externalConsole":false,"logging":{"engineLogging":true},"__configurationTarget":5,"__sessionId":"08fc5f59-0225-4622-92ec-e305ae95d875"}
          1: (138) Wait for connection completion.
          1: (382) ->=thread-group-added,id="i1"
          1: (383) ->~"GNU gdb (crosstool-NG esp-2021r1) 9.2.90.20200913-git\n"
[... cut ...]
          PS F:\BLE_WIFI\Espressif\SDK\blink\blink>  & 'c:\Users\wireless_tech\.vscode\extensions\ms-vscode.cpptools-1.6.0\debugAdapters\bin\WindowsDebugLauncher.exe' '--stdin=Microsoft-MIEngine-In-cuh0rh4v.k5e' '--stdout=Microsoft-MIEngine-Out-o3dc0r2v.uck' '--stderr=Microsoft-MIEngine-Error-ivjmntne.wjc' '--pid=Microsoft-MIEngine-Pid-pd0wx5py.a4t' '--dbgExe=F:\Tools\Espressif\IDF_Tools\.espressif\tools\riscv32-esp-elf\esp-2021r1-8.4.0\riscv32-esp-elf\bin\riscv32-esp-elf-gdb.exe' '--interpreter=mi'
          ```

xiaolongba, 2021-09-08:
Hi @gerekon Bro,

any updates about it?

NicolSpies, 2021-09-15:
Upgrading to latest ESP-IDF (Master branch, version 4.4) fixed the debugging problem using internal JTAG problem as reported.
Debugging works as expected except the problem previously reported regarding the serial monitoring of ESP32c3 via the JTAG UART while debugging or running without debugging.
  
Upgrade was done by deleting the `.espressif` and `esp-idf` folders and reconfiguring the ESP-IDF extension in VS Code.

xiaolongba, 2021-09-16:
Hi @NicolSpies 

Could you share with me your settings, especially the launch,json file? I tried to follow your tips above, but it still can not work well when using the following configuration:

```json
{
  "version": "0.2.0",
  "configurations": [
    {
      "type": "espidf",
      "name": "Launch",
      "request": "launch",
      "debugPort": 9998,
      "logLevel": 2,
      "mode": "manual",
      "skipVerifyAppBinBeforeDebug": false,
      "initGdbCommands": [
        "target remote :3333",
        "symbol-file F:/Project/blink/build/blink.elf",
        "mon reset halt",
        "flushregs",
        "thb app_main"
      ],
      "env": {
        "CUSTOM_ENV_VAR": "SOME_VALUE"
      }
    }
  ]
}
```

the error information is as below:

[image]

also, I tried to use another launch,json, it works, but it still has a few issues:

```json
{
  "version": "0.2.0",
  "configurations": [
    {
      "name": "GDB",
      "type": "cppdbg",
      "request": "launch",
      "MIMode": "gdb",
      "miDebuggerPath": "${command:espIdf.getXtensaGdb}",
      "program": "${workspaceFolder}/build/${command:espIdf.getProjectName}.elf",
      "windows": {
        "program": "${workspaceFolder}\\build\\${command:espIdf.getProjectName}.elf"
      },
      "cwd": "${workspaceFolder}",
      "environment": [{ "name": "PATH", "value": "${config:idf.customExtraPaths}" }],
      "setupCommands": [
        { "text": "target remote :3333" },
        { "text": "mon reset halt" },
        { "text": "thb app_main" },
        { "text": "flushregs" }
      ],
      "externalConsole": false,
      "logging": {
        "engineLogging": true
      }
    }
  ]
}
```

I make an animation to expand more as below:
[image]

@brianignacio5 @gerekon  Bro, could you give me a hand for the last step for debugging.

NicolSpies, 2021-09-16:
There is a process to follow when debugging otherwise the  ` Failed to init Debug Adapter!` problem returns. The working debug process in VS Code using the blink example:

1. Build and flash using the ESP32C3 internal JTAG.
2. Select `Start OpenOCD` and confirm it is running.
3. Select `ESP-IDF Monitor device` icon. `Terminal` will open, device will start and will stop at `I (531) heap_init: At 50000010 len 00001FF0 (7 KiB): RTCRAM`.
4. Start debugging by `F5` or the `Launch` icon. 

If step 3, device monitor is not opened first, the Debug Adaptor fails to initialize with the following error feedback: 

```
Windows PowerShell
Copyright (C) Microsoft Corporation. All rights reserved.

Try the new cross-platform PowerShell https://aka.ms/pscore6

PS D:\Dropbox\DATA\DEVELOPMENT\ESP32C3\blink> C:\Users\nicol\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:\Users\nicol\esp\esp-idf\tools\idf.py -p COM9 monitor
Executing action: monitor
Running idf_monitor in directory d:\dropbox\data\development\esp32c3\blink
Executing "C:\Users\nicol\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:\Users\nicol\esp\esp-idf\tools/idf_monitor.py -p COM9 -b 115200 --toolchain-prefix riscv32-esp-elf- --target esp32c3 --decode-panic backtrace d:\dropbox\data\development\esp32c3\blink\build\blink.elf -m 'C:\Users\nicol\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe' 'C:\Users\nicol\esp\esp-idf\tools\idf.py' '-p' 'COM9'"...
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM9 instead...
--- idf_monitor on \\.\COM9 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
Exception in thread Thread-2:
Traceback (most recent call last):
  File "threading.py", line 932, in _bootstrap_inner
  File "threading.py", line 870, in run
  File "C:\Users\nicol\esp\esp-idf\tools\idf_monitor_base\stoppable_thread.py", line 58, in _run_outer

  self.run()
  File "C:\Users\nicol\esp\esp-idf\tools\idf_monitor_base\serial_reader.py", line 56, in run
    self.serial.open()
  File "C:\Users\nicol\.espressif\python_env\idf4.4_py3.8_env\lib\site-packages\serial\serialwin32.py", line 80, in open
    self._reconfigure_port()
  File "C:\Users\nicol\.espressif\python_env\idf4.4_py3.8_env\lib\site-packages\serial\serialwin32.py", line 222, in _reconfigure_port
    raise SerialException(
serial.serialutil.SerialException: Cannot configure port, something went wrong. Original message: PermissionError(13, 'A device attached to the system is not functioning.', None, 31) 
```

When this failure occurred, the only way to get it working again is to:

1. Select `Stop OpenOCD Running`.
2. Disconnect USB cable from computer and reconnect.
3. Repeat steps 2 to 4 above.

NicolSpies, 2021-09-16:
> Could you share with me your settings, especially the launch,json file?

```
{
  "version": "0.2.0",
  "configurations": [
    {
      "type": "espidf",
      "name": "Launch",
      "request": "launch",
      "skipVerifyAppBinBeforeDebug": true
    }
  ]
}
```

The `"skipVerifyAppBinBeforeDebug": true` fix was reported a few days ago [https://esp32.com/viewtopic.php?t=23190](url)

xiaolongba, 2021-09-16:
> ```
> {
>   "version": "0.2.0",
>   "configurations": [
>     {
>       "type": "espidf",
>       "name": "Launch",
>       "request": "launch",
>       "skipVerifyAppBinBeforeDebug": true
>     }
>   ]
> } [...]

Big thanks bro, I followed your tips, it works smoothly. it also resolved the invalid issue when restart the debug, the details are as below:
[image]

@brianignacio5 Bro, I think you need to do some modifies about debugging to clarifying the debug issue when using the following configuration:

```json
{
  "version": "0.2.0",
  "configurations": [
    {
      "type": "espidf",
      "name": "Launch",
      "request": "launch",
      "debugPort": 9998,
      "logLevel": 2,
      "mode": "manual",
      "skipVerifyAppBinBeforeDebug": false,
      "initGdbCommands": [
        "target remote :3333",
        "symbol-file F:/Project/blink/build/blink.elf",
        "mon reset halt",
        "flushregs",
        "thb app_main"
      ],
      "env": {
        "CUSTOM_ENV_VAR": "SOME_VALUE"
      }
    }
  ]
}
```

@NicolSpies Bro, regarding the monitor you mentioned, it is not mandatory when debugging, you can close it here as below:
 
[image]

xiaolongba, 2021-09-16:
also, followed by the topic above, I found another issue, that is below:

[image]

as you can see above, the first time it can work, but it failed to debug when stopping and then start again, the operation pls refer to the animation above.

@brianignacio5 Bro, could you help me check, to set up one perfect environment with vscode + win10 + esp-idf extension, it took me at least several months, could you give me a hand for last setup?

NicolSpies, 2021-09-16:
> reset the device or unplug the cable line and then plug in again.

Yes, this is what I refer to above.

xiaolongba, 2021-09-16:
> Yes, this is what I refer to above.

that makes me crazy, it is too inconvenience

brianignacio5, 2021-09-17:
With regards to the `"skipVerifyAppBinBeforeDebug": true` I made it default value in GitHub master already. Sorry for the trouble. This is too verify that your project elf is the same as the one in the selected serial port device but apparently still need some work.

With regards the start/stop situation I'll take a look, it is probably not finishing the process correctly or hanging somewhere. 

@xiaolongba I'm sorry the debugging experience hasn't been smooth enough but thank you for testing this feature and giving us the feedback. Is there anything else not working properly ?

xiaolongba, 2021-09-17:
> With regards to the `"skipVerifyAppBinBeforeDebug": true` I made it default value in GitHub master already.

yes, I saw it already

> This is too verify that your project elf is the same as the one in the selected serial port device but apparently still need some work.

ok, waiting for good news

> Is there anything else not working properly ?

no at this moment

> With regards the start/stop situation I'll take a look, it is probably not finishing the process correctly or hanging somewhere.

OK, waiting for your good news

let me introduce the background  about debugging here:

I am focusing on the education for some students or newbies for the esp32 series. my team has released the specified dk for esp32-c3, and it has been sold in China and you can find it on Taobao if you are interested. we are preparing for some tutorials about it, at that moment, the basic chapters for esp32-c3, they are almost finished, the next step is to spend more time on BLE tutorials as materials about BLE for esp32 series are less, lots of them are WiFi related.  my team has published the completed SIG Mesh tutorials based on Nordic 52840 and receives lots of friends with a favor well. so espressif's support is very important for us. but sometimes, the response from officials is a little slow and it leads to the tutorials publishing is slow also. basically, we don't ask some stupid questions or simple questions. Only we face some issues we cannot resolve it and then we will request support from officials. 

anyway, many thanks for your great support, bro. at last, I can share with you something about tutorials if you are interested, but all of them are in Chinese. I am not sure if you can understand. of course, I can translate them into English, but it isnot the highest priority for my team at this moment.

again, thanks again.

NicolSpies, 2021-09-17:
> Is there anything else not working properly ?

@brianignacio5, The only outstanding thing for C3 debugging missing I am aware of at this stage is the monitoring of the C3 UART via the JTAG interface. It starts but stops at `I (227) heap_init: At 50000010 len 00001FF0 (7 KiB): RTCRAM` in the monitor. 
The verbose feedback in the monitor from `ESP_LOGI(TAG, "Turning the LED %s!", s_led_state == true ? "ON" : "OFF");` is not displayed.

closed 2021-09-19 by NicolSpies

Ooyekunle, 2022-05-22:
I have been having this issue all weekend using VScode trying to debug ESP32C3 Devkit v2. Here is the log from my openOCD launch attempt.

Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:43)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
User : 13 3 options.c:63 configuration_output_handler(): debug_level: 3
User : 14 8 options.c:63 configuration_output_handler(): 
Debug: 15 11 options.c:244 add_default_dirs(): bindir=/builds/idf/openocd-esp32/_build/../openocd-esp32/bin
Debug: 16 16 options.c:245 add_default_dirs(): pkgdatadir=/builds/idf/openocd-esp32/_build/../openocd-esp32/share/openocd
Debug: 17 22 options.c:246 add_default_dirs(): exepath=C:/Espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin
Debug: 18 31 options.c:247 add_default_dirs(): bin2data=../share/openocd
Debug: 19 35 configuration.c:42 add_script_search_dir(): adding C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts
Debug: 20 45 configuration.c:42 add_script_search_dir(): adding C:/Users/leksy/AppData/Roaming/OpenOCD
Debug: 21 53 configuration.c:42 add_script_search_dir(): adding C:/Espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin/../share/openocd/site
Debug: 22 67 configuration.c:42 add_script_search_dir(): adding C:/Espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin/../share/openocd/scripts
Debug: 23 77 configuration.c:97 find_file(): found C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts/board/esp32c3-builtin.cfg
Debug: 24 86 configuration.c:97 find_file(): found C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts/interface/esp_usb_jtag.cfg   
Debug: 25 95 command.c:146 script_debug(): command - adapter driver esp_usb_jtag
Info : 27 101 transport.c:117 allow_transports(): only one transport option; autoselect 'jtag'
Debug: 28 106 command.c:146 script_debug(): command - espusbjtag vid_pid 0x303a 0x1001
Info : 30 110 esp_usb_jtag.c:899 esp_usb_jtag_vid_pid(): esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Debug: 31 118 command.c:146 script_debug(): command - espusbjtag caps_descriptor 0x2000
Info : 33 125 esp_usb_jtag.c:912 esp_usb_jtag_caps_descriptor(): esp_usb_jtag: capabilities descriptor set to 0x2000
Debug: 34 133 command.c:146 script_debug(): command - adapter speed 40000
Debug: 36 136 core.c:1822 jtag_config_khz(): handle jtag khz
Debug: 37 139 core.c:1785 adapter_khz_to_speed(): convert khz to interface specific speed value
Debug: 38 144 core.c:1785 adapter_khz_to_speed(): convert khz to interface specific speed value
Debug: 39 151 configuration.c:97 find_file(): found C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts/target/esp32c3.cfg
Debug: 40 160 command.c:146 script_debug(): command - transport select jtag
Warn : 41 164 transport.c:286 jim_transport_select(): Transport "jtag" was already selected
Debug: 42 169 configuration.c:97 find_file(): found C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts/bitsbytes.tcl
Debug: 43 176 configuration.c:97 find_file(): found C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts/memory.tcl
Debug: 44 187 configuration.c:97 find_file(): found C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts/mmr_helpers.tcl
Debug: 45 195 configuration.c:97 find_file(): found C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts/target/esp_common.cfg       
Debug: 46 204 command.c:146 script_debug(): command - add_help_text program_esp write an image to flash, address is only required for binary images. verify, reset, exit, compress, restore_clock are optional
Debug: 48 219 command.c:1115 help_add_command(): added 'program_esp' help text
Debug: 49 225 command.c:146 script_debug(): command - add_usage_text program_esp <filename> [address] [verify] [reset] [exit] [compress] [restore_clock]
Debug: 51 234 command.c:1141 help_add_command(): added 'program_esp' usage text
Debug: 52 238 command.c:146 script_debug(): command - add_help_text program_esp_bins write all the images at address specified in flasher_args.json generated while building idf project
Debug: 54 249 command.c:1115 help_add_command(): added 'program_esp_bins' help text
Debug: 55 253 command.c:146 script_debug(): command - add_usage_text program_esp_bins <build_dir> flasher_args.json [verify] [reset] [exit] [compress] [restore_clock]
Debug: 57 264 command.c:1141 help_add_command(): added 'program_esp_bins' usage text
Debug: 58 268 command.c:146 script_debug(): command - add_help_text esp_get_mac Print MAC address of the chip. Use a `format` argument to return formatted MAC value
Debug: 60 276 command.c:1115 help_add_command(): added 'esp_get_mac' help text
Debug: 61 282 command.c:146 script_debug(): command - add_usage_text esp_get_mac [format]
Debug: 63 290 command.c:1141 help_add_command(): added 'esp_get_mac' usage text
Debug: 64 294 command.c:146 script_debug(): command - jtag newtap esp32c3 cpu -irlen 5 -expected-id 0x00005c25
Debug: 65 301 tcl.c:572 jim_newtap_cmd(): Creating New Tap, Chip: esp32c3, Tap: cpu, Dotted: esp32c3.cpu, 4 params
Debug: 66 312 tcl.c:596 jim_newtap_cmd(): Processing option: -irlen
Debug: 67 316 tcl.c:596 jim_newtap_cmd(): Processing option: -expected-id
[... cut ...]
Debug: 130 839 semihosting_common.c:100 semihosting_common_init():  
Error: 131 883 esp_usb_jtag.c:662 esp_usb_jtag_init(): esp_usb_jtag: could not find or open device!
Debug: 132 891 command.c:629 run_command(): Command 'init' failed with error code -4
User : 133 896 command.c:694 command_run_line():
Debug: 134 902 riscv.c:483 riscv_deinit_target(): riscv_deinit_target()
Debug: 135 909 riscv-013.c:1543 deinit_target(): riscv_deinit_target()
Debug: 136 918 target.c:2218 target_free_all_working_areas_restore(): freeing all working areas

brianignacio5, 2022-05-26:
Could you share the [troubleshoot](https://github.com/espressif/vscode-esp-idf-extension#troubleshooting) information @Ooyekunle ?