espressif/vscode-esp-idf-extension #511: [Feature Request]: Examples in sub-directory is not shown in "Show Example Projects".

vadaliya, 2021-09-03:
problem :
in the show example projects window, it only shows the example which is not in the sub-directory or we can say that examples are categorywise.

for example, in the Bluetooth section, only 3 examples show. but there are actually 4 more folders that have multiple examples in sub-directory style. so that sub-folders are not shown.

**Describe the solution you'd like**
if any folder has sub-folders, then in the example window, that folder shows the sub-folder in the sub-directory style. so the developer knows that this folder has more examples in it. 

--- Bluetooth
      |
        --- bluedroid
             |
             --- ble
             --- ble_50
             --- classic_bt
             --- coex
        --- blufi
        --- esp_ble_mesh
             |
             --- 
             ---
        --- esp_hid_decice
        --- esp_hid_host
        --- hci
             |
             --- 
             ---
        --- nimble
             |
             --- 
             ---

like this.... sorry for my bad english.

meltdown03, 2021-09-03:
Also the examples in `protocols/http_server` such as `restful_server` and `persistent_sockets` are not shown.

brianignacio5, 2021-09-03:
It is already fixed in master. Will release a new version on marketplace soon once we fix other important issues.

[image]

[image]

meltdown03, 2021-09-04:
I built it from master just now in Linux and it's sill not working for me. What commit fixed it or is it not on Github yet?

meltdown03, 2021-09-04:
It seems like the `webpack --mode development` (which the debug method uses) builds a much larger `dist/views/examples-bundle.js` file than the `--mode production`. (~733KB vs ~336KB) If I copy that larger file to the `${VSCODE_EXTENSION_DIR}/dist/views` and overwrite it, then it works outside of the VSCode Extension Development Host

brianignacio5, 2021-09-06:
Oh very weird. I'll take a look @meltdown03 thanks for reviewing it.

PR #513 (2021-09-07): fix recursive rendering on production
Fix #511 

Recursive rendering of component seems to break on web pack production if not using dynamic import within the component.
closed 2021-09-07 by brianignacio5