espressif/vscode-esp-idf-extension #512: [Feature Request]: Add CCache setting for faster rebuilds

meltdown03, 2021-09-03:
Would it be possible to add an option to enable CCache for faster rebuilds?
I manually added the option to the tasks.json files, but it only works with the Project Tasks and not the built-in commands, and it's not very convenient.

meltdown03, 2021-09-08:
Here is a portion of the build command output:
```
Executing action: all (aliases: build)
Running cmake in directory /home/neal/esp/esp-adf/examples/recorder/pipeline_raw_http/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32s2 -DCCACHE_ENABLE=0 /home/neal/esp/esp-adf/examples/recorder/pipeline_raw_http"...
-- Found Git: /usr/bin/git (found version "2.33.0") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
```
I don't think CCache is working. I tried with both the Project Task and the built-in build command and both seem to take a long time to rebuild. As you can see above,  `-DCCACHE_ENABLE=0` is set even thought enabled CCache in the settings.json file

brianignacio5, 2021-09-08:
And you set `idf.enableCCache` to `true` on settings.json ? Default is false.

meltdown03, 2021-09-09:
> And you set `idf.enableCCache` to `true` on settings.json ? Default is false.
> 
> You might want to delete the `build` folder before rebuild.

Yeah, I see it to true in the global settings.json. I tried deleting it multiple times so i could tell if it was using ccache

closed 2021-09-09 by brianignacio5

meltdown03, 2021-09-09:
Thanks, it's working with the built-in build command, but not the Tasks. I can just change the template before building the extension. Any plans to add DFU flashing, I see Arduino IDE has it for arduino-esp32?

brianignacio5, 2021-09-10:
For the task you just need to add `--cache` flag. Since CCache is optional for MacOS/Linux we could had added it to Windows but due to time constraint was not part of yesterday release. Apologies for that.

Yes we are planning to add DFU flashing but this is already available in command line with `idf.py dfu` and `idf.py dfu-flash` for ESP32-S2 and ESP32-S3 targets.

meltdown03, 2021-09-10:
In messing around, I found it easier to set the CCACHE_ENABLE option in the
CMakeLists.txt file. I like using the tasks because I can click monitor and
it will build, flash and monitor all with one click. If I want to use
ccache, I have to set it for all three tasks.

PR #517 (2021-09-08): add setting to enable ccache
Fix #512 

Add `idf.enableCCache` to enable CCache on the build task.

It could also be done by adding `IDF_CCACHE_ENABLE` to `idf.customExtraVars`.
PR #523 (2021-09-09): move ccache flag in build task
Using `IDF_CCACHE_ENABLE` only works if calling `idf.py build`. Since we are calling CMake directly, we need to pass `-DCCACHE_ENABLE=1` as an argument.

Fix #512