espressif/vscode-esp-idf-extension #515: [Bug Report]: "Select port to use" does not work on macOS

bpietsch, 2021-09-07:
**Describe the bug**
The "Select port to use" option" does not have an effect on macOS. I have to close the workspace, manually modify the "idf.port" option and re-open it for the change to take effect.

**To Reproduce**
N/A

**Expected behavior**
As described

**Screenshots**
N/A

**Environment (please complete the following information):**

- OS Name & Version: MacOS X 11.5.2
- VSCode Version: 1.59.1 (Espressif IDF extension: 1.1.1)
- ESP-IDF Version: 4.3
- Python Version: [e.g. 2.7.10]

> **NOTE**: You can use the `ESP-IDF: Doctor command` to generate a report of your configuration.

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS darwin x64 20.6.0 
System environment variable PATH 
 /Users/brian/.pyenv/shims:/Users/brian/.pyenv/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Applications/VMware Fusion.app/Contents/Public:/opt/X11/bin:/Library/Apple/usr/bin:/Applications/Wireshark.app/Contents/MacOS:/usr/local/opt/binutils/bin:/usr/local/opt/sphinx-doc/bin 
Visual Studio Code version 1.59.1 
Visual Studio Code language en 
Visual Studio Code shell zsh 
ESP-IDF Extension version 1.1.1 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) /Users/brian/Tools/esp/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
Custom extra paths (idf.customExtraPaths) /Users/brian/Tools/esp/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin:/Users/brian/Tools/esp/tools/xtensa-esp32s2-elf/esp-2021r1-8.4.0/xtensa-esp32s2-elf/bin:/Users/brian/Tools/esp/tools/xtensa-esp32s3-elf/esp-2021r1-8.4.0/xtensa-esp32s3-elf/bin:/Users/brian/Tools/esp/tools/riscv32-esp-elf/1.24.0.123_64eb9ff-8.4.0/riscv32-esp-elf/bin:/Users/brian/Tools/esp/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/Users/brian/Tools/esp/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/Users/brian/Tools/esp/tools/openocd-esp32/v0.10.0-esp32-20210401/openocd-esp32/bin
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"/Users/brian/Tools/esp/tools/openocd-esp32/v0.10.0-esp32-20210401/openocd-esp32/share/openocd/scripts"}
Virtual env Python Path (idf.pythonBinPath) /Users/brian/Tools/esp/python_env/idf4.3_py3.8_env/bin/python
Serial port (idf.port) /dev/cu.usbserial-14401
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) /Users/brian/Tools/esp
Git Path (idf.gitPath) git
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espIdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espIdfPath) false
Access to ESP-IDF Custom extra paths
Access to /Users/brian/Tools/esp/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin: true
Access to /Users/brian/Tools/esp/tools/xtensa-esp32s2-elf/esp-2021r1-8.4.0/xtensa-esp32s2-elf/bin: true
Access to /Users/brian/Tools/esp/tools/xtensa-esp32s3-elf/esp-2021r1-8.4.0/xtensa-esp32s3-elf/bin: true
Access to /Users/brian/Tools/esp/tools/riscv32-esp-elf/1.24.0.123_64eb9ff-8.4.0/riscv32-esp-elf/bin: true
Access to /Users/brian/Tools/esp/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin: true
Access to /Users/brian/Tools/esp/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin: true
Access to /Users/brian/Tools/esp/tools/openocd-esp32/v0.10.0-esp32-20210401/openocd-esp32/bin: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) true
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version 2.30.1
ESP-IDF version 4.3
Python version 3.8.5
Python's pip version 21.0.1
-------------------------------------------------- Python packages in idf.pythonBinPath ----------------------------------------------------
bitstring version: 3.1.9
Brotli version: 1.0.9
certifi version: 2021.5.30
cffi version: 1.14.6
charset-normalizer version: 2.0.3
click version: 8.0.1
construct version: 2.10.54
contextlib2 version: 21.6.0
cryptography version: 3.4.7
ecdsa version: 0.17.0
Flask version: 0.12.5
Flask-Compress version: 1.10.1
Flask-SocketIO version: 2.9.6
future version: 0.18.2
gcovr version: 5.0
gdbgui version: 0.13.2.0
gevent version: 1.5.0
greenlet version: 1.1.0
idf-component-manager version: 0.2.99b0
idna version: 3.2
itsdangerous version: 2.0.1
Jinja2 version: 3.0.1
kconfiglib version: 13.7.1
lxml version: 4.6.3
MarkupSafe version: 2.0.1
pip version: 21.0.1
psutil version: 5.8.0
pycparser version: 2.20
pyelftools version: 0.27
pygdbmi version: 0.9.0.2
Pygments version: 2.9.0
pyparsing version: 2.3.1
pyserial version: 3.5
python-engineio version: 3.14.2
python-socketio version: 4.6.1
PyYAML version: 5.4.1
reedsolo version: 1.5.4
requests version: 2.26.0
requests-toolbelt version: 0.9.1
schema version: 0.7.4
semantic-version version: 2.8.5
setuptools version: 57.4.0
six version: 1.16.0
tqdm version: 4.61.2
urllib3 version: 1.26.6
websocket-client version: 1.1.0
Werkzeug version: 0.16.1
wheel version: 0.36.2
xmlrunner version: 1.7.7
---------------------------------------------------- Check ESP-IDF python requirements.txt -------------------------------------------------
Check ESP-IDF Python packages Python requirements from /Users/brian/Tools/esp/esp-idf/requirements.txt are satisfied.
---------------------------------------------------- Check extension requirements.txt ------------------------------------------------------
Check Extension Python packages Python requirements from /Users/brian/.vscode/extensions/espressif.esp-idf-extension-1.1.1/requirements.txt are satisfied.
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages Python requirements from /Users/brian/.vscode/extensions/espressif.esp-idf-extension-1.1.1/esp_debug_adapter/requirements.txt are satisfied.
---------------------------------------------------- Visual Studio Code launch.json --------------------------------------------------------
{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "type": "espidf",
            "name": "Launch",
            "request": "launch",
        }
    ]
}
 
---------------------------------------------------- Visual Studio Code c_cpp_properties.json ----------------------------------------------
{
    "configurations": [
        {
            "name": "ESP-IDF",
            "cStandard": "c11",
            "cppStandard": "c++17",
            "includePath": [
                "${config:idf.espIdfPath}/components/**",
                "${config:idf.espIdfPathWin}/components/**",
                "${workspaceFolder}/**"
            ],
            "browse": {
                "path": [
                    "${config:idf.espIdfPath}/components",
                    "${config:idf.espIdfPathWin}/components",
                    "${workspaceFolder}"
                ],
                "limitSymbolsToIncludedHeaders": false
            },
            "compileCommands": "${workspaceFolder}/build/compile_commands.json"
        }
    ],
    "version": 4
} 
--------------------------------------------------------------------------------------------------------------------------------------------

brianignacio5, 2021-09-07:
This is a duplicate of #487 

This is a local settings.json overriding the global user settings.

Configuration settings precedence are:

1. Workspace folder configuration settings in `${workspaceFolder}/.vscode/settings.json`
2. Workspace configuration settings defined in the workspace's `<name>.code-workspace` file as shown below.
3. User settings defined in
   - *Windows* `%APPDATA%\Code\User\settings.json`
   - *macOS* `$HOME/Library/Application Support/Code/User/settings.json`
   - *Linux* `$HOME/.config/Code/User/settings.json`

This extension uses the `idf.saveScope` configuration setting to determine where to save configuration settings in features such as the `Select port to use` command. Use the `ESP-IDF: Select where to save configuration settings` to change `idf.saveScope` for the current scope.

brianignacio5, 2021-09-13:
Closing issue for lack of response. Please comment here if you need more support.

closed 2021-09-13 by brianignacio5

Elijahg, 2021-09-13:
I have had this issue too, the lack of UI feedback makes the lack of change seem like a bug. Perhaps when the user changes the port, it would be good to at least alert the user that the selected port won't apply due to a user settings override. It'd be very unlikely a user would want the GUI switcher to change the port it globally without it actually applying to the port associated with the project itself.

Alternatively, to me at least, it makes sense to have serial flashers in the workspace folder config (.vscode/settings.json) by default. Most projects have a 1:1 relationship with a PCB, and in my case at least, each PCB has its own serial to USB converter; therefore each project has a 1:1:1 relationship between project, board and port. This makes the global user settings wrong most of the time, so (with the default scope) I have to change the port each time I open a project, which wouldn't be the case if the port was always in the workspace folder config settings.