espressif/vscode-esp-idf-extension #516: [Bug Report]:

xiongyu0523, 2021-09-07:
My environment was good but suddenly it reports below error when I do ESP-IDF: Set Espressif device target

```
{ error: [Error: non zero exit code 1 Traceback (most recent call last): File "C:\Users\yuxiong\esp\esp-idf\tools\idf.py", line 794, in <module> ret = subprocess.call([WINPTY_EXE, sys.executable] + sys.argv, env=os.environ) File "subprocess.py", line 340, in call File "subprocess.py", line 854, in __init__ File "subprocess.py", line 1307, in _execute_child FileNotFoundError: [WinError 2] The system cannot find the file specified at ChildProcess.<anonymous> (c:\Users\yuxiong\.vscode\extensions\espressif.esp-idf-extension-1.1.1\dist\extension.js:1:4160) at ChildProcess.emit (events.js:315:20) at ChildProcess.EventEmitter.emit (domain.js:467:12) at Process.ChildProcess._handle.onexit (internal/child_process.js:277:12)] }
```

brianignacio5, 2021-09-07:
Something changed in your system environment. There was an old issue related to this:

https://github.com/espressif/vscode-esp-idf-extension/issues/14

Check your visual studio code configuration settings to see if `terminal.integrated.env.windows` has changed as shown in the linked issue.

xiongyu0523, 2021-09-07:
Checked my user setting file. There is no `terminal.integrated.env.windows` defined in settings.json

xiongyu0523, 2021-09-07:
How can I do a full clean uninstall to the environment, just delete `.espressif` and `esp` folder hold the esp-idf? If it is hard to catch root cause I can do a fresh re-installation

brianignacio5, 2021-09-07:
Settings.json can be defined in several places:

Configuration settings precedence are:

1. Workspace folder configuration settings in `${workspaceFolder}/.vscode/settings.json`
2. Workspace configuration settings defined in the workspace's `<name>.code-workspace` file as shown below.
3. User settings defined in
   - *Windows* `%APPDATA%\Code\User\settings.json`
   - *macOS* `$HOME/Library/Application Support/Code/User/settings.json`
   - *Linux* `$HOME/.config/Code/User/settings.json`
   
Have you changed the terminal you are using in visual studio code ? (like trying to use MINGW32 instead of regular powershell or cmd.exe ?)

Deleting these folders won't clean the environment. The issue is related to idf.py not being able to find WINPTY_EXE which should exists in your environment variables. Is hard to know what changed without additional information of your setup environment

xiongyu0523, 2021-09-10:
resolve by re-installation

closed 2021-09-10 by xiongyu0523