espressif/vscode-esp-idf-extension #527: [Bug Report]: ESP-IDF:  size analysis of binaries command invalid

xiaolongba, 2021-09-13:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [x] Make sure you have searched for existing bugs and features request before you post an issue.
- [x] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [x] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
**ESP-IDF:  size analysis of binaries** command is invalid after building the project

**To Reproduce**
1. Build the project
2. press **F1** and select **ESP-IDF:  size analysis of binaries**

**Expected behavior**
**ESP-IDF:  size analysis of binaries** command can work as expected

**Screenshots**
[image]

**Environment (please complete the following information):**

- OS Name & Version: win10
- VSCode Version: 1.60.0
- ESP-IDF Version: master branch
- Python Version: 3.8.7

> **NOTE**: You can use the `ESP-IDF: Doctor command` to generate a report of your configuration.

**Output**

```c

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19042 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 D:\Python39\Scripts\;D:\Python39\;%ALLEGRO_BRD2ODB%\nv\bin;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\windows\system32;C:\windows;C:\windows\System32\Wbem;C:\windows\System32\WindowsPowerShell\v1.0\;C:\windows\System32\OpenSSH\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;c:\Program Files (x86)\HP\HP Performance Advisor;D:\PuTTY\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin\;C:\Program Files\IVI Foundation\VISA\Win64\Bin\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin;D:\Git\cmd;C:\Program Files (x86)\STMicroelectronics\STM32 ST-LINK Utility\ST-LINK Utility;C:\Program Files (x86)\IVI Foundation\IVI\bin;C:\Program Files\IVI Foundation\IVI\bin;D:\CMake\bin;C:\Program Files\Nordic Semiconductor\nrf-command-line-tools\bin\;D:\doxygen\bin;D:\Mscgen;D:\stlink-1.3.0-win64\bin;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\ProgramData\chocolatey\bin;D:\nodejs\;C:\Program Files\Docker\Docker\resources\bin;C:\ProgramData\DockerDesktop\version-bin;C:\Users\wireless_tech\AppData\Local\Microsoft\WindowsApps;D:\Microsoft VS Code\bin;C:\Users\wireless_tech\AppData\Roaming\npm 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.60.0 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.2.0 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) F:/BLE_WIFI/Espressif/SDK/esp-idf-c3/
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
Custom extra paths (idf.customExtraPaths) F:\Tools\Espressif\IDF_Tools\.espressif\tools\xtensa-esp32-elf\esp-2021r1-8.4.0\xtensa-esp32-elf\bin;F:\Tools\Espressif\IDF_Tools\.espressif\tools\xtensa-esp32s2-elf\esp-2021r1-8.4.0\xtensa-esp32s2-elf\bin;F:\Tools\Espressif\IDF_Tools\.espressif\tools\xtensa-esp32s3-elf\esp-2021r1-8.4.0\xtensa-esp32s3-elf\bin;F:\Tools\Espressif\IDF_Tools\.espressif\tools\riscv32-esp-elf\esp-2021r1-8.4.0\riscv32-esp-elf\bin;F:\Tools\Espressif\IDF_Tools\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;F:\Tools\Espressif\IDF_Tools\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;F:\Tools\Espressif\IDF_Tools\.espressif\tools\cmake\3.20.3\bin;F:\Tools\Espressif\IDF_Tools\.espressif\tools\openocd-esp32\v0.10.0-esp32-20210401\openocd-esp32\bin;F:\Tools\Espressif\IDF_Tools\.espressif\tools\ninja\1.10.2;F:\Tools\Espressif\IDF_Tools\.espressif\tools\idf-exe\1.0.1;F:\Tools\Espressif\IDF_Tools\.espressif\tools\ccache\4.3\ccache-4.3-windows-64;F:\Tools\Espressif\IDF_Tools\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"F:\\Tools\\Espressif\\IDF_Tools\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20210401/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) F:/Tools/Espressif/IDF_Tools/.espressif/python_env/idf4.4_py3.8_env/Scripts/python.exe
Serial port (idf.port) COM35
OpenOCD Configs (idf.openOcdConfigs) board/esp32c3-builtin.cfg
ESP-IDF Tools Path (idf.toolsPath) F:\Tools\Espressif\IDF_Tools\.espressif
Git Path (idf.gitPath) F:/Tools/Espressif/IDF_Tools/.espressif/tools/idf-git/2.30.1/cmd/git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-IDF Custom extra paths
Access to F:\Tools\Espressif\IDF_Tools\.espressif\tools\xtensa-esp32-elf\esp-2021r1-8.4.0\xtensa-esp32-elf\bin: true
Access to F:\Tools\Espressif\IDF_Tools\.espressif\tools\xtensa-esp32s2-elf\esp-2021r1-8.4.0\xtensa-esp32s2-elf\bin: true
Access to F:\Tools\Espressif\IDF_Tools\.espressif\tools\xtensa-esp32s3-elf\esp-2021r1-8.4.0\xtensa-esp32s3-elf\bin: true
Access to F:\Tools\Espressif\IDF_Tools\.espressif\tools\riscv32-esp-elf\esp-2021r1-8.4.0\riscv32-esp-elf\bin: true
Access to F:\Tools\Espressif\IDF_Tools\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin: true
Access to F:\Tools\Espressif\IDF_Tools\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin: true
Access to F:\Tools\Espressif\IDF_Tools\.espressif\tools\cmake\3.20.3\bin: true
Access to F:\Tools\Espressif\IDF_Tools\.espressif\tools\openocd-esp32\v0.10.0-esp32-20210401\openocd-esp32\bin: true
Access to F:\Tools\Espressif\IDF_Tools\.espressif\tools\ninja\1.10.2: true
[... 97 lines cut ...]
            ],
            "browse": {
                "path": [
                    "${config:idf.espIdfPath}/components",
                    "${config:idf.espIdfPathWin}/components",
                    "${config:idf.espAdfPath}/components/**",
                    "${config:idf.espAdfPathWin}/components/**",
                    "${workspaceFolder}"
                ],
                "limitSymbolsToIncludedHeaders": false
            },
            "compileCommands": "${workspaceFolder}/build/compile_commands.json"
        }
    ],
    "version": 4
} 
--------------------------------------------------------------------------------------------------------------------------------------------
````

**Logs**

[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/7151709/esp_idf_vsc_ext.log)

**Additional context**
NULL

brianignacio5, 2021-09-13:
Based on the log, seems to be some error in the idf_size.py json output. 

I can't seem to reproduce the issue locally on Windows using the latest ESP-IDF master with the blink example.

Maybe you can share the output of running the following commands on the `Open ESP-IDF Terminal` and run:

`python.exe $Env:IDF_PATH/tools/idf_size.py build/blink.map --json`

`python.exe $Env:IDF_PATH/tools/idf_size.py build/blink.map --archives --json`

`python.exe $Env:IDF_PATH/tools/idf_size.py build/blink.map --file --json`

Are all output well-formed JSON?

You can try to update your ESP-IDF master git commit and submodules to latest commit.

xiaolongba, 2021-09-13:
Hi @brianignacio5 

I tired to create one new blink example project, it works well, but the previous example project still cannot work. However, the commands you mentioned above can work well , the details as below:

```c
PS F:\Project\25KB_1S_WiFi_C3>  F:/Tools/Espressif/IDF_Tools/.espressif/python_env/idf4.4_py3.8_env/Scripts/python.exe F:\BLE_WIFI\Espressif\SDK\esp-idf-c3\tools\idf_size.py .\build\esp32_c3_uart_tcp_25KB.map --json 
{
    "dram_data": 0,
    "dram_bss": 0,
    "dram_rodata": 0,
    "dram_other": 0,
    "used_dram": 0,
    "dram_total": 0,
    "used_dram_ratio": NaN,
    "dram_remain": 0,
    "iram_vectors": 0,
    "iram_text": 0,
    "iram_other": 0,
    "used_iram": 0,
    "iram_total": 0,
    "used_iram_ratio": NaN,
    "iram_remain": 0,
    "diram_data": 10956,
    "diram_bss": 16200,
    "diram_text": 58880,
    "diram_vectors": 0,
    "diram_rodata": 0,
    "diram_other": 0,
    "diram_total": 327680,
    "used_diram": 86036,
    "used_diram_ratio": 0.26256103515625,
    "diram_remain": 241644,
    "flash_code": 513344,
    "flash_rodata": 88146,
    "flash_other": 524544,
    "used_flash_non_ram": 1126034,       
    "total_size": 1195870
}
```

it is the same as other commands, why?

brianignacio5, 2021-09-14:
@xiaolongba it seems this is an ESP-IDF since as you can `NaN` is printed and it shouldn't.

We are working on a fix for this and will get back to you soon!

xiaolongba, 2021-09-14:
@brianignacio5 Ok, thanks, waiting for your news.

brianignacio5, 2021-09-26:
This should be fixed with `8d74bf87ec005a4b3ee987641963a29ae9adf564` in ESP-IDF master.

Please give it a try @xiaolongba

xiaolongba, 2021-09-26:
HI @brianignacio5 Bro,

I updated the esp-idf  and tried again, as seen from the log information, it seems to works, the details are as below:

```json
PS F:\Project\25KB_1S_WiFi_C3_Tcp_Client_Station> F:/Tools/Espressif/IDF_Tools/.espressif/python_env/idf4.4_py3.8_env/Scripts/python.exe F:\BLE_WIFI\Espressif\SDK\esp-idf\tools\idf_size.py .\build\esp32_c3_uart_tcp_client_25KB.map --json
{
    "dram_data": 0,  
    "dram_bss": 0,   
    "dram_rodata": 0,
    "dram_other": 0, 
    "used_dram": 0,
    "dram_total": 0,
    "used_dram_ratio": 0,
    "dram_remain": 0,
    "iram_vectors": 0,
    "iram_text": 0,
    "iram_other": 0,
    "used_iram": 0,
    "iram_total": 0,
    "used_iram_ratio": 0,
    "iram_remain": 0,
    "diram_data": 10956,
    "diram_bss": 16208,
    "diram_text": 58880,
    "diram_vectors": 0,
    "diram_rodata": 0,
    "diram_other": 0,
    "diram_total": 327680,
    "used_diram": 86044,
    "used_diram_ratio": 0.26258544921875,
    "diram_remain": 241636,
    "flash_code": 514752,
    "flash_rodata": 90722,
    "flash_other": 524544,
    "used_flash_non_ram": 1130018,
    "total_size": 1199854
}
```

however, the test result is different from the outputted information from esp-idf extension as below:
[image]

could you help me double-check?

thanks in advance.

brianignacio5, 2021-09-26:
I'll make a fix for the NaN KB result you see but basically values in terminal are bytes and those in the UI are in KB

xiaolongba, 2021-09-26:
understood, waiting for your good news.

thanks again, bro

brianignacio5, 2021-09-28:
Can you try #535 or this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/96978591.zip) ?

xiaolongba, 2021-09-28:
Hi @brianignacio5 bro,

I tried to update the extension, but the results are the same as before, the details as below:
[image]

brianignacio5, 2021-09-28:
How did you update the extension? Make sure to delete the extension folder before installing the new vsix installer

xiaolongba, 2021-09-28:
> How did you update the extension? Make sure to delete the extension folder before installing the new vsix installer

I delete it with the following steps, pls check it,
[image]

brianignacio5, 2021-09-30:
Thanks for testing it and the feedback. Will take a look at the production version to see what is happening.

xiaolongba, 2021-09-30:
Okay, could you share with me the production version for the further test first? I will give you feedback soon as I am almost working all day.

brianignacio5, 2021-10-08:
Maybe it is fixed with this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/97958316.zip) ?

xiaolongba, 2021-10-12:
Hi @igrr Bro,

many thanks for your detailed replies. regarding the concept of `dram*` and `diram*`, I am clear now. but I still think the content of the display is unreasonable. as I mentioned above

> Hi @brianignacio5 Bro,
> 
> Indeed, the NaN has gone, but the **.bss & .data** are always 0KB, could you double-check? I used the blink example project for testing. [image]

the total size is 303KB, FLASH CODE is 95KB and FLASH RODATA is 30KB, but where is the 303-95-30KB coming from? especially the DATA & BSS size is 0KB, as the red circle highlighted, the `diram_data `is 5214bytes, while `diram_bss `is 3712bytes, it is strange that the bar showed me that **.bss & .data** are always 0KB.

in my understanding, the bar shall show what is ram sizes and flash sizes are used. that will be more intuitive. at least, it should tell users what is dram,iram, diram and flash szie.

PR #535 (2021-11-24): fix divide zero value on idf size ui
Fix #527 

Add support for older and newer versions of idf_size.py json fields.