espressif/vscode-esp-idf-extension #530: Failed to install tools on Windows

meltdown03, 2021-09-18:
I'm on Windows 11 Preview, not sure if maybe that's the issue, but I cannot get the virtualenv created using the include `idf-python` version (3.8.7). I get an error stating that it's missing a `DLLs` folder.
Is there a way to use my systems Python version instead? I am able to create a virtualenv with it from a terminal.

```Using existing Toolchain for Xtensa (ESP32) based on GCC in C:\Users\User\.espressif\tools\xtensa-esp32-elf\esp-2021r1-8.4.0
Using existing Toolchain for Xtensa (ESP32-S2) based on GCC in C:\Users\User\.espressif\tools\xtensa-esp32s2-elf\esp-2021r1-8.4.0
Using existing Toolchain for Xtensa (ESP32-S3) based on GCC in C:\Users\User\.espressif\tools\xtensa-esp32s3-elf\esp-2021r1-8.4.0
Using existing Toolchain for 32-bit RISC-V based on GCC in C:\Users\User\.espressif\tools\riscv32-esp-elf\esp-2021r1-8.4.0
Using existing Toolchain for ESP32 ULP coprocessor in C:\Users\User\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205
Using existing Toolchain for ESP32-S2 and ESP32-S3 ULP coprocessors in C:\Users\User\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205
Using existing CMake build system in C:\Users\User\.espressif\tools\cmake\3.20.3
Using existing OpenOCD for ESP32 in C:\Users\User\.espressif\tools\openocd-esp32\v0.10.0-esp32-20210902
Using existing Ninja build system in C:\Users\User\.espressif\tools\ninja\1.10.2
Using existing IDF wrapper tool for Windows in C:\Users\User\.espressif\tools\idf-exe\1.0.1
Using existing Ccache (compiler cache) in C:\Users\User\.espressif\tools\ccache\4.3
Using existing dfu-util (Device Firmware Upgrade Utilities) in C:\Users\User\.espressif\tools\dfu-util\0.9

FileNotFoundError: [WinError 3] The system cannot find the path specified: 'C:\\Users\\User\\.espressif\\tools\\idf-python\\3.8.7\\DLLs'
Creating a new Python environment in C:\Users\User\.espressif\python_env\idf4.4_py3.8_env
Traceback (most recent call last):
  File "C:\Users\User\esp-idf\tools\idf_tools.py", line 1817, in <module>
    main(sys.argv[1:])
  File "C:\Users\User\esp-idf\tools\idf_tools.py", line 1813, in main
    action_func(args)
  File "C:\Users\User\esp-idf\tools\idf_tools.py", line 1517, in action_install_python_env
    subprocess.check_call([sys.executable, '-m', 'virtualenv', idf_python_env_path],
  File "subprocess.py", line 364, in check_call
subprocess.CalledProcessError: Command '['C:\\Users\\User\\.espressif\\tools\\idf-python\\3.8.7\\python.exe', '-m', 'virtualenv', 'C:\\Users\\User\\.espressif\\python_env\\idf4.4_py3.8_env']' returned non-zero exit status 1.
Command failed: C:\Users\User\.espressif\tools\idf-python\3.8.7\python.exe C:\Users\User\esp-idf\tools\idf_tools.py install-python-env
Traceback (most recent call last):
  File "C:\Users\User\esp-idf\tools\idf_tools.py", line 1817, in <module>
    main(sys.argv[1:])
  File "C:\Users\User\esp-idf\tools\idf_tools.py", line 1813, in main
    action_func(args)
  File "C:\Users\User\esp-idf\tools\idf_tools.py", line 1517, in action_install_python_env
    subprocess.check_call([sys.executable, '-m', 'virtualenv', idf_python_env_path],
  File "subprocess.py", line 364, in check_call
subprocess.CalledProcessError: Command '['C:\\Users\\User\\.espressif\\tools\\idf-python\\3.8.7\\python.exe', '-m', 'virtualenv', 'C:\\Users\\User\\.espressif\\python_env\\idf4.4_py3.8_env']' returned non-zero exit status 1.
Command failed: C:\Users\User\.espressif\tools\idf-python\3.8.7\python.exe C:\Users\User\esp-idf\tools\idf_tools.py install-python-env
Traceback (most recent call last):
  File "C:\Users\User\esp-idf\tools\idf_tools.py", line 1817, in <module>
    main(sys.argv[1:])
  File "C:\Users\User\esp-idf\tools\idf_tools.py", line 1813, in main
    action_func(args)
  File "C:\Users\User\esp-idf\tools\idf_tools.py", line 1517, in action_install_python_env
    subprocess.check_call([sys.executable, '-m', 'virtualenv', idf_python_env_path],
  File "subprocess.py", line 364, in check_call
subprocess.CalledProcessError: Command '['C:\\Users\\User\\.espressif\\tools\\idf-python\\3.8.7\\python.exe', '-m', 'virtualenv', 'C:\\Users\\User\\.espressif\\python_env\\idf4.4_py3.8_env']' returned non-zero exit status 1.
```

brianignacio5, 2021-09-18:
This is actually an issue with ESP-IDF embedded python itself. Need to see if the embedded python packages the necessary DLLs for Windows 11

meltdown03, 2021-09-18:
I ended up making the virtualenv manually with my systems Python. Then when the extension was trying to create it, I just hit cancel (it was confused and just sitting there spinning, lol) and it went to the completed page. It works now.

meltdown03, 2021-09-18:
> This is actually an issue with ESP-IDF embedded python itself. Need to see if the embedded python packages the necessary DLLs for Windows 11

The zip file it downloads does not have a DLLs folder, btw. (if that is what it requires)

brianignacio5, 2021-09-18:
Thanks for letting us know @meltdown03 ! Will review it and work on it

meltdown03, 2021-09-18:
I installed the esp-idf tools online installer for windows which has the same embedded python and also creates the virtualenv. It ended up working fine. I see it runs the same command to setup the virtualenv: `C:\Users\User\.espressif\tools\idf-python\3.8.7\python.exe -m virtualenv "C:\Users\User\.espressif\python_env\idf4.4_py3.8_env" -p "C:\Users\User\.espressif\tools\idf-python\3.8.7\python.exe"` (not sure what the -p does)
Once it's installed and I check the box in the installer to open an ESP-IDF Powershell or Command Prompt. I can delete the virtualenv and recreate it manually by running the virtualenv command. If I exit that prompt and start a new one, it fails. So weird! It might be setting some required environment variable before running the virtualenv command.

meltdown03, 2021-09-18:
Here is the log of the installer. It is doing a lot of importing of DLLs:
```
2021-09-18 09:45:02.246   Log opened. (Time zone: UTC-05:00)
2021-09-18 09:45:02.246   Setup version: Inno Setup version 6.2.0
2021-09-18 09:45:02.247   Original Setup EXE: C:\Users\User\Downloads\esp-idf-tools-setup-online-2.10.exe
2021-09-18 09:45:02.247   Setup command line: /SL5="$18064C,2455762,831488,C:\Users\User\Downloads\esp-idf-tools-setup-online-2.10.exe" 
2021-09-18 09:45:02.247   Windows version: 10.0.22458  (NT platform: Yes)
2021-09-18 09:45:02.247   64-bit Windows: Yes
2021-09-18 09:45:02.247   Processor architecture: x64
2021-09-18 09:45:02.247   User privileges: None
2021-09-18 09:45:02.256   Administrative install mode: No
2021-09-18 09:45:02.257   Install mode root key: HKEY_CURRENT_USER
2021-09-18 09:45:02.257   64-bit install mode: Yes
2021-09-18 09:45:02.270   Created temporary directory: C:\Users\User\AppData\Local\Temp\is-35H5Q.tmp
2021-09-18 09:45:02.279   -- DLL function import --
2021-09-18 09:45:02.279   Function name: idpAddFile
2021-09-18 09:45:02.279   DLL name: files:idp.dll
2021-09-18 09:45:02.279   Extracting temporary file: C:\Users\User\AppData\Local\Temp\is-35H5Q.tmp\idp.dll
2021-09-18 09:45:02.295   Dest DLL name: C:\Users\User\AppData\Local\Temp\is-35H5Q.tmp\idp.dll
2021-09-18 09:45:02.295   Importing the DLL function.
2021-09-18 09:45:02.404   Successfully imported the DLL function. Delay loaded? No
2021-09-18 09:45:02.404   -- DLL function import --
2021-09-18 09:45:02.404   Function name: idpAddFileComp
2021-09-18 09:45:02.404   DLL name: files:idp.dll
2021-09-18 09:45:02.404   Dest DLL name: C:\Users\User\AppData\Local\Temp\is-35H5Q.tmp\idp.dll
2021-09-18 09:45:02.404   Importing the DLL function.
2021-09-18 09:45:02.404   Successfully imported the DLL function. Delay loaded? No
2021-09-18 09:45:02.405   -- DLL function import --
2021-09-18 09:45:02.405   Function name: idpAddMirror
2021-09-18 09:45:02.405   DLL name: files:idp.dll
2021-09-18 09:45:02.405   Dest DLL name: C:\Users\User\AppData\Local\Temp\is-35H5Q.tmp\idp.dll
2021-09-18 09:45:02.405   Importing the DLL function.
2021-09-18 09:45:02.405   Successfully imported the DLL function. Delay loaded? No
2021-09-18 09:45:02.405   -- DLL function import --
2021-09-18 09:45:02.405   Function name: idpAddFtpDir
2021-09-18 09:45:02.405   DLL name: files:idp.dll
2021-09-18 09:45:02.405   Dest DLL name: C:\Users\User\AppData\Local\Temp\is-35H5Q.tmp\idp.dll
2021-09-18 09:45:02.405   Importing the DLL function.
2021-09-18 09:45:02.405   Successfully imported the DLL function. Delay loaded? No
2021-09-18 09:45:02.405   -- DLL function import --
2021-09-18 09:45:02.405   Function name: idpAddFtpDirComp
2021-09-18 09:45:02.405   DLL name: files:idp.dll
2021-09-18 09:45:02.405   Dest DLL name: C:\Users\User\AppData\Local\Temp\is-35H5Q.tmp\idp.dll
2021-09-18 09:45:02.405   Importing the DLL function.
2021-09-18 09:45:02.405   Successfully imported the DLL function. Delay loaded? No
2021-09-18 09:45:02.405   -- DLL function import --
2021-09-18 09:45:02.405   Function name: idpClearFiles
2021-09-18 09:45:02.405   DLL name: files:idp.dll
2021-09-18 09:45:02.406   Dest DLL name: C:\Users\User\AppData\Local\Temp\is-35H5Q.tmp\idp.dll
2021-09-18 09:45:02.406   Importing the DLL function.
2021-09-18 09:45:02.406   Successfully imported the DLL function. Delay loaded? No
2021-09-18 09:45:02.406   -- DLL function import --
2021-09-18 09:45:02.406   Function name: idpFilesCount
2021-09-18 09:45:02.406   DLL name: files:idp.dll
2021-09-18 09:45:02.406   Dest DLL name: C:\Users\User\AppData\Local\Temp\is-35H5Q.tmp\idp.dll
2021-09-18 09:45:02.406   Importing the DLL function.
2021-09-18 09:45:02.406   Successfully imported the DLL function. Delay loaded? No
2021-09-18 09:45:02.406   -- DLL function import --
2021-09-18 09:45:02.406   Function name: idpFtpDirsCount
2021-09-18 09:45:02.406   DLL name: files:idp.dll
2021-09-18 09:45:02.406   Dest DLL name: C:\Users\User\AppData\Local\Temp\is-35H5Q.tmp\idp.dll
2021-09-18 09:45:02.406   Importing the DLL function.
2021-09-18 09:45:02.406   Successfully imported the DLL function. Delay loaded? No
2021-09-18 09:45:02.406   -- DLL function import --
2021-09-18 09:45:02.406   Function name: idpFileDownloaded
2021-09-18 09:45:02.406   DLL name: files:idp.dll
2021-09-18 09:45:02.406   Dest DLL name: C:\Users\User\AppData\Local\Temp\is-35H5Q.tmp\idp.dll
2021-09-18 09:45:02.407   Importing the DLL function.
2021-09-18 09:45:02.407   Successfully imported the DLL function. Delay loaded? No
2021-09-18 09:45:02.407   -- DLL function import --
[... 187 lines cut ...]
2021-09-18 09:45:02.758   Configuration /IDFVERSION=
2021-09-18 09:45:02.758   Configuration /IDFVERSIONSURL=https://dl.espressif.com/dl/esp-idf/idf_versions.txt
2021-09-18 09:45:02.758   Configuration /OFFLINE=no
2021-09-18 09:45:02.758   Configuration /PYTHONNOUSERSITE=yes
2021-09-18 09:45:02.758   Configuration /PYTHONWHEELSURL=https://dl.espressif.com/pypi
2021-09-18 09:45:02.759   Configuration /PYTHONWHEELSVERSION=3.8-2021-06-15
2021-09-18 09:45:02.759   Configuration /SKIPSYSTEMCHECK=no
2021-09-18 09:45:02.759   Configuration /USEEMBEDDEDGIT=yes
2021-09-18 09:45:02.759   Configuration /USEEMBEDDEDPYTHON=yes
2021-09-18 09:45:02.766   Extracting temporary file: C:\Users\User\AppData\Local\Temp\is-35H5Q.tmp\idf-env.exe
[... cut ...]
2021-09-18 09:45:18.023    [OK]
2021-09-18 09:45:18.026   Executing: C:\Users\User\AppData\Local\Temp\is-35H5Q.tmp\idf-env.exe antivirus get --property displayName
2021-09-18 09:45:18.693   Result: Windows Defender
2021-09-18 09:45:18.693   Executing: C:\Users\User\AppData\Local\Temp\is-35H5Q.tmp\idf-env.exe antivirus get --property displayName
2021-09-18 09:45:18.917   Result: Windows Defender
2021-09-18 09:45:18.917   * Detected antivirus: Windows Defender
2021-09-18 09:45:18.920   * Check complete.
```

brianignacio5, 2021-09-23:
Seems this an issue with virtualenv itself as shown here https://github.com/espressif/esp-idf/issues/6509

Will add some fix for this in the extension to pass the python executable path to virtualenv

brianignacio5, 2021-09-28:
Can you try #536 or this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/97031245.zip) ?

meltdown03, 2021-09-29:
It still doesn't work in Windows 11:

```
FileNotFoundError: [WinError 3] The system cannot find the path specified: 'C:\\Users\\User\\.espressif\\tools\\idf-python\\3.8.7\\DLLs'
Command failed: "C:\Users\User\.espressif\tools\idf-python\3.8.7\python.exe" -m virtualenv "C:\Users\User\.espressif\python_env\idf4.4_py3.8_env" -p "C:\Users\User\.espressif\tools\idf-python\3.8.7\python.exe"
Command failed: "C:\Users\User\.espressif\tools\idf-python\3.8.7\python.exe" -m virtualenv "C:\Users\User\.espressif\python_env\idf4.4_py3.8_env" -p "C:\Users\User\.espressif\tools\idf-python\3.8.7\python.exe"
```

meltdown03, 2021-09-29:
Here is the extension log:
[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/7255247/esp_idf_vsc_ext.log)

meltdown03, 2021-10-14:
> The issue is already closed. Just to mention it here.
> The installer has support for using custom Python. You can start the
> installer with following option:
>
> idf-installer...exe /USEEMBEDDEDPYTHON=no
>
> This will enable check of available Python's. There's also check for
> problem with subprocess.py which might be some remain of previous Python
> installation.

Cool, thanks. I'd rather use my version