espressif/vscode-esp-idf-extension #532: [Bug Report]: Build icon not building

isarychkin, 2021-09-25:
when pressing build icon in the extension's task bar terminal opens up with the following:
########################################################################
> Executing task: cmake --build . <
Setting IDF_PATH: C:\esp\esp-idf
Adding ESP-IDF tools to PATH...
Name                           Value
----                           -----
IDF_PYTHON_ENV_PATH            C:\esp\.espressif\python_env\idf4.4_py3.8_env
Added to PATH
-------------
C:\esp\esp-idf\components\app_update
C:\esp\.espressif\python_env\idf4.4_py3.8_env\Scripts
%PATH%
Checking if Python packages are up to date...
Python requirements from C:\esp\esp-idf\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:
    idf.py build
###########################################################
Then exits to the terminal's prompt in the project's build folder.  At the same time a notification shows "Building project..." and nothing happens.  Typing "idf.py build" in the same terminal window builds the project OK.

**Environment :**

- OS Name & Version: Windows 10
- VSCode Version: 1.60.2
- ESP-IDF Version: v4.4-dev-3235-g3e370c4296
- Python Version: 3.8.7

> `ESP-IDF: Doctor command` REPORT:

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19043 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files\Common Files\Oracle\Java\javapath;C:\Python3\Scripts\;C:\Python3\;C:\Program Files (x86)\Microsoft SDKs\Azure\CLI2\wbin;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\ProgramData\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\PuTTY\;C:\ProgramData\chocolatey\bin;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\dotnet\;C:\Program Files (x86)\Microsoft SQL Server\150\DTS\Binn\;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\130\Tools\Binn\;C:\Program Files (x86)\Microsoft SQL Server\140\Tools\Binn\;C:\Program Files\Microsoft SQL Server\140\Tools\Binn\;C:\Program Files\Microsoft SQL Server\140\DTS\Binn\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\Git\cmd;C:\npm;C:\Program Files\nodejs\;C:\Program Files\PowerShell\7\;C:\Program Files\CMake\bin;C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.07\bin;C:\Users\myUser Name\AppData\Local\Programs\Python\Python39\Scripts\;C:\Users\myUser Name\AppData\Local\Programs\Python\Python39\;C:\Users\myUser Name\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\myUser Name\AppData\Roaming\npm;C:\Users\myUser Name\AppData\Local\GitHubDesktop\bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.60.2 
Visual Studio Code language en 
Visual Studio Code shell powershell.exe 
ESP-IDF Extension version 1.2.0 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:/esp/esp-idf/
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
Custom extra paths (idf.customExtraPaths) c:\esp\.espressif\tools\xtensa-esp32-elf\esp-2021r1-8.4.0\xtensa-esp32-elf\bin;c:\esp\.espressif\tools\xtensa-esp32s2-elf\esp-2021r1-8.4.0\xtensa-esp32s2-elf\bin;c:\esp\.espressif\tools\xtensa-esp32s3-elf\esp-2021r1-8.4.0\xtensa-esp32s3-elf\bin;c:\esp\.espressif\tools\riscv32-esp-elf\esp-2021r1-8.4.0\riscv32-esp-elf\bin;c:\esp\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;c:\esp\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;c:\esp\.espressif\tools\cmake\3.20.3\bin;c:\esp\.espressif\tools\openocd-esp32\v0.10.0-esp32-20210902\openocd-esp32\bin;c:\esp\.espressif\tools\ninja\1.10.2;c:\esp\.espressif\tools\idf-exe\1.0.1;c:\esp\.espressif\tools\ccache\4.3\ccache-4.3-windows-64;c:\esp\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"c:\\esp\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20210902/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) C:/esp/.espressif/python_env/idf4.4_py3.8_env/Scripts/python.exe
Serial port (idf.port) COM5
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,target/esp32.cfg
ESP-IDF Tools Path (idf.toolsPath) c:\esp\.espressif
Git Path (idf.gitPath) C:/esp/.espressif/tools/idf-git/2.30.1/cmd/git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-IDF Custom extra paths
Access to c:\esp\.espressif\tools\xtensa-esp32-elf\esp-2021r1-8.4.0\xtensa-esp32-elf\bin: true
Access to c:\esp\.espressif\tools\xtensa-esp32s2-elf\esp-2021r1-8.4.0\xtensa-esp32s2-elf\bin: true
Access to c:\esp\.espressif\tools\xtensa-esp32s3-elf\esp-2021r1-8.4.0\xtensa-esp32s3-elf\bin: true
Access to c:\esp\.espressif\tools\riscv32-esp-elf\esp-2021r1-8.4.0\riscv32-esp-elf\bin: true
Access to c:\esp\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin: true
Access to c:\esp\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin: true
Access to c:\esp\.espressif\tools\cmake\3.20.3\bin: true
Access to c:\esp\.espressif\tools\openocd-esp32\v0.10.0-esp32-20210902\openocd-esp32\bin: true
Access to c:\esp\.espressif\tools\ninja\1.10.2: true
Access to c:\esp\.espressif\tools\idf-exe\1.0.1: true
Access to c:\esp\.espressif\tools\ccache\4.3\ccache-4.3-windows-64: true
[... 50 lines cut ...]
reedsolo version: 1.5.4
requests version: 2.26.0
requests-toolbelt version: 0.9.1
schema version: 0.7.4
semantic-version version: 2.8.5
setuptools version: 58.1.0
six version: 1.16.0
tqdm version: 4.62.3
urllib3 version: 1.26.7
websocket-client version: 1.2.1
Werkzeug version: 0.16.1
wheel version: 0.36.2
windows-curses version: 2.2.0
xmlrunner version: 1.7.7
---------------------------------------------------- Check ESP-IDF python requirements.txt -------------------------------------------------
Check ESP-IDF Python packages Python requirements from C:\esp\esp-idf\requirements.txt are satisfied.
---------------------------------------------------- Check extension requirements.txt ------------------------------------------------------
Check Extension Python packages Python requirements from c:\Users\Igor Sarychkin\.vscode\extensions\espressif.esp-idf-extension-1.2.0\requirements.txt are satisfied.
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages Python requirements from c:\Users\Igor Sarychkin\.vscode\extensions\espressif.esp-idf-extension-1.2.0\esp_debug_adapter\requirements.txt are satisfied.
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at ENOENT: no such file or directory, open 'c:\esp\hello_world\.vscode\launch.json'
--------------------------------------------------------------------------------------------------------------------------------------------

Click the menu `View` and select `Output`. This extension uses several output channels: ESP-IDF, ESP-IDF Debug Adapter, Heap Trace, OpenOCD and SDK Configuration Editor.

brianignacio5, 2021-09-27:
I'm unable to reproduce your issue but it might be related to spaces in PATH. ESP-IDF still have issue to find tools when there spaces in PATH.

isarychkin, 2021-09-28:
@brianignacio5  I removed v1.20 and installed v.1.00 (7 months old) - all works as expected.  Then upgraded to 1.20 and the "build" icon is broken again.

brianignacio5, 2021-10-08:
I remembered we used that custom terminal because of an issue with PlatformIO modifying the environment PATH.

Can you try disabling other extensions and see if this fixes your issue ?

MartyMcFlyInTheSky, 2022-02-09:
I have the same issue since some time already:

My settings.json looks like this (I have sys env variables configured)

Excerpt from workspace settings.json:

```
{
   ...
    "idf.adapterTargetName": "esp32",
    "idf.espIdfPath": "${env:IDF_PATH}",
    "idf.toolsPathWin": "${env:IDF_TOOLS_PATH}",
    "idf.customExtraPaths": "C:/dev/esp/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin;C:/dev/esp/.espressif/tools/xtensa-esp32s2-elf/esp-2021r1-8.4.0/xtensa-esp32s2-elf/bin;C:/dev/esp/.espressif/tools/xtensa-esp32s3-elf/esp-2021r1-8.4.0/xtensa-esp32s3-elf/bin;C:/dev/esp/.espressif/tools/riscv32-esp-elf/esp-2021r1-8.4.0/riscv32-esp-elf/bin;C:/dev/esp/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin;C:/dev/esp/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin;C:/dev/esp/.espressif/tools/cmake/3.16.4/bin;C:/dev/esp/.espressif/tools/openocd-esp32/v0.10.0-esp32-20210401/openocd-esp32/bin;C:/dev/esp/.espressif/tools/ninja/1.10.2;C:/dev/esp/.espressif/tools/idf-exe/1.0.1;C:/dev/esp/.espressif/tools/ccache/3.7;C:/dev/esp/.espressif/tools/dfu-util/0.9/dfu-util-0.9-win64",
    "idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"C:/dev/esp/.espressif/tools/openocd-esp32/v0.10.0-esp32-20210401/openocd-esp32/share/openocd/scripts\",\"IDF_CCACHE_ENABLE\":\"1\"}",
    "idf.pythonBinPathWin": "C:/dev/esp/.espressif/python_env/idf4.4_py3.8_env/Scripts/python.exe",
    "idf.openOcdConfigs": [
        "interface/ftdi/esp32_devkitj_v1.cfg",
        "target/esp32.cfg"
    ],
    "idf.portWin": "COM5",
   ...
}
```

As Isarychkin mentioned, hitting the build-Icon just launches export.bat and doesn't subsequently trigger the build (it just displays the msg window forever). Building via idf.py build from console directly works well.

Note that I do not use PlatformIO