espressif/vscode-esp-idf-extension #539: [Bug Report]: Configuring extension fails with error "certificate has expired"

sinope-db, 2021-09-30:
**Describe the bug**

I get Error: certificate has expired errors when configuring the extension and/or installing IDF and toolchain.

Everything was working this morning, and I went to reconfigure to change path and it started to fail.
I reinstalled VS code, removed and reinstalled the extension with same result.

After much troubleshooting, it seems it could be related to DST Root CA X3 expiration (Let's Encrypt root). It expired this morning (local time).

**To Reproduce**
From a clean install, just execute ESP-IDF Configure ESP-IDF extension command and watch 

**Environment (please complete the following information):**
```
OS win32 x64 6.1.7601 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.60.2 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\cmd.exe 
ESP-IDF Extension version 1.2.0 
```
Also saw same error on colleague running Win10.

**Output**

channel ESP-IDF:
```
Error: certificate has expired
	at TLSSocket.onConnectSecure (_tls_wrap.js:1497:34)
	at TLSSocket.emit (events.js:315:20)
	at TLSSocket._finishInit (_tls_wrap.js:932:8)
	at TLSWrap.ssl.onhandshakedone (_tls_wrap.js:706:12)
Error opening esp-idf version list file. HTTP/HTTPS Request error https://dl.espressif.com/dl/esp-idf/idf_versions.txt
```

**Logs**
[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/7262001/esp_idf_vsc_ext.log)

**Extension Configuration settings**
    "idf.adapterTargetName": "esp32",
    "idf.gitPath": "C:\\Users\\X\\.espressif\\tools\\idf-git\\2.30.1\\cmd\\git.exe",
    "idf.pythonBinPathWin": ""

sinope-db, 2021-10-01:
Additional info: 
- I can download the file (https://dl.espressif.com/dl/esp-idf/idf_versions.txt) from a browser just fine.
- If I change the date on the computer before  DST Root CA X3 expiration, it works.

prying, 2021-10-01:
I get the same issue which by setting the date to the 29th of the 9th can be resolved. 
Thanks for the workaround while the cirtificates expiration date is being fixed.

zhengyangliu, 2021-10-02:
> 我遇到了同样的问题，通过将日期设置为 9 日的 29 日可以解决。 感谢您在 cirtificates 到期日期修复期间的解决方法。

Same error for me. Set the system date to 9.29 will solve this error.

vlaskz, 2021-10-04:
> > 我遇到了同样的问题，通过将日期设置为 9 日的 29 日可以解决。 感谢您在 cirtificates 到期日期修复期间的解决方法。
> 
> Same error for me. Set the system date to 9.29 will solve this error.

In fact, it solves the problem. But Espressif must renew this certificate soon as possible...

zhengyangliu, 2021-10-04:
> > > 我遇到了同样的问题，通过将日期设置为 9 日的 29 日可以解决。 感谢您在 cirtificates 到期日期修复期间的解决方法。
> > 
> > 
> > Same error for me. Set the system date to 9.29 will solve this error.
> 
> In fact, it solves the problem. But Espressif must renew this certificate soon as possible...

Yes, after download these content. I have to synchronize the date again to download the remaining resources. Otherwise it will get stuck on the next download and installation step。

sinope-db, 2021-10-05:
Issue seems resolved now, looks like Espressif updated the certificate chain on their server

closed 2021-10-05 by sinope-db