espressif/vscode-esp-idf-extension #542: [Feature Request]: ESP32-S3-DevkitC-1 Debugging Support

meltdown03, 2021-10-10:
I manually changed the "idf.openOcdConfigs" in the extension.ts** file, line 1344 to `board/esp32s3-builtin.cfg` since it has built-in Jtag. I also added the classes in the esp_debug_adapter `hw_specific/esp.py` file, copying the original esp32. It works until it gets to an unknown symbol, which when I looked up is at 0x400011e8 (memset). It won't let me step anymore. It also **does** this in the command-line version of xtensa-esp32s3-elf-gbd so it's not a problem with the debug adapter. 
So it's just these few changes to get the ESP32-S3 added to this:
[vscode-ext.txt](https://github.com/espressif/vscode-esp-idf-extension/files/7317645/vscode-ext.txt)
[esp_debug_adapter.txt](https://github.com/espressif/vscode-esp-idf-extension/files/7317646/esp_debug_adapter.txt)

meltdown03, 2021-10-21:
> Thanks @meltdown03 for your changes. Will update soon. You can manually modify `idf.openOcdConfigs` in settings.json or use the `Select OpenOCD Board Configuration` command which uses the information from openOCD boards directly.
> 
> Can you give a little more details about the unknown symbol issue you are having here ? ESP-IDF version, minimal code to reproduce, etc.

Yeah, now i cannot even get OpenOCD to start, before it worked until it got to the `memset` symbol. It did it within your extension and running GDB and OpenOCD from the command-line so I don't think it was an issue with the extension. I attached the OpenOCD log I get know if maybe you have some thoughts. Also, I can no longer view the serial monitor when using the USB/JTAG port. It disconnects immediately and disappears from the system (ArchLinux) forcing me to reset the ESP32-S3. This all seemed to happen after I tried to set the default Console log output to use the USB/JTAG Controller instead of the UART0.
[openocd.log](https://github.com/espressif/vscode-esp-idf-extension/files/7385563/openocd.log)

meltdown03, 2021-11-06:
Is the esp-debug-adapter getting updated to support the ESP32-S3? I added the classes locally on mine and they worked up until recently when all debugging broke on the S3. Changes to OpenOCD and ESP-IDF are to blame I think.

meltdown03, 2021-11-06:
```log
Open On-Chip Debugger  v0.10.0-esp32-20210902 (2021-09-02-09:38)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

Info : only one transport option; autoselect 'jtag'
Warn : Transport "jtag" was already selected
WARNING: ESP flash support is disabled!
WARNING: ESP flash support is disabled!
force hard breakpoints
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32s3.cpu0: Debug controller was reset.
Info : esp32s3.cpu0: Core was reset.
Info : Listening on port 3333 for gdb connections
Info : esp32s3.cpu1: Debug controller was reset.
Info : esp32s3.cpu1: Core was reset.
Info : accepting 'gdb' connection on tcp/3333
Warn : No symbols for FreeRTOS!
Info : esp32s3.cpu0: Target halted, PC=0x42017976, debug_reason=00000000
Info : Set GDB target to 'esp32s3.cpu0'
Info : esp32s3.cpu1: Target halted, PC=0x42017976, debug_reason=00000000
Info : New GDB Connection: 1, Target esp32s3.cpu0, state: halted
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32s3.cpu0: Debug controller was reset.
Info : esp32s3.cpu0: Core was reset.
Info : esp32s3.cpu0: Target halted, PC=0x500000E7, debug_reason=00000000
Info : esp32s3.cpu0: Core was reset.
Info : esp32s3.cpu0: Target halted, PC=0x40000400, debug_reason=00000000
Info : esp32s3.cpu1: Debug controller was reset.
Info : esp32s3.cpu1: Core was reset.
Info : esp32s3.cpu1: Target halted, PC=0x40000400, debug_reason=00000000
esp
  esp apptrace [start file://<outfile> [poll_period [trace_size [stop_tmo [wait4halt
            [skip_size]]]]] | [stop] | [status] | [dump file://<outfile>]
  esp gcov [dump]
  esp semihost_basedir dir
  esp sysview [start file://<outfile1> [file://<outfile2>] [poll_period [trace_size
            [stop_tmo [wait4halt [skip_size]]]]] | [stop] | [status]
  esp sysview_mcore [start file://<outfile> [poll_period [trace_size [stop_tmo [wait4halt
            [skip_size]]]]] | [stop] | [status]
esp32
  esp32 smp [on|off]
  esp32 smp_gdb
  esp32 smp_off
  esp32 smp_on
esp32s3.cpu0
  esp32s3.cpu0 arp_examine ['allow-defer']
  esp32s3.cpu0 arp_halt
  esp32s3.cpu0 arp_halt_gdb
  esp32s3.cpu0 arp_poll
  esp32s3.cpu0 arp_reset
  esp32s3.cpu0 arp_waitstate
  esp32s3.cpu0 array2mem arrayname bitwidth address count
  esp32s3.cpu0 cget target_attribute
  esp32s3.cpu0 configure [target_attribute ...]
  esp32s3.cpu0 curstate
  esp32s3.cpu0 esp
    esp32s3.cpu0 esp apptrace [start file://<outfile> [poll_period [trace_size [stop_tmo [wait4halt
              [skip_size]]]]] | [stop] | [status] | [dump file://<outfile>]
    esp32s3.cpu0 esp gcov [dump]
    esp32s3.cpu0 esp semihost_basedir dir
    esp32s3.cpu0 esp sysview [start file://<outfile1> [file://<outfile2>] [poll_period [trace_size
[... 72 lines cut ...]
    esp32s3.cpu1 xtensa perfmon_enable <counter_id> <select> [mask] [kernelcnt] [tracelevel]
    esp32s3.cpu1 xtensa set_permissive [0|1]
    esp32s3.cpu1 xtensa smpbreak [none|breakinout|runstall] | [BreakIn] [BreakOut] [RunStallIn]
              [DebugModeOut]
    esp32s3.cpu1 xtensa tracedump <outfile1> <outfile2>
    esp32s3.cpu1 xtensa tracestart [pc <pcval>/[maskbitcount]] [after <n> [ins|words]]
    esp32s3.cpu1 xtensa tracestop
esp_get_mac [format]
espusbjtag
  espusbjtag log logfile.txt
  espusbjtag setio tdi tms tck trst srst
  espusbjtag tdo
program_esp <filename> [address] [verify] [reset] [exit] [compress] [restore_clock]
program_esp_bins <build_dir> flasher_args.json [verify] [reset] [exit] [compress]
          [restore_clock]
❌ Error: invalid subcommand "appimage_offset 0x10000"

Info : dropped 'gdb' connection
Info : accepting 'gdb' connection on tcp/3333
Info : New GDB Connection: 1, Target esp32s3.cpu0, state: halted
Info : esp32s3.cpu1: Debug controller was reset.
Info : esp32s3.cpu1: Core was reset.
```

Here's what I get when I try to debug now.

atanisoft, 2021-11-06:
@meltdown03 why did you close the issue? Support is not available (yet) in an official release of this plugin so I think it should remain open for now.

meltdown03, 2021-11-06:
Sorry, I must've clicked the wrong button, I didn't mean to close it.

brianignacio5, 2022-01-24:
Could you please try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/147436665.zip) ? With latest openOCD and this vsix installer esp32 S3 debugging should be available !

@meltdown03 @atanisoft

meltdown03, 2022-01-24:
> Could you please try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/147436665.zip) ? With latest openOCD and this vsix installer esp32 S3 debugging should be available !
> 
> @meltdown03 @atanisoft

Side note: I noticed that they changed the `requirements.txt` file to `requirements.core.txt` and `requirements.gdbgui.txt`, so I had to rename the file to reinstall the extension and then rename it back when installing the the tools.

OpenOCD starts fine with the S3, but not the debug adapter or GDB maybe:

``` python
2022-01-24 06:46:46,187 - Debug Adapter (main) - DEBUG - START_TIME_2022_01_24_06-46-46
2022-01-24 06:46:46,187 - Debug Adapter (main) - DEBUG - Init errors:None
2022-01-24 06:46:46,187 - Debug Adapter (main) - INFO - Working directory: /home/neal
2022-01-24 06:46:46,188 - Debug Adapter (main) - INFO - Arguments: 
{   'app_flash_off': 65536,
    'board_type': None,
    'cmdfile': None,
    'core_file': (),
    'debug': 5,
    'developer_mode': 'none',
    'device_name': 'esp32s3',
    'elfpath': (   '/home/neal/esp32s3_examples/nmea0183_parser/build/nmea_parser.elf',),
    'log_file': '/home/neal/esp32s3_examples/nmea0183_parser/debug.log',
    'log_mult_files': None,
    'log_no_debug_console': None,
    'oocd': 'openocd',
    'oocd_args': None,
    'oocd_ip': 'localhost',
    'oocd_mode': 'connect_to_instance',
    'oocd_scripts': None,
    'port': 43474,
    'postmortem': False,
    'tmo_scale_factor': 1,
    'toolchain_prefix': 'xtensa-esp32s3-elf-'}
DEBUG_ADAPTER_STARTED
2022-01-24 06:46:46,188 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STARTED
2022-01-24 06:46:46,188 - Debug Adapter (main) - INFO - Starting. Cmd: /home/neal/.vscode/extensions/espressif.esp-idf-extension-1.3.0/esp_debug_adapter/debug_adapter_main.py -d 5 -e /home/neal/esp32s3_examples/nmea0183_parser/build/nmea_parser.elf -l /home/neal/esp32s3_examples/nmea0183_parser/debug.log -p 43474 -dn esp32s3 -a 0x10000 -t xtensa-esp32s3-elf-

2022-01-24 06:46:46,190 - Debug Adapter (main) - INFO - Listening on port 43474
DEBUG_ADAPTER_READY2CONNECT
2022-01-24 06:46:46,190 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT
2022-01-24 06:46:46,192 - Debug Adapter (main) - DEBUG - Got connection
2022-01-24 06:46:46,193 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Debug adapter -> Extension: DEBUG_ADAPTER_STARTED\n", "source": {}}, "seq": 0}

2022-01-24 06:46:46,194 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Starting. Cmd: /home/neal/.vscode/extensions/espressif.esp-idf-extension-1.3.0/esp_debug_adapter/debug_adapter_main.py -d 5 -e /home/neal/esp32s3_examples/nmea0183_parser/build/nmea_parser.elf -l /home/neal/esp32s3_examples/nmea0183_parser/debug.log -p 43474 -dn esp32s3 -a 0x10000 -t xtensa-esp32s3-elf-\n\n", "source": {}}, "seq": 1}

2022-01-24 06:46:46,195 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Listening on port 43474\n", "source": {}}, "seq": 2}

2022-01-24 06:46:46,195 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT\n", "source": {}}, "seq": 3}

2022-01-24 06:46:46,195 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Got connection\n", "source": {}}, "seq": 4}

2022-01-24 06:46:46,198 - Debug Adapter (ReaderThread) - DEBUG - read line: >>b'Content-Length: 410\\r\\n'<<

2022-01-24 06:46:46,199 - Debug Adapter (ReaderThread) - DEBUG - read line: >>b'\\r\\n'<<

2022-01-24 06:46:46,199 - Debug Adapter(Command Processor) - DEBUG - Got json: {
    "arguments": {
        "adapterID": "espidf",
        "clientID": "vscode",
        "clientName": "Visual Studio Code",
        "columnsStartAt1": true,
        "linesStartAt1": true,
        "locale": "en-us",
        "pathFormat": "path",
        "supportsInvalidatedEvent": true,
        "supportsMemoryReferences": true,
        "supportsProgressReporting": true,
        "supportsRunInTerminalRequest": true,
        "supportsVariablePaging": true,
        "supportsVariableType": true
[... 78 lines cut ...]

2022-01-24 06:46:46,243 - Debug Adapter (main) - DEBUG - Adapter is Stopping...
2022-01-24 06:46:46,243 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Debug Adapter stopped\n", "source": {}}, "seq": 10}

2022-01-24 06:46:46,243 - Debug Adapter (main) - DEBUG - Stopping target poller
2022-01-24 06:46:46,244 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Adapter is Stopping...\n", "source": {}}, "seq": 11}

2022-01-24 06:46:46,244 - Debug Adapter (main) - DEBUG - Closing the GDB process
2022-01-24 06:46:46,244 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Stopping target poller\n", "source": {}}, "seq": 12}

2022-01-24 06:46:46,245 - Debug Adapter (main) - DEBUG - Closing the Socket
2022-01-24 06:46:46,245 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Closing the GDB process\n", "source": {}}, "seq": 13}

[... cut ...]
2022-01-24 06:46:46,246 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Stopping of the Writer thread\n", "source": {}}, "seq": 15}

2022-01-24 06:46:46,247 - Debug Adapter (main) - DEBUG - Stopping of the Reader thread
2022-01-24 06:46:46,247 - Debug Adapter (main) - DEBUG - Adapter is Stopped
DEBUG_ADAPTER_STOPPED
2022-01-24 06:46:46,247 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STOPPED
2022-01-24 06:46:46,247 - Debug Adapter(Command Processor) - DEBUG - Processed command: disconnect
[Stopped] : ESP-IDF Debug Adapter
```

meltdown03, 2022-01-24:
here is the esp_idf_extension.log contents when it fails:
``` json
{"message":"Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:42)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n","stack":"Error: Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:42)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n\n\tat /home/neal/.vscode/extensions/espressif.esp-idf-extension-1.3.0/dist/extension.js:1:7294\n\tat ChildProcess.exithandler (child_process.js:308:7)\n\tat ChildProcess.emit (events.js:315:20)\n\tat ChildProcess.EventEmitter.emit (domain.js:467:12)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)","level":"error","timestamp":"2022-01-24T13:03:02.690Z"}
{"user":true,"message":"OpenOCD Exit with non-zero error code 1","stack":"Error: Spawn exit with non-zero1\n\tat ChildProcess.<anonymous> (/home/neal/.vscode/extensions/espressif.esp-idf-extension-1.3.0/dist/extension.js:1:89697)\n\tat ChildProcess.emit (events.js:315:20)\n\tat ChildProcess.EventEmitter.emit (domain.js:467:12)\n\tat maybeClose (internal/child_process.js:1048:16)\n\tat Socket.<anonymous> (internal/child_process.js:439:11)\n\tat Socket.emit (events.js:315:20)\n\tat Socket.EventEmitter.emit (domain.js:467:12)\n\tat Pipe.<anonymous> (net.js:673:12)","level":"error","timestamp":"2022-01-24T13:03:06.081Z"}
{"level":"error","message":"Invalid error level 'error' for user notification. OpenOCD Exit with non-zero error code 1","timestamp":"2022-01-24T13:03:06.082Z"}
```

atanisoft, 2022-01-24:
@meltdown03 it looks like the gdb binary is expecting a python 2.7 version, which is unfortunate considering all other tools moved up to python 3.x some time ago. You will likely need to re-install python 2.7 for it to work.

atanisoft, 2022-01-24:
It seems I'm still facing issues with the S3 USB/JTAG and ESP-IDF, not sure what is causing it yet:
```
openocd -f board/esp32s3-builtin.cfg -d
Open On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:42)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 13 2 options.c:63 configuration_output_handler(): debug_level: 3
User : 14 2 options.c:63 configuration_output_handler(): 
...
Debug: 234 5 xtensa.c:2379 xtensa_build_reg_cache(): Special reg 'mesr' (160) does not exist
Debug: 235 5 xtensa.c:2379 xtensa_build_reg_cache(): Special reg 'mecr' (161) does not exist
Debug: 236 5 xtensa.c:2379 xtensa_build_reg_cache(): Special reg 'mevaddr' (162) does not exist
Error: 237 11 libusb_helper.c:381 jtag_libusb_get_serial(): libusb_get_string_descriptor_ascii() failed with -9
Debug: 238 12 command.c:629 run_command(): Command 'init' failed with error code -4
User : 239 12 command.c:694 command_run_line(): 
Debug: 240 12 esp_xtensa.c:131 esp_xtensa_target_deinit(): start
Debug: 241 12 xtensa.c:2533 xtensa_target_deinit(): start
openocd: ../src/jtag/core.c:343: jtag_checks: Assertion `jtag_trst == 0' failed.
Debug: 242 12 server.c:609 sig_handler(): Terminating on Signal 6
Aborted
```

Even `idf.py monitor` reports an odd failure:
```
idf.py monitor --port=/dev/ttyACM1
Executing action: monitor
Running idf_monitor in directory .../esp32cs
Executing "idf5.0_py3.9_env/bin/python .../idf_monitor.py -p /dev/ttyACM1 -b 115200 --toolchain-prefix  --target esp32s3 .../build/ESP32CommandStation.elf -m '.../idf5.0_py3.9_env/bin/python' '.../idf.py'"...
--- idf_monitor on /dev/ttyACM1 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 973, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 910, in run
    self._target(*self._args, **self._kwargs)
  File "/home/mdunston/esp-idf/tools/idf_monitor_base/stoppable_thread.py", line 58, in _run_outer
    self.run()
  File "/home/mdunston/esp-idf/tools/idf_monitor_base/serial_reader.py", line 51, in run
    self.serial.open()
  File ".../idf5.0_py3.9_env/lib/python3.9/site-packages/serial/serialposix.py", line 338, in open
    self._update_rts_state()
  File ".../idf5.0_py3.9_env/lib/python3.9/site-packages/serial/serialposix.py", line 708, in _update_rts_state
    fcntl.ioctl(self.fd, TIOCMBIC, TIOCM_RTS_str)
BrokenPipeError: [Errno 32] Broken pipe
```

I'm suspecting it is something specific to Linux but not 100% sure what yet as I don't see this sort of failure on the ESP32-S2 (still need to test ESP32-C3, my C3 DevKit's are rev2 without USB)

meltdown03, 2022-01-25:
> @meltdown03 it looks like the gdb binary is expecting a python 2.7 version, which is unfortunate considering all other tools moved up to python 3.x some time ago. You will likely need to re-install python 2.7 for it to work.

You were correct, I installed python2 and it worked! @brianignacio5

meltdown03, 2022-01-25:
> It seems I'm still facing issues with the S3 USB/JTAG and ESP-IDF, not sure what is causing it yet:
>
> ```
> openocd -f board/esp32s3-builtin.cfg -d
> Open On-Chip Debugger  v0.11.0-esp32-20211220 [...]

Are you using the latest master branch of IDF? I was getting those errors last month but some commits fixed that issue.

meltdown03, 2022-01-25:
I still get these errors when debugging :
``` json
2022-01-24 19:00:30,765 - Gdb - DEBUG - RESULT: error {'msg': 'Cannot find bounds of current function'}
2022-01-24 19:00:30,765 - Debug Adapter (main) - ERROR - Failed to step over!
Traceback (most recent call last):
  File "/home/neal/.vscode/extensions/espressif.esp-idf-extension-1.3.0/esp_debug_adapter/debug_adapter/command_processor.py", line 51, in __call__
    on_request(protocol_message)
  File "/home/neal/.vscode/extensions/espressif.esp-idf-extension-1.3.0/esp_debug_adapter/debug_adapter/command_processor.py", line 643, in on_next_request
    result = self.da.step()
  File "/home/neal/.vscode/extensions/espressif.esp-idf-extension-1.3.0/esp_debug_adapter/debug_adapter/debug_adapter.py", line 764, in step
    self._gdb.exec_next()
  File "/home/neal/.vscode/extensions/espressif.esp-idf-extension-1.3.0/esp_debug_adapter/debug_adapter/debug_backend/debug_backend/gdb.py", line 348, in exec_next
    raise DebuggerError('Failed to step over!')
debug_adapter.debug_backend.debug_backend.defs.DebuggerError: Failed to step over!
```
[image]

atanisoft, 2022-01-25:
> Are you using the latest master branch of IDF? I was getting those errors last month but some commits fixed that issue.

That seems to be it, I was on a version from a week or two back and it still had the issue. Now I'm not seeing any issues.

brianignacio5, 2022-01-25:
I've tested using the current GitHub master with the blink example and it seems to work (step over, stop at breakpoint, etc.)

Maybe you could share a small project so can I try to reproduce the issue.

meltdown03, 2022-01-25:
> I've tested using the current GitHub master with the blink example and it seems to work (step over, stop at breakpoint, etc.)
> 
> Maybe you could share a small project so can I try to reproduce the issue.
> 
> Remember to run the setup wizard or install the python packages again.
> 
> I'll fix the requirements.txt -> requirements.core.txt issue in another [PR](https://github.com/espressif/vscode-esp-idf-extension/pull/620)

I don't get the error that I posted above on the `Blink` example.  I did notice on the Blink example that stepping forware would _always_ Step Into functions. I don't remember that happening before. Is that what it's doing for you too? 

Any plans to add features like showing Assembly code or Registers like the Platformio debugger? Not that I know assembly, hehe.

brianignacio5, 2022-01-26:
> I don't get the error that I posted above on the Blink example. I did notice on the Blink example that stepping forware would always Step Into functions. I don't remember that happening before. Is that what it's doing for you too?

Not really, it step forward and step into seems to work as expected. I'll test on Linux and Windows too (was testing on MacOS).

> Any plans to add features like showing Assembly code or Registers like the Platformio debugger? Not that I know assembly, hehe.

For sure yes ! Part of the backlog hopefully this Q1 or Q2 we should have something for you.

brianignacio5, 2022-01-26:
The step issue seems to happen on gdb itself too. 

Tested in a terminal I have

```
idf.py gdb
Executing action: gdb
GNU gdb (crosstool-NG esp-2021r2) 9.2.90.20200913-git
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-host_apple-darwin12 --target=xtensa-esp32s3-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
configure_led () at ../main/blink_example_main.c:44
44      {
JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
esp32s3.cpu0: Debug controller was reset.
esp32s3.cpu0: Core was reset.
esp32s3.cpu0: Target halted, PC=0x500000EF, debug_reason=00000000
esp32s3.cpu0: Core was reset.
esp32s3.cpu0: Target halted, PC=0x40000400, debug_reason=00000000
esp32s3.cpu1: Debug controller was reset.
esp32s3.cpu1: Core was reset.
esp32s3.cpu1: Target halted, PC=0x40000400, debug_reason=00000000
Hardware assisted breakpoint 1 at 0x4200562f: file ../main/blink_example_main.c, line 74.
esp32s3.cpu1: Debug controller was reset.
esp32s3.cpu1: Core was reset.
esp32s3.cpu0: Target halted, PC=0x4200562F, debug_reason=00000001
Set GDB target to 'esp32s3.cpu0'
esp32s3.cpu1: Target halted, PC=0x42017D4A, debug_reason=00000000
[New Thread 1070548148]
[New Thread 1070546260]
[New Thread 1070536952]
[New Thread 1070541400]
[New Thread 1070535312]
[Switching to Thread 1070545912]

Thread 1 hit Temporary breakpoint 1, app_main () at ../main/blink_example_main.c:74
74          configure_led();
(gdb) s
configure_led () at ../main/blink_example_main.c:44
44      {
(gdb) 
```

brianignacio5, 2022-01-26:
Can we close this ?

atanisoft, 2022-01-26:
@brianignacio5 I think we can as long as the docs have been updated to guide users through configuring things end-to-end. I'm not 100% certain my config is correct and also willing to purge existing config and reconfigure per docs if they are available.

atanisoft, 2022-01-26:
One thing that was "missing" was the dependencies.txt -> dependencies.core.txt / dependencies.gdbgui.txt split, but I think that can be tracked via another issue.

meltdown03, 2022-01-26:
And maybe mention that python 2 is required.

brianignacio5, 2022-01-27:
The thing is that the debug adapter still uses an older `pygdbmi` package which depends on Python 2.7 and there are some changes required in the debug adapter and debug backend to update it. Hopefully we do it in this Q1.

With regards to the `requirements.txt`, it is fixed in [ this PR](https://github.com/espressif/vscode-esp-idf-extension/pull/620). The `requirements.gdbgui.txt` is not required for the extension since this package is debugging with `gdbgui`.

PR #617 (2022-01-29): update debug adapter to use generic chip classes
Support debug backend chip generic classes. Now for chips which has no explicit GDB/OOCD class in debug backend generic ones are used basing on provided target triple extracted from toolchain prefix. So it is mandatory to provide toolcahin prefix (--toolchain-prefix argument) for debugging new chips.

Fix #542 

Fix #603
closed 2022-01-29 by brianignacio5