espressif/vscode-esp-idf-extension #543: flasher_args.json file is missing from the build directory, can't proceed, please build properly!!

ua100k, 2021-10-11:
Hi, I have done the install of Visual Studio Code using the video provided espressif  - https://www.youtube.com/watch?v=Lc6ausiKvQM

One issue was that the scripts could not be executed in powershell due to group policy. I fixed that. 

I can build the code but I cannot flash it. This is the error that I get. 

I see this file in the esp-idf sub directory but shouldn't the build generate this file? 
Also where do I put this file so that I can flash using Visual Studio. 

Thank you.

brianignacio5, 2021-10-11:
Could you please more details about your issue ?

Please look at the [troubleshooting section](https://github.com/espressif/vscode-esp-idf-extension#troubleshooting) to get related output.

ua100k, 2021-10-11:
Hi Brian,

Thank you very much for responding. I deleted all the code and installation and followed your detailed instructions here 
https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/basic_use.md
and it is now working. 
One issue is that the certificates were expired. And the answer here to disable the certs. 
https://github.com/espressif/esp-idf/issues/7621

Thank you very much for your help. You guys have very responsive. I really appreciate it. 
Cheers

ua100k, 2021-10-11:
Thank you very much