espressif/vscode-esp-idf-extension #546: Failed to init debug adapter

ankit-thealchemist, 2021-10-17:
Hello

I am debugging blink example using the esp prog. I am using the esp idf extension in vs code. Ocd server runs fine. but on debugging its giving error failed to init debug adapter. 

I am attaching the esp doctor output and the screen shot for more info

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 6.3.9600 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files\Python38\Scripts\;C:\Program Files\Python38\;C:\Program Files\Git\cmd;C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\bin;C:\Program Files\Microsoft VS Code\bin;D:\Program Files\nodejs\;D:\Program Files\PuTTY\;d;C:\Users\programmer\AppData\Local\GitHubDesktop\bin;c:\windows\system32;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;D:\IOTCode\framework\toolchain4.0\.espressif\tools\cmake\3.16.4\bin;C:\Users\programmer\AppData\Roaming\npm 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.60.0 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.2.0 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) D:\IOTCode\framework\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
Custom extra paths (idf.customExtraPaths) d:\IOTCode\framework\toolchain4.0\.espressif\tools\xtensa-esp32-elf\esp-2021r1-8.4.0\xtensa-esp32-elf\bin;d:\IOTCode\framework\toolchain4.0\.espressif\tools\xtensa-esp32s2-elf\esp-2021r1-8.4.0\xtensa-esp32s2-elf\bin;d:\IOTCode\framework\toolchain4.0\.espressif\tools\xtensa-esp32s3-elf\esp-2021r1-8.4.0\xtensa-esp32s3-elf\bin;d:\IOTCode\framework\toolchain4.0\.espressif\tools\riscv32-esp-elf\esp-2021r1-8.4.0\riscv32-esp-elf\bin;d:\IOTCode\framework\toolchain4.0\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;d:\IOTCode\framework\toolchain4.0\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;d:\IOTCode\framework\toolchain4.0\.espressif\tools\cmake\3.16.4\bin;d:\IOTCode\framework\toolchain4.0\.espressif\tools\openocd-esp32\v0.10.0-esp32-20210401\openocd-esp32\bin;d:\IOTCode\framework\toolchain4.0\.espressif\tools\ninja\1.10.2;d:\IOTCode\framework\toolchain4.0\.espressif\tools\idf-exe\1.0.1;d:\IOTCode\framework\toolchain4.0\.espressif\tools\ccache\3.7;d:\IOTCode\framework\toolchain4.0\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"d:\\IOTCode\\framework\\toolchain4.0\\.espressif\\tools\\openocd-esp32\\v0.10.0-esp32-20210401/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) d:\IOTCode\framework\toolchain4.0\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe
Serial port (idf.port) COM4
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,target/esp32.cfg
ESP-IDF Tools Path (idf.toolsPath) d:\IOTCode\framework\toolchain4.0\.espressif
Git Path (idf.gitPath) d:\IOTCode\framework\toolchain4.0\.espressif\tools\idf-git\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-IDF Custom extra paths
Access to d:\IOTCode\framework\toolchain4.0\.espressif\tools\xtensa-esp32-elf\esp-2021r1-8.4.0\xtensa-esp32-elf\bin: true
Access to d:\IOTCode\framework\toolchain4.0\.espressif\tools\xtensa-esp32s2-elf\esp-2021r1-8.4.0\xtensa-esp32s2-elf\bin: true
Access to d:\IOTCode\framework\toolchain4.0\.espressif\tools\xtensa-esp32s3-elf\esp-2021r1-8.4.0\xtensa-esp32s3-elf\bin: true
Access to d:\IOTCode\framework\toolchain4.0\.espressif\tools\riscv32-esp-elf\esp-2021r1-8.4.0\riscv32-esp-elf\bin: true
Access to d:\IOTCode\framework\toolchain4.0\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin: true
Access to d:\IOTCode\framework\toolchain4.0\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin: true
Access to d:\IOTCode\framework\toolchain4.0\.espressif\tools\cmake\3.16.4\bin: true
Access to d:\IOTCode\framework\toolchain4.0\.espressif\tools\openocd-esp32\v0.10.0-esp32-20210401\openocd-esp32\bin: true
Access to d:\IOTCode\framework\toolchain4.0\.espressif\tools\ninja\1.10.2: true
Access to d:\IOTCode\framework\toolchain4.0\.espressif\tools\idf-exe\1.0.1: true
Access to d:\IOTCode\framework\toolchain4.0\.espressif\tools\ccache\3.7: true
Access to d:\IOTCode\framework\toolchain4.0\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) true
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version 2.30.1.windows.1
ESP-IDF version 4.3.1
Python version 3.8.7
Python's pip version 20.3.3
-------------------------------------------------- Python packages in idf.pythonBinPath ----------------------------------------------------
bitstring version: 3.1.9
Brotli version: 1.0.9
certifi version: 2021.5.30
cffi version: 1.14.6
charset-normalizer version: 2.0.4
click version: 8.0.3
colorama version: 0.4.4
construct version: 2.10.54
contextlib2 version: 21.6.0
cryptography version: 3.4.7
ecdsa version: 0.17.0
esp-windows-curses version: 0.1
Flask version: 0.12.5
Flask-Compress version: 1.10.1
[... 63 lines cut ...]
            "cppStandard": "c++17",
            "includePath": [
                "${config:idf.espIdfPath}/components/**",
                "${config:idf.espIdfPathWin}/components/**",
                "${config:idf.espAdfPath}/components/**",
                "${config:idf.espAdfPathWin}/components/**",
                "${workspaceFolder}/**"
            ],
            "browse": {
                "path": [
                    "${config:idf.espIdfPath}/components",
                    "${config:idf.espIdfPathWin}/components",
                    "${config:idf.espAdfPath}/components/**",
                    "${config:idf.espAdfPathWin}/components/**",
                    "${workspaceFolder}"
                ],
                "limitSymbolsToIncludedHeaders": false
            }
        }
    ],
    "version": 4
}
 
--------------------------------------------------------------------------------------------------------------------------------------------
[image]

brianignacio5, 2021-10-19:
Based on the OpenOCD output, there seems to be a lack of configuration of drivers in your computer.

You can use the following script to install the drivers as mentioned in the [Configure JTAG debugging](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/configure-ft2232h-jtag.html)

```
Invoke-WebRequest 'https://dl.espressif.com/dl/idf-env/idf-env.exe' -OutFile .\idf-env.exe; .\idf-env.exe driver install --espressif --ftdi --silabs
```

ankit-thealchemist, 2021-10-19:
> Based on the OpenOCD output, there seems to be a lack of configuration of drivers in your computer.
> 
> There is a documentation about [Configure JTAG debugging](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/configure-ft2232h-jtag.html) but you can use the following script to install the drivers in powershell:
> 
> ```
> Invoke-WebRequest 'https://dl.espressif.com/dl/idf-env/idf-env.exe' -OutFile .\idf-env.exe; .\idf-env.exe driver install --espressif --ftdi --silabs
> ```

sorry for the hassle. I am using the gpio 12 for blinking the led. This is hindering the communication with ocd server.  I am able to start the debugger but with some issues. 
I start the debugger by going to Run---> start debugging.  I got the following error...
[image]

I click on the x. This is the output of the ocd server .

[image]

Then I again start the debugging.. Then I got this error. 
[image]

After clicking the x. Then the debugger start automatically.

ankit-thealchemist, 2021-10-20:
> What is the application offset in your partition table. You can modify it in the launch.json with the `appOffset` as defined in the [debugging](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/DEBUGGING.md) documentation.

I donot know how its working now. every things working and I donot know why?

I am using the E-Prog. Could anyone know why we have boot and reset button on e-prog? seems like clicking thrm on few times does the trick!!