espressif/vscode-esp-idf-extension #568: [Bug Report]: this.openOcdConfigFilesList.forEach is not a function

Mair, 2021-11-05:
In vscode when using the extension, trying to launch OpenOCD I get an error message pop up this.openOcdConfigFilesList.forEach is not a function. If I run the command manually via command line it works.

**To Reproduce**
click on [OpenOCD Server] then start OpenOCD

**Expected behavior**
should work

**Screenshots**
[image]

**Environment (please complete the following information):**

- OS Name & Version: windows 11
- VSCode Version: 1.2.0
- ESP-IDF Version: v4.3.1 / release branch
- Python Version:  3.8.8

> **NOTE**: You can use the `ESP-IDF: Doctor command` to generate a report of your configuration.

see [doctor.txt](https://github.com/espressif/vscode-esp-idf-extension/files/7481564/doctor.txt)

**Output**

Click the menu `View` and select `Output`. This extension uses several output channels: ESP-IDF, ESP-IDF Debug Adapter, Heap Trace, OpenOCD and SDK Configuration Editor.
```
[05/11/2021, 6:48:10 pm] Unable to resolve configuration with compilerPath "c:\esp\tools\tools\xtensa-esp32-elf\esp-2021r1-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe
C:\esp\tools\.espressif\tools\xtensa-esp32-elf-clang\esp-12.0.1-20210823-x86_64-pc-windows-msvc\bin\xtensa-esp32-elf-gcc.exe".  Using "cl.exe" instead.
```
although... It seems to be correct
[image]

**Logs**
>If applicable, please share the log file which can be obtained from

see [esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/7481575/esp_idf_vsc_ext.log)

**Extension Configuration settings**
>Open Command Palette (F1 or View Menu -> Command Palette) and type `Preferences: Open Settings (JSON)` or use `{CURRENT_PROJECT}/.vscode/settings.json` to get the following extension configuration settings value:

```json
{
    "C_Cpp.intelliSenseEngine": "Default",
    "idf.adapterTargetName": "esp32",
    "idf.customExtraPaths": "c:\\esp\\tools\\tools\\xtensa-esp32-elf\\esp-2021r1-8.4.0\\xtensa-esp32-elf\\bin;c:\\esp\\tools\\tools\\xtensa-esp32s2-elf\\esp-2021r1-8.4.0\\xtensa-esp32s2-elf\\bin;c:\\esp\\tools\\tools\\xtensa-esp32s3-elf\\esp-2021r1-8.4.0\\xtensa-esp32s3-elf\\bin;c:\\esp\\tools\\tools\\riscv32-esp-elf\\esp-2021r1-8.4.0\\riscv32-esp-elf\\bin;c:\\esp\\tools\\tools\\esp32ulp-elf\\2.28.51-esp-20191205\\esp32ulp-elf-binutils\\bin;c:\\esp\\tools\\tools\\esp32s2ulp-elf\\2.28.51-esp-20191205\\esp32s2ulp-elf-binutils\\bin;c:\\esp\\tools\\tools\\cmake\\3.16.4\\bin;c:\\esp\\tools\\tools\\openocd-esp32\\v0.10.0-esp32-20210902\\openocd-esp32\\bin;c:\\esp\\tools\\tools\\ninja\\1.10.2;c:\\esp\\tools\\tools\\idf-exe\\1.0.1;c:\\esp\\tools\\tools\\ccache\\3.7;c:\\esp\\tools\\tools\\dfu-util\\0.9\\dfu-util-0.9-win64",
    "idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"c:\\\\esp\\\\tools\\\\tools\\\\openocd-esp32\\\\v0.10.0-esp32-20210902/openocd-esp32/share/openocd/scripts\",\"IDF_CCACHE_ENABLE\":\"1\"}",
    "idf.espIdfPathWin": "c:\\esp\\esp-idf",
    "idf.openOcdConfigs": "interface/ftdi/esp32_devkitj_v1.cfg,target/esp32.cfg",
    "idf.portWin": "COM3",
    "idf.pythonBinPathWin": "c:\\esp\\tools\\python_env\\idf4.3_py3.8_env\\Scripts\\python.exe",
    "idf.toolsPathWin": "c:\\esp\\tools"
}
```

- `idf.espIdfPath`  `c:\esp\esp-idf`
- `idf.pythonBinPath`  `python
- `idf.openOcdConfigs` its empty.... hmmmm 
- `idf.customExtraPaths` `c:\esp\tools\tools\xtensa-esp32-elf\esp-2021r1-8.4.0\xtensa-esp32-elf\bin;c:\esp\tools\tools\xtensa-esp32s2-elf\esp-2021r1-8.4.0\xtensa-esp32s2-elf\bin;c:\esp\tools\tools\xtensa-esp32s3-elf\esp-2021r1-8.4.0\xtensa-esp32s3-elf\bin;c:\esp\tools\tools\riscv32-esp-elf\esp-2021r1-8.4.0\riscv32-esp-elf\bin;c:\esp\tools\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;c:\esp\tools\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;c:\esp\tools\tools\cmake\3.16.4\bin;c:\esp\tools\tools\openocd-esp32\v0.10.0-esp32-20210902\openocd-esp32\bin;c:\esp\tools\tools\ninja\1.10.2;c:\esp\tools\tools\idf-exe\1.0.1;c:\esp\tools\tools\ccache\3.7;c:\esp\tools\tools\dfu-util\0.9\dfu-util-0.9-win64`
- `idf.customExtraVars` {"OPENOCD_SCRIPTS":"c:\\esp\\tools\\tools\\openocd-esp32\\v0.10.0-esp32-20210902/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}`.
- `idf.adapterTargetName` esp32`.
- `idf.customAdapterTargetName` empty

**Additional context**
This happens in both my local machine and if I use the docker container

brianignacio5, 2021-11-08:
Thank you for submitting this report. There should be an error validation when `this.openOcdConfigFilesList` is not defined. 

Apparently your error seems to come from having an empty `this.openOcdConfigFilesList`. Please consider that if you set a value in `{CURRENT_PROJECT}/.vscode/settings.json` even an empty one, it will override the value in User or Workspace settings in `Preferences: Open Settings (JSON)`. Just delete the value in your project settings.json so the value used is the one from User settings. 

About [Visual Studio Code settings precedence](https://code.visualstudio.com/docs/getstarted/settings#_settings-precedence).

Mair, 2021-11-08:
@brianignacio5 Thanks for the response.
my settings were generated by using the extensions *ESP-IDF: New Project* which has the following
```json
{
    "C_Cpp.intelliSenseEngine": "Tag Parser",
    "idf.adapterTargetName": "esp32",
    "idf.customExtraPaths": "c:\\esp\\tools\\tools\\xtensa-esp32-elf\\esp-2021r1-8.4.0\\xtensa-esp32-elf\\bin;c:\\esp\\tools\\tools\\xtensa-esp32s2-elf\\esp-2021r1-8.4.0\\xtensa-esp32s2-elf\\bin;c:\\esp\\tools\\tools\\xtensa-esp32s3-elf\\esp-2021r1-8.4.0\\xtensa-esp32s3-elf\\bin;c:\\esp\\tools\\tools\\riscv32-esp-elf\\esp-2021r1-8.4.0\\riscv32-esp-elf\\bin;c:\\esp\\tools\\tools\\esp32ulp-elf\\2.28.51-esp-20191205\\esp32ulp-elf-binutils\\bin;c:\\esp\\tools\\tools\\esp32s2ulp-elf\\2.28.51-esp-20191205\\esp32s2ulp-elf-binutils\\bin;c:\\esp\\tools\\tools\\cmake\\3.16.4\\bin;c:\\esp\\tools\\tools\\openocd-esp32\\v0.10.0-esp32-20210902\\openocd-esp32\\bin;c:\\esp\\tools\\tools\\ninja\\1.10.2;c:\\esp\\tools\\tools\\idf-exe\\1.0.1;c:\\esp\\tools\\tools\\ccache\\3.7;c:\\esp\\tools\\tools\\dfu-util\\0.9\\dfu-util-0.9-win64",
    "idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"c:\\\\esp\\\\tools\\\\tools\\\\openocd-esp32\\\\v0.10.0-esp32-20210902/openocd-esp32/share/openocd/scripts\",\"IDF_CCACHE_ENABLE\":\"1\"}",
    "idf.espIdfPathWin": "c:\\esp\\esp-idf",
    //"idf.openOcdConfigs": "C:\\projects\\esp32-projects\\test-openocd\\debug\\ftdi_ft2322.cfg, target/esp32.cfg",
    "idf.portWin": "COM12",
    "idf.pythonBinPathWin": "c:\\esp\\tools\\python_env\\idf4.3_py3.8_env\\Scripts\\python.exe",
    "idf.toolsPathWin": "c:\\esp\\tools",
}
```
If it's in error perhaps it might be wise to remove it from the generated template.😁

Once I removed that line I retried and I get the following
[image]

even though the project was built. hitting the build button seems to go into an infinite loop of building /%IDF_PATH%/examples/get_started/hello_world 🤔

any way... hitting the button again and I get 
[image]

and my output displays the following
```bash
Open On-Chip Debugger  v0.10.0-esp32-20210902 (2021-09-02-09:33)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

adapter speed: 20000 kHz

WARNING: boards/esp32-wrover.cfg is deprecated, and may be removed in a future release.
         If your board is ESP32-WROVER-KIT, use board/esp32-wrover-kit-1.8v.cfg instead.
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
❌ Error: libusb_open() failed with LIBUSB_ERROR_NOT_FOUND
❌ Error: no device found
Error: unable to open ftdi device with vid 0403, pid 6010, description '*', serial '*' at bus location '*'
❌ Error: no device found
Error: unable to open ftdi device with vid 0403, pid 6014, description '*', serial '*' at bus location '*'

Assertion failed!

Program: c:\esp\tools\tools\openocd-esp32\v0.10.0-esp32-20210902\openocd-esp32\bin\openocd.exe
File: ../src/jtag/core.c, Line 343

Expression: jtag_trst == 0
[Stopped] : OpenOCD Server
```

So it looks like its closer but not there yet I think it might be the cfg file its using which brings me to my next question...
Looking at the user settings I see
```json
 "idf.gitPath": "c:\\esp\\tools\\tools\\idf-git\\2.30.1\\cmd\\git.exe",
    "idf.espIdfPathWin": "c:\\esp\\esp-idf",
    "idf.pythonBinPathWin": "c:\\esp\\tools\\python_env\\idf4.3_py3.8_env\\Scripts\\python.exe",
    "idf.toolsPathWin": "c:\\esp\\tools",
    "idf.customExtraPaths": "c:\\esp\\tools\\tools\\xtensa-esp32-elf\\esp-2021r1-8.4.0\\xtensa-esp32-elf\\bin;c:\\esp\\tools\\tools\\xtensa-esp32s2-elf\\esp-2021r1-8.4.0\\xtensa-esp32s2-elf\\bin;c:\\esp\\tools\\tools\\xtensa-esp32s3-elf\\esp-2021r1-8.4.0\\xtensa-esp32s3-elf\\bin;c:\\esp\\tools\\tools\\riscv32-esp-elf\\esp-2021r1-8.4.0\\riscv32-esp-elf\\bin;c:\\esp\\tools\\tools\\esp32ulp-elf\\2.28.51-esp-20191205\\esp32ulp-elf-binutils\\bin;c:\\esp\\tools\\tools\\esp32s2ulp-elf\\2.28.51-esp-20191205\\esp32s2ulp-elf-binutils\\bin;c:\\esp\\tools\\tools\\cmake\\3.16.4\\bin;c:\\esp\\tools\\tools\\openocd-esp32\\v0.10.0-esp32-20210902\\openocd-esp32\\bin;c:\\esp\\tools\\tools\\ninja\\1.10.2;c:\\esp\\tools\\tools\\idf-exe\\1.0.1;c:\\esp\\tools\\tools\\ccache\\3.7;c:\\esp\\tools\\tools\\dfu-util\\0.9\\dfu-util-0.9-win64",
    "idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"c:\\\\esp\\\\tools\\\\tools\\\\openocd-esp32\\\\v0.10.0-esp32-20210902/openocd-esp32/share/openocd/scripts\",\"IDF_CCACHE_ENABLE\":\"1\"}",
    "idf.adapterTargetName": "esp32",
    "git.autofetch": true,
    "git.confirmSync": false,
    "idf.flashType": "UART",
    "idf.portWin": "COM3",
```
So no mention of anything openOCD related in there
my jtag cfg file that works for my jtag looks like this
```bash
interface ftdi
ftdi_vid_pid 0x0403 0x6010
ftdi_layout_init 0x0038 0x003b
transport select jtag
adapter_khz 200
```
I haven't found one on the OPENOCD folder that has worked for me so I usually copy the above cfg file to my project and reference it directly when I launch openOCD via the command line.
How do I tell the extension to use that cfg file?

brianignacio5, 2021-11-08:
Just change `idf.openOcdConfigs` from  `"C:\\projects\\esp32-projects\\test-openocd\\debug\\ftdi_ft2322.cfg, target/esp32.cfg"` to `//"idf.openOcdConfigs": ["C:\\projects\\esp32-projects\\test-openocd\\debug\\ftdi_ft2322.cfg"," target/esp32.cfg"]` or use the `Select OpenOCD Board Configuration` extension command.

I'll fix the issue of the New Project wizards, thanks. Your openOCD output error suggest 2 things:

1) Need to set `idf.openOcdConfigs` to the correct value of your board.
2) You need to configure the usb drivers as explained [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/configure-ft2232h-jtag.html#windows)

Mair, 2021-11-08:
@brianignacio5 Thanks, I'll try number1. Point number 2 has already been done and I am able to debug using my own launch.json

Mair, 2021-11-08:
@brianignacio5 
seems like I had a dud cable **_DOH!_**

I changed my settings.json to
```
{
   ...
    "idf.openOcdConfigs": ["C:\\projects\\esp32-projects\\test-openocd\\debug\\ftdi_ft2322.cfg","target/esp32.cfg"],
    ...
}
```
and now my output says: a couple of errors (see screen shot)
[image]

So it looks like openocd starts but my config on the cfg files seems wrong,

as mentioned it does work with my own settings.json which looks like this
```json
{
      "name": "ESP32 OpenOCD",
      "type": "cppdbg",
      "request": "launch",
      "miDebuggerPath": "C:/esp/tools/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gdb.exe",
      "cwd": "${workspaceFolder}/build",
      "program": "C:/projects/esp32-projects/test-openocd/build/test-openocd.elf",
      "setupCommands": [
        {
          "description": "Enable pretty-printing for gdb",
          "text": "-enable-pretty-printing",
          "ignoreFailures": true
        },
        { "text": "file 'C:/projects/esp32-projects/test-openocd/build/test-openocd.elf'" },
        { "text": "target remote 127.0.0.1:3333"},
        {"text": "set remote hardware-watchpoint-limit 2" },
        { "text": "monitor reset halt" },
        { "text": "flushregs" }
      ],
      "externalConsole": false,
      "logging": {}
    }
```
and I launch it like this
`openocd -f .\debug\ftdi_ft2322.cfg -f .\debug\esp-wroom-32.cfg`

the above looks like this
[image]

So I can get it to work but using the extension would be super nice.

brianignacio5, 2021-11-08:
In `launch.json` add `"skipVerifyAppBinBeforeDebug": false` and try again. This is an issue not yet fixed in marketplace but already fixed in master.

Mair, 2021-11-08:
tried that
```json
 {
      "type": "espidf",
      "name": "Launch",
      "request": "launch",
      "skipVerifyAppBinBeforeDebug": false
    }
```
unfortunately same error
[image]

brianignacio5, 2021-11-08:
Can you share the latest extension log ? I'm not sure what is happening in the extension here.

Can you launch openOCD from the extension itself ?

brianignacio5, 2021-11-09:
The elf file is recognized automatically from your projects' CMakeLists.txt file. Is your project using the CMake build system ?

Mair, 2021-11-09:
@brianignacio5, yip. My setup as follows
1. installed the idf using the tool installer version 4.3.1 (release branch)
2. in vscode ->command pallet -> Configure esp-idf extension 
3. Express 
4. Selected the IDF on my system and redownloaded the tool chain

This is a brand new fresh windows 11 install. I bet if you have a vm and set it up in the same way you would get the same results.
This is my jtag https://www.aliexpress.com/item/1005002448969402.html?spm=a2g0s.9042311.0.0.48b64c4daRIxZX

log file 
[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/7503124/esp_idf_vsc_ext.log)

brianignacio5, 2021-11-10:
Oh sorry was supposed to be `"skipVerifyAppBinBeforeDebug": true`

Mair, 2021-11-10:
thanks @brianignacio5 I'll give it a try this evening and let you know how it goes

Mair, 2021-11-11:
@brianignacio5 
I'm getting a different error with that

[image]

brianignacio5, 2021-11-11:
There should be an error in `esp_idf_vsc_ext.log` or `OpenOCD` or `Debug Adapter` Output channels.

Mair, 2021-11-11:
Debug Adaptor
```
DEBUG_ADAPTER_STARTED
2021-11-11 22:05:51,801 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STARTED
DEBUG_ADAPTER_READY2CONNECT
2021-11-11 22:05:51,808 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT
DEBUG_ADAPTER_STOPPED
2021-11-11 22:06:02,032 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STOPPED
[Stopped] : ESP-IDF Debug Adapter
```
OpenOCD
```
Open On-Chip Debugger  v0.10.0-esp32-20210902 (2021-09-02-09:33)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

adapter speed: 200 kHz

Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
❌ Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : clock speed 200 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32.cpu0: Debug controller was reset.
Info : esp32.cpu0: Core was reset.
Info : esp32.cpu1: Debug controller was reset.
Info : esp32.cpu1: Core was reset.
Info : Listening on port 3333 for gdb connections
Info : esp32.cpu0: Debug controller was reset.
Info : esp32.cpu0: Core was reset.
Info : esp32.cpu1: Debug controller was reset.
Info : esp32.cpu1: Core was reset.
Info : accepting 'gdb' connection on tcp/3333
Warn : No symbols for FreeRTOS!
Info : esp32.cpu0: Target halted, PC=0x400E3282, debug_reason=00000000
Info : Set GDB target to 'esp32.cpu0'
Info : esp32.cpu1: Target halted, PC=0x400E3282, debug_reason=00000000
Info : esp32.cpu0: Target halted, PC=0x400928DA, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 26 KB
Info : Flash mapping 1: 0x20020 -> 0x400d0020, 78 KB
Info : esp32.cpu0: Target halted, PC=0x400928DA, debug_reason=00000001
Info : Auto-detected flash bank 'esp32.cpu0.flash' size 8192 KB
Info : Using flash bank 'esp32.cpu0.flash' size 8192 KB
Info : esp32.cpu0: Target halted, PC=0x400928DA, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 26 KB
Info : Flash mapping 1: 0x20020 -> 0x400d0020, 78 KB
Info : Using flash bank 'esp32.cpu0.irom' size 80 KB
Info : esp32.cpu0: Target halted, PC=0x400928DA, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 26 KB
Info : Flash mapping 1: 0x20020 -> 0x400d0020, 78 KB
Info : Using flash bank 'esp32.cpu0.drom' size 28 KB
Info : New GDB Connection: 1, Target esp32.cpu0, state: halted
Info : dropped 'gdb' connection
```
Interesting I get the ❌ Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED even when I use it with my config and it does still work

[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/7519962/esp_idf_vsc_ext.log)

Hope that helps

brianignacio5, 2021-11-16:
Could you increase the log level of the debug adapter by setting in `launch.json` and share the `Debug Adapter` output ?

launch.json
```json
{
   ...
   "logLevel": 5,
}
```

Still not sure what is happening in the debug adapter to be failing, need more details in the Debug Adapter Output.

Mair, 2021-11-25:
Hi @brianignacio5, Sorry for the delay.

this is my launch.json configuration now
```json
{
      "type": "espidf",
      "name": "Launch",
      "request": "launch",
      //"skipVerifyAppBinBeforeDebug": false,
      "skipVerifyAppBinBeforeDebug": true,
      "logLevel": 5,
    },
```
my output window
```bash
2021-11-26 08:58:43,239 - Debug Adapter (main) - DEBUG - START_TIME_2021_11_26_08-58-43
2021-11-26 08:58:43,239 - Debug Adapter (main) - DEBUG - Init errors:None
2021-11-26 08:58:43,239 - Debug Adapter (main) - INFO - Working directory: C:\Users\mairs\AppData\Local\Programs\Microsoft VS Code
2021-11-26 08:58:43,239 - Debug Adapter (main) - INFO - Arguments: 
{   'app_flash_off': 65536,
    'board_type': None,
    'cmdfile': None,
    'core_file': (),
    'debug': 5,
    'developer_mode': 'none',
    'device_name': 'esp32',
    'elfpath': (   'c:\\projects\\esp32-projects\\test-openocd\\build\\test-openocd.elf',),
    'log_file': 'c:\\projects\\esp32-projects\\test-openocd\\debug.log',
    'log_mult_files': None,
    'log_no_debug_console': None,
    'oocd': 'openocd',
    'oocd_args': None,
    'oocd_ip': 'localhost',
    'oocd_mode': 'connect_to_instance',
    'oocd_scripts': None,
    'port': 43474,
    'postmortem': False,
    'tmo_scale_factor': 1,
    'toolchain_prefix': None}
DEBUG_ADAPTER_STARTED
2021-11-26 08:58:43,239 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STARTED
2021-11-26 08:58:43,239 - Debug Adapter (main) - INFO - Starting. Cmd: c:\Users\mairs\.vscode\extensions\espressif.esp-idf-extension-1.2.0\esp_debug_adapter\debug_adapter_main.py -d 5 -e c:\projects\esp32-projects\test-openocd\build\test-openocd.elf -l c:\projects\esp32-projects\test-openocd\debug.log -p 43474 -dn esp32 -a 0x10000

2021-11-26 08:58:43,247 - Debug Adapter (main) - INFO - Listening on port 43474 (host: mair-hh,local ip: 192.168.0.117)
DEBUG_ADAPTER_READY2CONNECT
2021-11-26 08:58:43,247 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT
2021-11-26 08:58:43,248 - Debug Adapter (main) - DEBUG - Got connection
2021-11-26 08:58:43,249 - Debug Adapter (ReaderThread) - DEBUG - read line: >>b'Content-Length: 410\\r\\n'<<

2021-11-26 08:58:43,249 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Debug adapter -> Extension: DEBUG_ADAPTER_STARTED\n", "source": {}}, "seq": 0}

2021-11-26 08:58:43,250 - Debug Adapter (ReaderThread) - DEBUG - read line: >>b'\\r\\n'<<

2021-11-26 08:58:43,250 - Debug Adapter(Command Processor) - DEBUG - Got json: {
    "arguments": {
        "adapterID": "espidf",
        "clientID": "vscode",
        "clientName": "Visual Studio Code",
        "columnsStartAt1": true,
        "linesStartAt1": true,
        "locale": "en-us",
        "pathFormat": "path",
        "supportsInvalidatedEvent": true,
        "supportsMemoryReferences": true,
        "supportsProgressReporting": true,
        "supportsRunInTerminalRequest": true,
        "supportsVariablePaging": true,
        "supportsVariableType": true
    },
    "command": "initialize",
[... 109 lines cut ...]
Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT
Got connection
Got json: {
    "arguments": {
        "adapterID": "espidf",
        "clientID": "vscode",
        "clientName": "Visual Studio Code",
        "columnsStartAt1": true,
        "linesStartAt1": true,
        "locale": "en-us",
        "pathFormat": "path",
        "supportsInvalidatedEvent": true,
        "supportsMemoryReferences": true,
        "supportsProgressReporting": true,
        "supportsRunInTerminalRequest": true,
        "supportsVariablePaging": true,
        "supportsVariableType": true
    },
    "command": "initialize",
    "seq": 1,
    "type": "request"
}
Created gdb object of type: GdbEsp32

```

brianignacio5, 2021-12-10:
@Mair  sorry for late reply.

Have you configured the usb drivers as mentioned here: 

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/configure-ft2232h-jtag.html#configure-usb-drivers

brianignacio5, 2021-12-14:
The original issue is fixed with #569 and the debug error seems like a different issue. 

I've tested this with ESP-IDF v4.3.1 and v0.10.0-esp32-20210401 (the one listed in this esp-idf version tools) for ESP32 and debug seems to be working. Please comment here for further review.

closed 2021-12-14 by brianignacio5