espressif/vscode-esp-idf-extension #570: [Bug Report]: Debug Adapter (main) - ERROR - 'frame'

stefaandesmet2003, 2021-11-08:
Hi, 
this vscode extension is great, but I'm having an issue to get debugging working with the 'espidf' launch type.
my setup is a 'Wemos' ESP32 devboard connected to a FT232H USB adapter for JTAG debugging.
the ESP-IDF Debug adapter plugin systematically raises an exception before it even connects to the running openocd server.
The extension shows a popup with no information except 'open launch.json' :

The espidf launch config is the default as per instruction
[image]

The setup does work correctly with a cppdbg launch configuration. Both configurations are attached. (doctor.txt)
I had the issue with ESP-IDF4.3, now moved to master branch, but same result.
OpenOCD view shows a running openocd server that receives no command from the ESP-IDF Debug Adapter plugin.
Thanks in advance for helping out!
Stefaan

**Environment (please complete the following information):**

[doctor.txt](https://github.com/espressif/vscode-esp-idf-extension/files/7498765/doctor.txt)

--------------------------------------------------------------------------------------------------------------------------------------------
OpenOCD View:
--------------------------------------------------------------------------------------------------------------------------------------------

Open On-Chip Debugger  v0.10.0-esp32-20210902 (2021-09-02-09:38)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

adapter speed: 20000 kHz

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32.cpu0: Target halted, PC=0x400D483F, debug_reason=00000001
Info : esp32.cpu1: Target halted, PC=0x400E550A, debug_reason=00000000
Info : Listening on port 3333 for gdb connections

--------------------------------------------------------------------------------------------------------------------------------------------
ESP-IDF Debug Adapter View:
--------------------------------------------------------------------------------------------------------------------------------------------

2021-11-08 17:53:28,127 - Debug Adapter (main) - DEBUG - START_TIME_2021_11_08_17-53-28
2021-11-08 17:53:28,128 - Debug Adapter (main) - DEBUG - Init errors:None
2021-11-08 17:53:28,128 - Debug Adapter (main) - INFO - Working directory: /home/stefaan
2021-11-08 17:53:28,129 - Debug Adapter (main) - INFO - Arguments: 
{   'app_flash_off': 65536,
    'board_type': None,
    'cmdfile': None,
    'core_file': (),
    'debug': 4,
    'developer_mode': 'none',
    'device_name': '',
    'elfpath': ('/home/stefaan/Documents/dev/esp32/blink/build/blink.elf',),
    'log_file': '/home/stefaan/Documents/dev/esp32/blink/debug.log',
    'log_mult_files': None,
    'log_no_debug_console': None,
    'oocd': 'openocd',
    'oocd_args': None,
    'oocd_ip': 'localhost',
    'oocd_mode': 'connect_to_instance',
    'oocd_scripts': None,
    'port': 43474,
    'postmortem': False,
    'tmo_scale_factor': 1,
    'toolchain_prefix': None}
DEBUG_ADAPTER_STARTED
[... 239 lines cut ...]
2021-11-08 17:53:33,609 - Debug Adapter (main) - DEBUG - Adapter is Stopped
DEBUG_ADAPTER_STOPPED
2021-11-08 17:53:33,610 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STOPPED
2021-11-08 17:53:33,614 - Debug Adapter(Command Processor) - DEBUG - Processed command: disconnect
[Stopped] : ESP-IDF Debug Adapter

**Logs**
If applicable, please share the log file which can be obtained from

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

**Extension Configuration settings**
Open Command Palette (F1 or View Menu -> Command Palette) and type `Preferences: Open Settings (JSON)` or use `{CURRENT_PROJECT}/.vscode/settings.json` to get the following extension configuration settings value:

- `idf.espIdfPath` (MacOS or Linux) or `idf.espIdfPathWin` (Windows) also called `Path to locate ESP-IDF framework (IDF_PATH)` in `Preferences: Open Settings (UI)`.
- `idf.pythonBinPath` (MacOS or Linux) or `idf.pythonBinPathWin` (Windows) also called `Python absolute binary path used to execute ESP-IDF Python Scripts` in `Preferences: Open Settings (UI)`.
- `idf.openOcdConfigs` also called `List of configuration files inside OpenOCD Scripts directory` in `Preferences: Open Settings (UI)`.
- `idf.customExtraPaths` also called `Paths to be appended to PATH` in `Preferences: Open Settings (UI)`.
- `idf.customExtraVars` also called `Variables to be added to system environment variables` in `Preferences: Open Settings (UI)`.
- `idf.adapterTargetName` also called `Target name for ESP-IDF Debug Adapter` in `Preferences: Open Settings (UI)`.
- `idf.customAdapterTargetName` is used when `idf.adapterTargetName` is set to `custom`.

**Additional context**
Add any other context about the problem here.

stefaandesmet2003, 2021-11-08:
Hi, 
apparently device name , wasn't set; must have missed that somehow. corrected now to "idf.adapterTargetName": "esp32" in settings.json
And problem looks mostly solved.
Sorry to spam this repo :/

closed 2021-11-08 by stefaandesmet2003

brianignacio5, 2021-11-09:
I'm glad you had it fixed! 

Let us know if there something we can help with 😄